esp8266/Arduino #1483: nodeMCU + ArduinoIDE zero crossing

brunoudi, 2016-01-22:
I'm trying to make a dimmer with nodeMCU and ArduinoIDE, but I'm having a problem with the zero cross detection.

Considering 60hz, i would get a interrupt every 8,33 ms, but in 10% of the cases I get a interrupt at 7,86ms and then another at 0,47ms. 

Those "mistakes" is causing the lamp to flicker.

My code is below, as my schematic.

I would like to hear your opinion and sugetions.

Thanks.

const int pulsePin = D2; //pin to pulse the triac

cont inst zcPin = D1; //pin to zero cross detection

int tempo = 0; //time to wait to send the pulse in microseconds

void zeroCrossing() {
	if (tempo == 0) {
		digitalWrite(pulsePin, LOW);
	}
	else {
		delayMicroseconds(tempo);
		digitalWrite(pulsePin, HIGH);
		delayMicroseconds(20);
		digitalWrite(pulsePin, LOW);
	}

void setup() {
	Serial.begin(115200);
	pinMode(pulsePin, OUTPUT);
	pinMode(zcPin, INPUT);
	attachInterrupt(digitalPinToInterrupt(zcPin), zeroCrossing, FALLING);
}

void loop () {
int temp;
	if (Serial.available()) {
		temp = Serial.parseInt();
		if (temp != 0) {
		tempo = temp;
		}
	}
}
[image]

Links2004, 2016-01-22:
if you get the interrupt before the real one it must be some noise on the line.
simulate or calculate if you can add some capacitor to filter it. 

keep in mind the the WiFi and OS from the SDK, 
have interrupts too and this can delay other interrupts.
for this reason i habe used a AVR + a ESP01 for my project.

I have used this on the AVR:
[image]

drmpf, 2016-01-22:
I am making a wifi controlled house light.  A retrofit for an existing light without changing the house wiring and keeping the existing wall switch operational.
There is the circuit.  Note this is only detecting missing AC not zero crossing, but has some features that may help you.  
[image]

It has larger resistors to reduce the power loss, a high frequency filter on the front of the opto-isolator, and a reverse diode to protect the optos led from reverse voltage and a diode on the GPIO2 input so you can still flash the ESP with the circuit connected.

Zero crossing can by very noise prone.  You may do better to use a circuit like this and infer the next zero cross from the mid point of the input pulse.  i.e. top of the AC waveform

drmpf, 2016-01-22:
p.s. As been mentioned elsewhere the standard ESP wifi blocks on sending packets so you should use the newer Async library. https://github.com/esp8266/Arduino/issues/1430
  Even then you may have delays introduced by the Wifi code.  The newer ESP32 may solve this

me-no-dev, 2016-01-24:
@drmpf looking at the code above, can't the triac be controlled by the PWM driver? So you do not need to delay and switch in the callback. If only one interrupt is attached, I have measured about 2-3us delay between the event and the time the callback is called. At 50Hz that should be OK :)
What are the requirements for the signal to the triac?

me-no-dev, 2016-01-24:
Maybe if not the PWM driver, then use timer0 or timer1 to control switching and not delay at all in the callback.
I have noticed that as little you do in those callback, the more the end result is reliable. Switching a pin and starting the timer will take less than 1us to complete and let the rest of the system run without noticing at all that process. Then when the timer hits, you only switch the pin (way under 1us) and you'll be ready to get the next zero pulse

ilendemli, 2016-04-15:
@me-no-dev do you have an example? @brunoudi have you figured a reliable way to detect zero crossing? Is there a way to add timer intervals?

me-no-dev, 2016-04-15:
https://gist.github.com/me-no-dev/2e185a909000f6ac9605
this is one example using timers and interrupts

mkeyno, 2016-04-16:
hi @me-no-dev , the links you passed,it is  implemented in ESP?, I don't recognize GPOS  or GPI term in the code
@drmpf why don't you using MOC3040 as cross detector , I saw many sketch use such opto device 
[image]
[image]

me-no-dev, 2016-04-16:
yes that is ESP :) it's direct register manipulation (if you look in the core files you will see many things like that)

devyte, 2017-10-18:
Interesting discussion, but closing per #3655 .

closed 2017-10-18 by devyte

junior700, 2018-06-10:
I sugest this circuit: 
[https://www.circuitar.com.br/media/product/26/files/Zero_Cross_v1.0.pdf](url)

junior700, 2018-06-10:
https://www.circuitar.com.br/media/product/26/files/Zero_Cross_v1.0.pdf

junior700, 2018-06-10:
> @drmpf why don't you using MOC3040 as cross detector , I saw many sketch use such opto device

Hi @mkeyno , this circuit is useful for heating resistor, but has a lot of flickering on the lamps!

junior700, 2018-06-10:
You could use multitask (free RTOS) with ESP.