esp8266/Arduino #1486: EEPROM Lost after Reboot/Power Off

codemasterPL, 2016-01-23:
Hi Guys,
So far i had _no_ issues with programming Esp8266, great platform for IoT until i started saving/loading configuration to/from EEPROM.

The Setup i am having:
Mac OSX 10.11.2 (El Captain)
Arduino IDE: 1.6.7
Programmer: pyesptool (only this one works with Mac)
Board: LoLin Node MCU V3 (http://www.upgradeindustries.com/media/ebay/Photos/nodemcu_new/nodemcu_front_wm.jpg)

programmers.txt
pyesptool.name=pyesptool
pyesptool.communication=serial
pyesptool.protocol=esp
pyesptool.program.protocol=esp
pyesptool.program.tool=pyesptool
pyesptool.program.extra_params=

platform.txt
tools.pyesptool.cmd=esptool.py
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}/hardware/tools/esp8266/pyesptool
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern=python "{path}/{cmd}" {program.verbose} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"

Problem:
I can easily save/load data from EEPROM as long as ESP8266 is powered ON.
When i poweroff and poweronn - EEPROM comes back with random values

Code:

void CSense::LoadConfig(tConfig& rtConfig)
{ 
  //
  rtConfig.m_strWiFiSSID = "";
  rtConfig.m_strWiFiPassword = "";
  
  
  // Laod SSID First
  for (int i = 0; i < m_iMaxSSIDLength; i++)
  {
      rtConfig.m_strWiFiSSID += char(EEPROM.read(i));
  }

   // Load PASS 
   for (int i = 0; i < m_iMaxPasswordLength; i++)
   {
      rtConfig.m_strWiFiPassword += char(EEPROM.read(m_iMaxSSIDLength+i));
   }  
 
}
void CSense::SaveConfig(const tConfig& rtConfig)
{  
  // Clear EEPROM
  for(int a = 0; a < m_iMaxSSIDLength + m_iMaxPasswordLength;a++)
  {
    EEPROM.write(a, 0); 
  }

  // Save SSID
  for (int i = 0; i < m_iMaxSSIDLength; i++)
  {
    if(i < rtConfig.m_strWiFiSSID.length())
      EEPROM.write(i, rtConfig.m_strWiFiSSID[i]); 
  }

  // Save Password
  for (int i = 0; i < m_iMaxPasswordLength; i++)
  {
    if(i < rtConfig.m_strWiFiPassword.length())
      EEPROM.write(i + m_iMaxSSIDLength, rtConfig.m_strWiFiPassword[i]);    
  }
  EEPROM.commit();
  yield();

}

Please let me know if there is anything else i need to share, i am new to this world - be patient :)

igrr, 2016-01-23:
Does your esptool.py contain [this fix](https://github.com/themadinventor/esptool/pull/53)?

Off topic, but do you know that it's not necessary to use EEPROM to save SSID/password? These are stored by the SDK itself, so for most use cases you can just set SSID/password once and the ESP will connect automatically.

codemasterPL, 2016-01-23:
Hi @igrr, thank you for your reply.

the reason i am saving and loading ssid/pass from EEPROM is due to the fact that the board starts 'unconfigured' in default - meaning, enters AP mode for user to be able to set his own ssid/password via http. After the user saves this,data gets saved into EEPROM and board is rebooted. From now on board is connecting to the AP configured by user. 

I am sorry but how do i apply this fix to the esptool.py ?

codemasterPL, 2016-01-23:
Ok, downloaded newest esptool from the url provided above, still trash in EEPROM after reset.

igrr, 2016-01-23:
Okay, does `EEPROM.commit();` return `true`?

codemasterPL, 2016-01-23:
Good point, 

EEPROM.commit(); returns false;

igrr, 2016-01-23:
Please check that your flash size setting in the IDE matches the size of your flash chip.
There is a CheckFlashConfig example bundled with the IDE which can help you with that task.

codemasterPL, 2016-01-23:
Working on it, thanks very MUCH!

codemasterPL, 2016-01-23:
Here are the results of the sketch:

Flash ide  size: 524288
Flash ide speed: 40000000
Flash ide mode:  QIO
Flash Chip configuration wrong!

WereCatf, 2016-01-23:
I have three LoLin Nodemcu V3 - boards and they all return correct configuration. The issue may lie with your setup instead; I am using Arduino IDE 1.6.7 and the latest ESP8266-code from Github.

codemasterPL, 2016-01-23:
@WereCatf , do you code on Mac OSX too?

WereCatf, 2016-01-23:
@codemasterPL No, I use Linux for that. You could try installing IDE 1.6.7 and latest code from Github on OSX and seeing if the problem goes away, and if it doesn't then you could install Linux in a virtual machine and testing there -- if it works with Linux with 1.6.7 and latest Github, but not under OSX with the same then it's a platform-specific bug in the tools/code and needs to be fixed.

codemasterPL, 2016-01-23:
What is strange i have noticed is that after booting, it kind of shows not only random values but sometimes the part of strings that are in the program already. Meaning EEPROM has part of Strings that were declared in the scope of the program.. 

[image]

igrr, 2016-01-23:
Got an idea... Perhaps esptool.py overwrites binary image header which contains flash size configuration? That would explain why CheckFlashConfig reports "Flash ide size: 524288".

igrr, 2016-01-23:
The behavior you are observing is explained as follows: when IDE compiles the sketch, it sets the flash layout (via a linker script). However in your case run-time configuration differs from compile-time one. This means that EEPROM library tries to use sectors which are out of bounds, leading to these issues.

codemasterPL, 2016-01-23:
@igrr
maybe that will help, output from the compiler:

python /Users/p/Desktop/Arduino.app/Contents/Java/hardware/tools/esp8266/pyesptool/esptool.py --baud 115200 --port /dev/cu.wch ch341 USB=>RS232 1410 write_flash 0x00000 /var/folders/qg/wd73lgzn0dq7kp38qmx9_frh0000gp/T/build1baecff4f031bf9cc1383287f4cbe0cd.tmp/One.ino.bin

Links2004, 2016-01-23:
default for esptool.py  is 512K you need additional parameters.
```--flash_size```
https://github.com/Links2004/esptool/blob/master/esptool.py#L556-L557

codemasterPL, 2016-01-23:
Thank You @Links2004 , which one should i pick knowing :

Flash real id: 001640E0
Flash real size: 4194304

and

choices = ['4m', '2m', '8m', '16m', '32m', '16m-c1', '32m-c1', '32m-c2'], default = '4m')

codemasterPL, 2016-01-23:
Good news!

Worked like a charm!

Thank you all for your input!

closed 2016-01-25 by igrr