esp8266/Arduino #1487: "Hardware" SPI Pin out?

shayo, 2016-01-23:
Hi,

I am trying to interface with a library (RC522, https://github.com/miguelbalboa/rfid).
It uses hardware SPI to communicate with a device.
Is there support for SPI with the esp8266? How do I define which pins are the MISO,MOSI,CLK and SS?

Thanks,

Shay

igrr, 2016-01-23:
Hardware SPI pins are fixed, please check the pin map picture here: http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html#digital-io
Arduino SPI library works with the 8266, except that SPI polarity is fixed.

jimibi, 2016-01-23:
You must use include spi.h, for SS, so certain pins are fixed:

SPI CLK = GPIO14
SPI MOSI = GPIO13
SPI MISO = GPIO12
All other pins can be assigned to any available GPIO:
CS
RST

shayo, 2016-01-23:
Works!

closed 2016-01-23 by shayo

muke1970, 2016-06-30:
Hi.
Thanks for your work and time and sorry about my funny english :)
I'm developing a project and I need more informations about SPI on ESP8266.
If I understood ...
There are 2 hardware SPI on ESP8266 (i'm using ESP12-F).
One is called SPI (digital pin 6-11) but is not available because it usually used to connect memory.
One is called HSPI (GPIO12-15).
Ok?
The question: 
If I need use two SPI device and connect    
SPI CLK = GPIO14
SPI MOSI = GPIO13
SPI MISO = GPIO12
CS1 = GPIO16
CS2 = GPIO4
...
Can I use HardwareSPI or must use softwareSPI?
Thanks.

chaeplin, 2016-06-30:
@muke1970 
check this https://github.com/esp8266/Arduino/issues/1429
HardwareSPI with manual CS selection(digitalWrite to LOW).

julioantolin, 2017-02-28:
Hello, good morning, I want to know if you can help me to know how I can use the SPI interface, which is connected to FLASH memory, using two free GPIO for CS and D / C. In this SPI interface I want to connect a screen with ILI9341 chip , I already have the ILI in the SPI that everyone uses with the pines:
SPI CLK = GPIO14
SPI MOSI = GPIO13
SPI MISO = GPIO12
And it works very well, but I need hares more pins. I use the ESP8266MOD with access to the pines SPI using the FLASH, I think that for some reason they were left available, being that they are all busy for FLASH control. If possible a serious good news. I use the Arduino compiler. Greetings and thank you very much.

Nookesh, 2018-04-13:
Hello good morning.I am using ESP8266-12f breakout board.i am not using the development board.My requirement is i need two uarts(2 pair of Rx and Rx) and SPI. Can i use the SPI which is connected to digital pins 6-11??

Looking for a help.Any help or suggestions highly appreciated??
Thanks

muke1970, 2018-04-16:
GPIO6 ... GPIO11 are used to connect SPI flash chip that holds your program so ... I suppose isn't a very good idea ;-)

gitcnd, 2018-11-22:
SPI supports lots of different devices on the same bus - they all ignore data other than their own (i.e. when their CS pin doesn't tell them to read it)... so why do people think it's not OK to share the flash chip SPI with other devices ?

Makuna, 2018-11-22:
@gitcnd Your program is stored in flash, it is not read once from it, it constantly being accessed as needed to pull small amounts into ram to run.  Some code is marked to keep in ram (I_RAM_CACHE) like ISRs.  Loading your code to run while your code was "holding" the same bus for its purposes could cause contention that is not recoverable.

5chufti, 2018-11-29:
there is possibility to use hwspi, but with lilitation that flash has higher priority.
see [here](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#spi) under extended mode

julioantolin, 2018-11-29:
thanks you

Enviado de mi MI MAX 2