esp8266/Arduino #1488: Deep sleep and wake up using interrupt

shayo, 2016-01-23:
Hi,

I saw the option to enter deep sleep for a predefined amount of time (  ESP.deepSleep)
Is it possible to wake up during that time with an external interrupt?
If so, how?

Thanks,

Shay

chaeplin, 2016-01-23:
check esp api in https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#esp-specific-apis
```
ESP.deepSleep(microseconds, mode) will put the chip into deep sleep. mode is one of 
WAKE_RF_DEFAULT, WAKE_RFCAL, WAKE_NO_RFCAL, WAKE_RF_DISABLED.
```
```
 (GPIO16 needs to be tied to RST to wake from deepSleep.)
```

shayo, 2016-01-23:
That doesn't really answer my question. I understand that if order to wake up from a deep sleep timer, GPIO16 needs to be wired to RST, but is it possible to wake up before that, using an external event on some GPIO line?

chaeplin, 2016-01-23:
@shayo 
No, you can't use an external event on some GPIO line.

duncan-a, 2016-01-23:
But, if GPIO16 triggers the 'wake from deep sleep' by firing a Reset, why can't a Low pulse on RST achieve the same result?

Am I missing something?

Links2004, 2016-01-23:
GPIO16 is connected to CH_DP for deepSleep.
and pulled low during deepSleep.
if CH_DP goes high the chip wakes up.
for what reason the high signal comes from does not mater.
can be the RTC timer or any other electrical stuff connected to CH_DP.
so adding a pullup to CH_DP and a switch that is normally close to GPIO16 will work fine.

chaeplin, 2016-01-24:
@duncan-a External interrupt to rst is ok. But @shayo's  question is "using an external event on some GPIO line".

@Links2004 GPIO16 is in HIGH during deepSleep.

me-no-dev, 2016-01-24:
As an oprtion, you can have the ESP in deep sleep waiting for the RTC to trigger pin 16 for wakeup and use the reset button at the same time as manual reset. Reading the reset cause can tell you which event woke up the ESP :) then you can programmatically split and act different on both.

Humancell, 2016-04-03:
@chaeplin ... thanks for the schematic and comments.

I'm curious ... when you say "flip flop" does this mean it fires the reset on both opening AND closing the switch?

chaeplin, 2016-04-03:
@Humancell  It's monostable flip flop. check here  -> http://www.bowdenshobbycircuits.info/page9.htm
 http://www.bowdenshobbycircuits.info/mono.gif
http://chaeplin.github.io/project-01-button.html

Humancell, 2016-04-04:
Thank you ... I saw that you had posted about your solution here also:  https://github.com/chaeplin/esp8266_and_arduino/tree/master/_48-door-alarm-deepsleep

I"m working on a similar idea, and am curious ... with your door sensor, it appears that you have the deep sleep timer still working, but then you can be triggered by the reed switch opening ... or closing?

And with the added circuitry (did you use the monostable flip-flop with your reed switch door sensor?) what is the final deep sleep current that you are seeing?  Do any of these components add to the current draw during deep sleep?

chaeplin, 2016-04-04:
@Humancell Trigged when opening, cause the type of reed switch is normally opened(Q3 is added to invert signal).

With C2, R4 and R9, a low signal is converted to reset pulse(Arduino reset method), so no more monostable flip-flop is needed.

My µCurrent GOLD shows 0A while in deepsleep.

Humancell, 2016-04-04:
Ok ... this makes sense.  So the reed you are using is open, and then closes with the magnet nearby?

If I need to do the reset on the reed switch both opening AND closing?  I want to create an event when the door opens ... AND when it closes ... which obviously adds complexity.  Is this where the flip-flop comes into play?

Sorry for being slow on this ...

chaeplin, 2016-04-04:
@Humancell _48-door-alarm-deepsleep is doing what you want. 
I can't make a reset signal when door is closing, So I use modem sleep and light sleep while door is opened.

Humancell, 2016-04-05:
Thank you for the wealth of links and information.  This is exactly what I was looking for.  Now I can dedicate an ATtiny85 to be a strictly "interrupt-woken" peripheral that will produce the low-pulse required to wake the ESP-12F.  I can then also use the ESP-12F deep sleep timer to wake at regular intervals to produce "heartbeats" from the sensor when the door is open or closed for extended periods.

With your suggestion on pins above, would I be passing the ESP deep sleep sugnal on pin 16 through the ATtiny85 also?  Is that what your pin #3 is above?  So teh ATtiny85 would wake on either the reed switch to ESP GPIO wake signal?

chaeplin, 2016-04-05:
@Humancell I use esp-01 for the circuit. Pin no 3 is used to present state of door to esp GPIO 1(Attiny85 control debounce). As you use ESP-12F, GPIO16 can be used to signal ATtiny85 as your suggestion.

Ukritkhonglao, 2016-04-17:
@chaeplin Hi,
Im working on a project with esp12e and want to wake it up wirelessly from deep sleep using RF transmitter. 
the RF receiver(SRX882) data pin -> pin3 ATtiny85. 
esp12e RST -> pin6 ATtiny85. 
So my understanding is that ATtiny85 will generate pulse to RST of esp12e in order to wake up from deep sleep?

chaeplin, 2016-04-17:
@Ukritkhonglao 
You don't need pwm. A short low pulse will do.
```
digitalWrite(pin5, LOW);
delay(10);
digitalWrite(pin5, HIGH);
```

fjaraujo, 2016-06-28:
Hi @chaeplin, I tried to implement your schemes (your edited version of monostable flip flop and the version in github) and in both cases didn't work. I have used 3904 instead of 4401,... could be by this? Other point is the capacitors... which is the polarity?

thanks in advance for your help

joostmarkerink, 2016-07-01:
Isn't it easier and more power efficient to let the attiny control the vcc of the esp using a transistor?

pieman64, 2016-07-29:
@chaeplin I am using D0 for something other than wake from deepsleep. D0 is set to be permanently low for this other function. Is there any way that I can still use D0 to wake from deepsleep i.e. by changing pinMode on the fly or setting D0 high or low just before it goes to sleep? I have tried several combinations but haven't been able to get D0 to be "multi purpose".

cavazorro, 2016-08-28:
Hello everybody, I'm doing a project where I have several buttons waiting to be pushed, once a button is pushed, I need to send the button number to a server using the ESP8266. I want to have everything sleeping while it is waiting for activity. I like the ATTiny85 option but don't know if I should use it for cutting power to the ESP8266 using a transistor, or if I should use it to wake it up. 
Also, if anyone has an idea of how to best wire the buttons so I can send the button number to the ESP8266, it would be greatly appreciated. There are 4 buttons in total.

dariocdj, 2016-09-22:
Hi guys, @chaeplin i've don't understand well some issues : 
1) if i use 48-door-alarm-deepsleep with your resistor and capacitor circuits, you've wrote that you "use modem sleep and light sleep while door is opened".. so you can send a signal when door is opened and closed also, right? if i leave door opened for 1 hour and than i close it, esp wake from modem or light sleep and tell me that is closed, right?  
2) What must be connected to LIGHT SLEEP IN on your circuit ??? is DOOR_INT_PIN 5 of sketch ?
3) GPIO IN is DOOR_PIN 4 of your sketch ? 
4)  what is the advantage to use attiny85? that esp goes always in deepsleep ? in that case a sketch must be loaded on attiny? where i can find it ? 
Thanks a lot for your answer and sorry for my bad english
Dario

dariocdj, 2016-09-22:
Guys, someone has implemented a working circuit to wake esp on reed open or close? with attiny or with  R/C/TR circuit ?

oriolrius, 2016-09-25:
@dariocdj Hi Dario, I'm also interested on your previous questions. Have you found any answer to them? I'm very interested on implementing R/C/TR circuit but I have the same stoppers like you.

Regards and thanks for so interesting thread guys.

Oriol

dariocdj, 2016-09-28:
@oriolrius no sorry, i'm searching for an answer yet...... please update this post if you understand the right way...

oriolrius, 2016-09-28:
@dariocdj I decided to start trying with ATTiny85 based solution because it seems better IMHO. I don't like to implement logic using passive electronics. I prefer to have a master-slave configuration based on two MCU. I don't expect having anything done before a month because I have other priorities. Anyway when I finish that I'll publish my solution on my blog page (http://oriolrius.cat).

Regards.
Oriol

avolag, 2016-11-07:
Dear all, thanks a lot for your information. I'm really new on esp programming. I have a question: To record temperature data every a known interval of time, the dsleep() function is perfect (connecting GPIO16 to the RESET). But I also need to acquire data from a rain gauge (that have a reed switch). But I don't understant how I can do it. I want use reed input as external trigger as interrupt, so when the rain activate the reed, it can reset the ESP, without wait the RTC set time. Even more obscure for me, the possibility to use RTC memory to store data. Rain data could be stored in the RTC memory and send it when the ESP wake from deepsleep. Do you have same advice?

Marmachine, 2017-05-03:
@chaeplin you inspired me, thanks for sharing your info!

I want to trigger a wake up by a PIR sensor, but only when in deep sleep.
I think the situation is quite similar to the above, although the PIR gives a (2.2V HIGH) signal instead of the magnetic switch in the door sensor.

So i came up with the idea to use a relay and i think that my below scheme could work. It's **still a draft** though, i haven't got all the parts yet to setup a test, but maybe someone already has some ideas or suggestions regarding this? I'm no expert in electronics so i could be disapointed when i put this all together!?

[image]

tbalon, 2017-07-10:
I had good success simply controlling power to the ESP8266.  My sensor used a LiPo battery with a boost/buck regulator that had an enable pin. I used the regulator enable as a power switch to the ESP01. In my case, I used a normally open reed switch connected to an ATiny85 which was in sleep mode. When the switch closed, it grounded a pin on the ATiny generating an interrupt. The ATiny then turned enabled the regulator which in turn starts up the ESP01. The ATiny85 only uses a few uA in sleep mode, the Microchip regulator only uses 2 uA in standby... I'm using a 400ma LiPo and a supercap to off load in-rush current during startup. I configured the ESP01 to use N-mode wifi to limit power to 120ma during posts.

At boot, the ESP01 sets its GPIO_0 high to indicate it's busy. It then reads the door state on its RX pin and posts json updates. Once its all done it goes into deep-sleep which causes GPIO_0 low. The ATiny detects this and shuts off power and goes back into sleep.  A watchdog on the ESP01 forces a deep-sleep after 15 seconds in case it cannot associate with the AP or the door is propped open.

[image]

devyte, 2017-10-18:
Interesting discussion, but closing per #3655 .

closed 2017-10-18 by devyte

HobbytronicsPK, 2018-02-08:
If someone is looking for waking ESP8266 from deepsleep with multiple buttons interrupt, have a look on to the following link.
[Wake ESP8266 by input signals during Deep Sleep the same way as Pin Change Interrupts](https://hobbytronics.com.pk/esp-deep-sleep-input-wakeup/)

Tank159, 2019-01-08:
by using a nodemcu board, the easiest way to wake it up during the deepSleep time, is by connecting the two contacts of the build in RST button through an intterupt. Maybe this will help

Rob58329, 2020-06-06:
The schematics shown above in this post seem to be problematic for the Wemos D1mini ESP8266, because the D1mini has an inbuilt 100nF-Capacitor to ground and 10k-Pullup-Resistor on its RST pin.  (IE. any reset circuit needs to both discharge this 100nF capacitor, and be capable of sinking both this discharge current, the current from the 10k-pullup, and another several mA of current from the RST pin itself, and still generate a reset pulse which drops to approx 0.8V).

The below circuit achieves this by using the STM1061 [2.2V trigger] (a voltage monitor with an open-drain output) to isolate the RST-pin/100nF-inbuilt-capacitor from the RC timing circuit.  (I use this circuit to monitor the interrupt output from the TLS2561 and TLS2591 light sensors, which have an open-drain output with is active-low .)

[image]

The STM1061 has an SOT23-3 package, but is relatively easy to mount on an SOT23-3 adapter for breadboard or stripboard use.  The RED section in the above schematic (the Schottky-Diode) is optional (ie. only needed if you want ESP8266 to be woken from DeepSleep by internal timer).  GPIO16 is OpenCircuit during normal ESP8266 operation, but goes HIGH during DeepSleep (biasing the 2N2222 which thereby enables the interrupt circuit).  The Schottky-Diode is necessary so the STM1061 can still pull the RST-pin low even when GPIO16 is HIGH.  The 360R resistor limits the peak discharge current to approx 9mA.

The Reset-Pulses generated by this circuit with a D1mini are shown below.  Note that even though the pulse from GPIO16 only drops to 1.0V, this still seems to work fine on all my D1mini units.

[image]

[image]