esp8266/Arduino #1489: Unexpected behaviour with analogWriteFreq and analogWrite

paccerdk, 2016-01-24:
Using analogWriteFreq together with certain analogWrite duty cycles yields unexpected results:

For instance, calling:
`analogWriteFreq(20000);`
`analogWrite(pin, 1);`
produces a duty cycle of nearly 100%.

Calling:
`analogWriteFreq(12000);`
`analogWrite(pin, 1022);` or `analogWrite(pin, 1021);`
produces a duty cycle of nearly 0%.

Double checked with logic analyzer

Using latest git version, cloned yesterday (Jan. 23)

me-no-dev, 2016-01-24:
those freqs are a bit hight for what the software PWM can do. If you want 10KHz freq, your range should be around 256 (8 bit) if you want any success.

devyte, 2017-10-18:
No MCVE sketch, no basic info, no IDE settings.
I think the pwm has been rewritten in the interim. In any case, the pwm is still software-based, so there are inherent limitations.
Closing. If there is still a problem, please open a new issue and fill in the requested info.

closed 2017-10-18 by devyte