esp8266/Arduino #1490: Holding GPIO16 low seems to cause some sort of reset

rogerclarkmelbourne, 2016-01-24:
I'm trying to use GPIO16 as a input, where I have pulled GPIO16 high through a resistor, and its pulled low though a push button.

(I'm using an ESP-12E)

If the button is held in for about 1 second, the ESP seems to go into some sort of reset

and outputs this on serial
```
Exception (0):
epc1=0x40106405 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff2f60 end: 3fff3130 offset: 01a0

>>>stack>>>
3fff3100:  3fffdc20 00000000 3fff3184 402084ae  
3fff3110:  3fffdc20 00000000 3fff20f4 4020b162  
3fff3120:  00000000 00000000 3fff2110 40100114  
<<<stack<<<
```
Then seems to reboot

I'm just using this really cut down example program that does nothing but set the Serial and pinMode and the start counting up and displaying the count (via Serial)

```
#define BUTTON_PIN 16
void setup(void)
{
  Serial.begin(9600);

  pinMode(BUTTON_PIN,INPUT);  // Set button as input 
  
  int count=0;
  while(1)
  {
  delay(500);
  Serial.println(count++);
  }
}
```

I've not seen anything in the docs that suggest I can't use GPIO16 as a normal input.

I've checked my circuit and GPIO16 is not connected to anything else

So I don't really understand why this is happening

BTW. If I briefly press the button i.e less about 1 second, the button can be read using digitalRead 

It just seems to be invoking some other function if its held in

Is there some feature e.g. sleep mode that I need to disable to use GPIO 16 ?

PS. I'm not using the very latest version as it doesnt seem to be compatible with my IDE version (1.6.5r5), but its only about 2 or 3 months old.

me-no-dev, 2016-01-24:
code looks fine. GPIO16 should by all means behave normally (it's slightly slower than the other GPIOs, but much faster than Arduino). I'm using the pin all of the time and have seen no such issues.

rogerclarkmelbourne, 2016-01-24:
Umm

Very odd.

I'll see if I can reproduce the same fault on a totally different ESP board

At the moment I am using an ESP-12E soldered to a custom made PCB, that has the button and some LED's etc i.e custom breakout board.

I've used a continuity checker to make sure I've not solder bridged to another pin etc, 

Its strange that this problem only seems to occur if I hold the button in. i.e if I press and release quickly its fine.

But I'm glad to know that my design was technically correct, even if it doesn't work ;-)

supersjimmie, 2016-01-24:
What is the value of the resistor (and have you verified it)? can't it be that the combination of the pull-up resistor and the pull-down switch take just too much current? (thus causing an instable system)

me-no-dev, 2016-01-24:
I tried both on my nodemcu and my bare modules and can not see the behavior you explain, so must be something power related or as @supersjimmie says

rogerclarkmelbourne, 2016-01-25:
Thanks guys

I've tried running the board from a USB PSU and even from a bench PSU, I also put a 470uF capacitor rightr across the ESP-12E's power lines, but I'm still getting this issue.

I checked my pullup on GPIO16 and its 10K, I also monitored the current when I pushed the button with both an analog and a digital meter, and neither saw any change when I pushed the button.

I have a 10k pullup on Reset, and another 10k pullup on CH_PD
I had to pull GPIO15 low through 10k, as I want to use GPIO15, as its one of the alternative pins for the Serial UART. I pull up GPIO0 via 10k (PROGRAM) as well.

I guess this is some sort of power related issue, as I get fairly random results when I press the button.
Sometimes it seems to go into Program mode and sometimes it just resets

I tried putting 1k in series with the button, and it then works fine.

I measured the resistance between GPIO16 and reset and it seems to be around 15k, but that could be irrelevant, as I measured the voltage on the reset pin and it didn't seem to change at all when I connect GPIO16 to GND.

Anyway, when I get time I'll check it on one of my other ESP modules that is not a 12E (but which has GPIO16 broken out)

rogerclarkmelbourne, 2016-01-25:
Guys

I can only replicate this issue on one module. So possibly it has some sort of fault.

So I'll close this issue

Thanks again for your help

closed 2016-01-25 by rogerclarkmelbourne

goetzFromKoblenz, 2019-01-01:
I know, that this is closed for quite a while.
At least I'd like to tell you that I came across exactly the same problem and what it turned out. 
I used a NodeMCU V3 board, had the same resets when taking GPIO16 to low and I also measured 15k between GPIO16 and reset. 
For comparison I took a brand new ESP-12E module and measured between the two pins and there was no connection between them (several MOhms). 
So I tried to find out if there was some kind of a bridging resistor on the NodeMCU board. Taking a very close look at the ESP module soldered onto the board I discovered a hair-fine solder bridge between GPIO16 and CH_PD at the edge of its PCB. I scraped it away and all my problems disappeared.
As the whole board isn't soldered too well, I guess it's a problem of the production process...

Even if I had to solve it myself, this thread helped me a lot!