esp8266/Arduino #1491: Connection lost after few seconds. Using WS and JSON library

qblacksheep, 2016-01-24:
Hi, i have problem with my program. My sensor need to send data by WebSockets, but after about minute its send empty json: {} and after this WS session its disconnect and can't reconnect.
Here is terminal ss:
[image]
 
and after few sec:
[image]

Do u know why? I am using library https://github.com/Links2004/arduinoWebSockets
Here is my code when i am sending data:
```javascript
void dataSend(String light, String voltage)
{
  
  JsonArray& jsonArray = jsonBuffer.createArray();
  JsonObject& root = jsonBuffer.createObject();
  JsonObject& lightJson = signalJson("int_0", "light", light);
  JsonObject& voltageJson = signalJson("int_1", "voltage", voltage);
  jsonArray.add(lightJson);
  jsonArray.add(voltageJson);
  root["signals"] = jsonArray;
  root["action"] = "data";
 
  char buffer[256];
 
  size_t size = root.printTo(buffer, sizeof(buffer));
  if(registrationFlag)
  {
 
  webSocket.sendTXT(buffer, size);
for( int i = 0; i < sizeof(buffer);  ++i )
{
   buffer[i] = (char)0;
}
 
  delay(100);
  }
```

hallard, 2016-01-24:
@qblacksheep 
As far as I remember using JSON library (if you use this one https://github.com/bblanchon/ArduinoJson)) SHOULD use a static JSON buffer declared like this :

```
StaticJsonBuffer<200> jsonBuffer;
```
Are you using like that ?

Links2004, 2016-01-25:
is the same happening when you simple put a Static string as response (not build dynamic with arduinoJSON)?

qblacksheep, 2016-01-26:
Its only happening when i am using JSON library. I chagne my buffer to static but now i can't change value of atributes.

These functions worked propety when i used dynamicJsonBuffer. Now when i change to static, its doesnt working.
``` javascript
StaticJsonBuffer<200> jsonBufferRegister;
String registration = 
 "{\"stored_id\":\"undefined\"\,\"action\":\"register\",\"device\":{\"name\":\"sensor\",\"interfaces\":[{\"direction\":\"output\",\"data_type\":\"light\",\"id\":\"int_0\"},{\"direction\":\"output\",\"data_type\":\"voltage\",\"id\":\"int_1\"}]}}";
 
void updateDeviceID(const char* newID) {
  
  JsonObject& json = jsonBufferRegister.parseObject(registration);
  json["stored_id"] = newID;
  delay(200);
  char buffer[256];
  json.printTo(buffer, sizeof(buffer));
  delay(200);
  registration = String(buffer);
  
}

JsonObject& signalJson(String interfaceID, String dataType, String value) {
  
 StaticJsonBuffer<200>  jsonBufferSignal;

 JsonObject& signal = jsonBufferSignal.createObject();
 JsonObject& message = jsonBufferSignal.createObject();
 
 message["data_type"] = dataType;
 message["value"] = value;
 signal["interface_id"] = interfaceID;
 signal["message"] = message;
 
 return signal;
 jsonBufferSignal.clear();
}

void dataSend(String light, String voltage)
{
  StaticJsonBuffer<200> jsonBufferSend;
  
  JsonArray& jsonArray = jsonBufferSend.createArray();
  JsonObject& root = jsonBufferSend.createObject();
  JsonObject& lightJson = signalJson("int_0", "light", light);
  JsonObject& voltageJson = signalJson("int_1", "voltage", voltage);
  jsonArray.add(lightJson);
  jsonArray.add(voltageJson);
  root["signals"] = jsonArray;
  root["action"] = "data";
 
  char buffer[256];
 
  size_t size = root.printTo(buffer, sizeof(buffer));
  if(registrationFlag)
  {
 
  webSocket.sendTXT(buffer, size);
for( int i = 0; i < sizeof(buffer);  ++i )
{
   buffer[i] = (char)0;
}
  delay(100);
  }
;
```
As you see i want change my device_id and type, value and interface id in sending json. With staticJsonBuffer its doesnt working, in dynamic i get what i mentioned in the first post.

hallard, 2016-01-26:
@qblacksheep 
If you don't do a intensive usage of JSON like it seems because you're sending using websocket, may be you can try to construct your JSON string yourself. It's not so smart but pretty efficient

This is what I'm doing, not my favorite, but works ;-)
```arduino
void getSensorsJSONData(String & response)
{
  char buffer[32];
  bool first_item = true;

  // Json start
  response += FPSTR(FP_JSON_START);

  // SI7021 Array  
  response += F("\"si7021\":[\r\n");
  if (config.sensors.en_si7021) {
    response += F("{\"temperature\":"); 
    response += String(si7021_temperature/100.0);
    response += F(",\r\n\"humidity\":"); 
    response += String(si7021_humidity/100.0);
    response += F(",\r\n\"seen\":"); 
    response += si7021_last_seen;
    response += F("}\r\n");
  }
  response += F("],\r\n");

  // SHT10 Array  
  response += F("\"sht10\":[\r\n");
  if (config.sensors.en_sht10 && (config.config & CFG_SHT10) ) {
    response += F("{\"temperature\":"); 
    response += String(sht1x_temperature/100.0);
    response += F(",\r\n\"humidity\":"); 
    response += String(sht1x_humidity/100.0);
    response += F(",\r\n\"seen\":"); 
    response += sht1x_last_seen;
    response += F("}\r\n");
  }
  response += F("]\r\n");

  // Json end
  response += FPSTR(FP_JSON_END);
}
```

andig, 2016-01-31:
@hallard if you do this mass of concats I'd recommend also `reserve()`ing the string upfront.

devyte, 2017-10-18:
From the code snippets, it looks to me like the jsonBuffer is global, which means it is being reused, which is wrong.
Closing as usage error.

closed 2017-10-18 by devyte