esp8266/Arduino #1494: eraseConfig() does not seems to clear SDK Wifi Credentials

hallard, 2016-01-24:
Does the `ESP.eraseConfig()` should also erase SDK Wifi Credentials ? 
I've 2 button on my Web Interface to do 
* Reset (just reset the board)
* Factory Reset to clear all and reset.

Factory Reset restore my default board config settings including clearing my client SSID and PSK, and then call a `ESP.eraseConfig()`to also clear SDK config. Then issue a normal reset.

Unfortunately, just after Factory Reset, my board connect to my Old Access point even if I don't call Wifi setting and connection (because my config is empty on those settings, yeah cleared that!!)
Anyone tried this with success ?

igrr, 2016-01-26:
I think if you do `ESP.restart()` after `ESP.eraseConfig()`, SDK libraries will actually restore the configuration from RAM before restarting. I'm not sure this is documented, but I saw this while going through disassembly of some older version of the SDK (1.0.0-like).
You could try to crash instead of calling `restart`, i.e. do something like `*((int*) 0) = 0;`. It's not a clean solution, though. I would prefer if the SDK didn't do something under the hood unless instructed to do so explicitly, but well... I guess it's a task for another time.

hallard, 2016-01-27:
@igrr 
you're absolutely right, It did the trick ;-)
I just replaced 
```
*((int*) 0) = 0;
```
by a watchdog reset, it works but is it correct ?
```
// Fire Watchdog Reset
while(1);
```
thanks

Links2004, 2016-01-27:
```cpp
ESP.reset();
```
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L116

hallard, 2016-01-27:
@didn't tried that one, it will not restore config like restart() ?

Links2004, 2016-01-27:
yes (last tried on SDK 1.3),
its simple a other way too reset the ESP.
more clean then WDT or triggering a exception.

closed 2016-02-03 by igrr

barsznica, 2018-01-02:
WiFi.disconect() worked for me first go.

ESP.eraseConfig then ESP.reset() OR ESP.eraseConfig() then ESP.restart() did not work for me.
I did not try the *((int*) 0) = 0; thing.

The answer to this question was covered [here](https://github.com/tzapu/WiFiManager/issues/142) and [here](http://www.esp8266.com/viewtopic.php?f=32&t=8204&start=4).