esp8266/Arduino #1498: Template issues?!

tuxedo0801, 2016-01-25:
Hi there,

I'm not a realy C++ expert, but trying to port a arduino library to esp8266/Arduino...

I thought doing a #ifdef here and there would do it. But now I'm struggling with this issue:

When compiling, I get the following error:

```
...../KnxDevice.cpp:236:106: error: duplicate explicit instantiation of 'e_KnxDeviceStatus KnxDevice::read(byte, T&) [with T = unsigned char; byte = unsigned char]' [-fpermissive]
 template e_KnxDeviceStatus KnxDevice::read <unsigned char>(byte objectIndex, unsigned char& returnedValue);
                                                                                                          ^
...../KnxDevice.cpp:283:98: error: duplicate explicit instantiation of 'e_KnxDeviceStatus KnxDevice::write(byte, T) [with T = unsigned char; byte = unsigned char]' [-fpermissive]
 template e_KnxDeviceStatus KnxDevice::write <unsigned char>(byte objectIndex, unsigned char value);
```

And that's the corresponding KnxDevice.cpp (line 235-243):

```c++
template e_KnxDeviceStatus KnxDevice::read <boolean>(byte objectIndex, boolean& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <unsigned char>(byte objectIndex, unsigned char& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <char>(byte objectIndex, char& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <unsigned int>(byte objectIndex, unsigned int& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <int>(byte objectIndex, int& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <unsigned long>(byte objectIndex, unsigned long& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <long>(byte objectIndex, long& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <float>(byte objectIndex, float& returnedValue);
template e_KnxDeviceStatus KnxDevice::read <double>(byte objectIndex, double& returnedValue);
```
and the related snippet from the header file:
```c++
    template <typename T>  e_KnxDeviceStatus read(byte objectIndex, T& returnedValue);
```

Of course it's a kind of duplicate, but with a different type flavor.
With arduino board the code compiles fine.

Any hints on this?

tuxedo0801, 2016-01-25:
Is the "template" feature missing with the compiler the esp8266/Arduino is using?

tuxedo0801, 2016-01-26:
Come on... Anyone out there working with templates?

igrr, 2016-01-26:
A bunch of ESP8266 libraries use templates, like ESP8266WebServer and EEPROM, so I don't think there is a constraint from the compiler end. Perhaps you could point to the complete code you are trying to port?

Links2004, 2016-01-26:
template are working on the esp8266
example:
https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/libraries/EEPROM/EEPROM.h#L41-L58

tuxedo0801, 2016-01-26:
Thanks for feedback.

The code I'm trying to port:

https://github.com/KONNEKTING/KonnektingDeviceLibrary

As I said: I'm not a real c++ expert. Would be great if someone could give me a hint on what's wrong?! (even if this is not a topic for this issue-tracker)

thanks a lot,

   Alex

tuxedo0801, 2016-01-30:
Had a bit time to further investigate...

If I testwise remove the templates which use "unsigned char", I no longer get the duplicate-error. But:

> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp.o:(.text._Z9knxEventsh+0x10): undefined reference to `e_KnxDeviceStatus KnxDevice::write<bool>(unsigned char, bool)'
> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp.o: In function `knxEvents(unsigned char)':
> /media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary/examples/KnxDevice_PushButton_with_Programming/KnxDevice_PushButton_with_Programming.ino:35: undefined reference to `e_KnxDeviceStatus KnxDevice::write<bool>(unsigned char, bool)'
> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x14): undefined reference to `__cxa_guard_acquire'
> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x18): undefined reference to `__cxa_guard_release'
> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x37): undefined reference to `__cxa_guard_acquire'
> /tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o: In function `KnxTpUart::RXTask()':
> /media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp:515: undefined reference to `__cxa_guard_release'
> collect2: error: ld returned 1 exit status

This tells me: the template with the unsigned char type is missing ?!

So, it has to do - somehow - with the unsigned char template.

Any ideas?

best regards,
Alex

Links2004, 2016-01-30:
undefined reference to __cxa_guard_release is the problem here.
the problem is tracked here: #500

tuxedo0801, 2016-01-30:
Of course it's undefined, I commented out the template with the "unsigned char" type...

The actual problem is, when I use the template with type "unsigned char", I get duplicate erors... See above.

tuxedo0801, 2016-01-30:
Steps to reproduce:

Checkout https://github.com/KONNEKTING/KonnektingDeviceLibrary
Load sketch https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/examples/KnxDevice_PushButton_with_Programming/KnxDevice_PushButton_with_Programming.ino

try to compile for:

"Generic ESP8266 Module, Serial, 80Mhz, QIO, 115200, 512K (64KSPIFFS), ck, Disable, None"

You will get error:

```
/home/foobar/arduino-1.6.7/arduino-builder -dump-prefs -logger=machine -hardware "/home/foobar/arduino-1.6.7/hardware" -hardware "/home/foobar/.arduino15/packages" -tools "/home/foobar/arduino-1.6.7/tools-builder" -tools "/home/foobar/arduino-1.6.7/hardware/tools/avr" -tools "/home/foobar/.arduino15/packages" -built-in-libraries "/home/foobar/arduino-1.6.7/libraries" -libraries "/media/bigdisk/Programming/Arduino/Sketchbook/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=qio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10607 -build-path "/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary/examples/KnxDevice_PushButton_with_Programming/KnxDevice_PushButton_with_Programming.ino"
/home/foobar/arduino-1.6.7/arduino-builder -compile -logger=machine -hardware "/home/foobar/arduino-1.6.7/hardware" -hardware "/home/foobar/.arduino15/packages" -tools "/home/foobar/arduino-1.6.7/tools-builder" -tools "/home/foobar/arduino-1.6.7/hardware/tools/avr" -tools "/home/foobar/.arduino15/packages" -built-in-libraries "/home/foobar/arduino-1.6.7/libraries" -libraries "/media/bigdisk/Programming/Arduino/Sketchbook/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=qio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10607 -build-path "/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary/examples/KnxDevice_PushButton_with_Programming/KnxDevice_PushButton_with_Programming.ino"
"/home/foobar/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266" "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/variants/generic" "/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp" -o "/dev/null"
"/home/foobar/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266" "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/variants/generic" "-I/media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary" "/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp" -o "/dev/null"
"/home/foobar/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266" "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/variants/generic" "-I/media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary" "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/EEPROM" "/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp" -o "/dev/null"
"/home/foobar/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10607 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/cores/esp8266" "-I/home/ [... cut ...]
                                                                                                  ^
Bibliothek KonnektingDeviceLibrary im Ordner: /media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary (legacy) wird verwendet
Bibliothek EEPROM in Version 1.0 im Ordner: /home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/EEPROM  wird verwendet
Bibliothek SoftwareSerial in Version 1.0 im Ordner: /home/foobar/.arduino15/packages/esp8266/hardware/esp8266/2.1.0-rc2/libraries/SoftwareSerial  wird verwendet
exit status 1
Fehler beim Kompilieren.
```

tuxedo0801, 2016-01-30:
@Links2004 
Your're right, must have to do with the linked issue. Will check this...

tuxedo0801, 2016-01-30:
If I understood the linked issue correctly, one should not declare static variables and initialize them by a methode call at once.

Problem is: I don't find such a "declare static and initialize by method call" constellation in my code.

My code makes use of a static class instance to access the API of the library.

I try to explain with some code snippets:

From the header (of the library I try to port to ESP8266):

The static instance of the main class of the library:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/KnxDevice.h#L134
```c++
public:
    static KnxDevice Knx; // unique KnxDevice instance (singleton design pattern)
```

The constructor itself is private, singleton-pattern... just use the static instance.
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/KnxDevice.h#L128
```c++
  // Constructor, Destructor
    KnxDevice();  // private constructor (singleton design pattern)
    ~KnxDevice() {}  // private destructor (singleton design pattern)
    KnxDevice (const KnxDevice&); // private copy constructor (singleton design pattern) 
```

The user can then use the singleton to access the library:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/examples/KnxDevice_PushButton_with_Programming/KnxDevice_PushButton_with_Programming.ino#L32
```c++
Knx.write(2, true );
```

The above shown "write" command is one of the templated-methods:
https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/master/KnxDevice.cpp#L297
```c++
template e_KnxDeviceStatus KnxDevice::write <boolean>(byte objectIndex, boolean value);
template e_KnxDeviceStatus KnxDevice::write <unsigned char>(byte objectIndex, unsigned char value);
template e_KnxDeviceStatus KnxDevice::write <char>(byte objectIndex, char value);
template e_KnxDeviceStatus KnxDevice::write <unsigned int>(byte objectIndex, unsigned int value);
template e_KnxDeviceStatus KnxDevice::write <int>(byte objectIndex, int value);
template e_KnxDeviceStatus KnxDevice::write <unsigned long>(byte objectIndex, unsigned long value);
template e_KnxDeviceStatus KnxDevice::write <long>(byte objectIndex, long value);
template e_KnxDeviceStatus KnxDevice::write <float>(byte objectIndex, float value);
template e_KnxDeviceStatus KnxDevice::write <double>(byte objectIndex, double value);
```

So, I don't see the same case explained in issue #500, but according to the latest build error (fixed some other smaller things, the duplicate is now gone), it's somehow related to this issue:

```
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp.o:(.text._Z9knxEventsh+0x10): undefined reference to `e_KnxDeviceStatus KnxDevice::write<bool>(unsigned char, bool)'
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/sketch/KnxDevice_PushButton_with_Programming.ino.cpp.o: In function `knxEvents(unsigned char)':
/tmp/arduino_bb4fcaf6d0d53941d44b612db42cc988/KnxDevice_PushButton_with_Programming.ino:35: undefined reference to `e_KnxDeviceStatus KnxDevice::write<bool>(unsigned char, bool)'
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x14): undefined reference to `__cxa_guard_acquire'
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x18): undefined reference to `__cxa_guard_release'
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o:(.text._ZN9KnxTpUart6RXTaskEv+0x37): undefined reference to `__cxa_guard_acquire'
/tmp/buildbb4fcaf6d0d53941d44b612db42cc988.tmp/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp.o: In function `KnxTpUart::RXTask()':
/media/bigdisk/Programming/Arduino/Sketchbook/libraries/KonnektingDeviceLibrary/KnxTpUart.cpp:515: undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status
```

So I'm still lost :-(

tuxedo0801, 2016-02-08:
Both issues fixed:

the guard issue is fixed by a PR and the duplicate error is fixed by changing one template from type "boolean" to "bool".

closed 2016-02-08 by tuxedo0801

Merdeka, 2016-02-23:
tuxedo0801 can you explain how you fixed this?

I replaced all the boolean to bool in KnxDevice.cpp and KnxDevice.h.
That resolved the template issue.

But i don't really understand the "the guard issue is fixed by a PR" part.

tuxedo0801, 2016-02-24:
The actual issue was already discussed here: https://github.com/esp8266/Arduino/issues/500

And @iggr fixed this about 18 days ago: https://github.com/esp8266/Arduino/commit/c62858fd42291e7a5b6f911bd05209112983d3bb

But there no release that contains this bugfix. I guess it will be available soon. 

https://github.com/esp8266/Arduino/releases

Look forward for release 2.1.0-RC3 or final 2.1.0 ...