esp8266/Arduino #1502: Library no longer compiles

mares0909, 2016-01-25:
I have been using my NodeMCU 1.0 with the ESP8266 Arduino library with no problem. I have recently updated the library and it stopped working: it won't compile. I have tried reinstalling all my libraries, everything to the latest but I can't figure out why I get the below error message:

```
/Users/matemarschalko/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress)':
/Users/matemarschalko/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp:218:45: error: 'wifi_softap_set_dhcps_lease_time' was not declared in this scope
     if(!wifi_softap_set_dhcps_lease_time(720)) {
                                             ^
Error compiling.
```

mares0909, 2016-01-26:
This has been resolved by deleting my ESP8266 library from the libraries folder and reinstalling the board with the board manager and using the library files that come with it by default.

closed 2016-01-26 by mares0909

tailwheel, 2016-01-26:
Of course, the problem with the newer library still exists.

milose, 2016-02-17:
I can't get this to work with platformio.