esp8266/Arduino #1505: Mini Router WN523N2 shocked ESP8266

CombiesGit, 2016-01-26:
I noticed it at my OTA experiments. Once mDNS is used, the ESP throws an exception(0). It does this exactly 1 times per Minute. It makes no difference whether the WN523N2 is running in bridge or repeater mode. The error only occurs with mDNS. Web servers, etc. are stable

A new router firmware I have not found.
All other devices in the WLAN working well.

CombiesGit, 2016-01-28:
Logs and Source files attached

[Log_with_repeater.txt](https://github.com/esp8266/Arduino/files/108182/Log_with_repeater.txt)
[Log_without_repeater.txt](https://github.com/esp8266/Arduino/files/108183/Log_without_repeater.txt)

[OTA03.ino.txt](https://github.com/esp8266/Arduino/files/108184/OTA03.ino.txt)

CombiesGit, 2016-01-28:
The Bug is marked with  <<<< --------------- 
An ill Statement.

How can I fix this?

[obj_dump.txt](https://github.com/esp8266/Arduino/files/108784/obj_dump.txt)

Links2004, 2016-01-28:
the address is in the SDK part, only espressif can fix it.

CombiesGit, 2016-02-01:
Espressif has an Answer.
See http://bbs.espressif.com/viewtopic.php?f=66&t=1701

A simple exchange of the sdk throws an error:
```

In file included from E:\Programme\Arduino\hardware\esp8266com\esp8266\cores\esp8266/esp8266_peri.h:24:0,

                 from E:\Programme\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:38,

                 from E:\Temp\build8d1efe75cc663826dc7dfe795abebf41.tmp\sketch\OTA03.ino.cpp:1:

E:\Programme\Arduino\hardware\esp8266com\esp8266/tools/sdk/include/c_types.h:15:29: error: conflicting declaration 'typedef long unsigned int uint32_t'

 typedef unsigned long       uint32_t;
```

igrr, 2016-02-01:
SDK update is a bit more involved than replacing files. I was planning to update to 1.5.2, hold on a bit.

CombiesGit, 2016-04-03:
Hi
THX for the update to  1.5.2
the Problem persist.

igrr, 2016-05-07:
Should be fixed in git version. If you're still having issues, please follow up in #1826.

closed 2016-05-07 by igrr

CombiesGit, 2016-05-07:
> If you're still having issues
Yes!
No Connection over the Repeater/Bridge.
No Problem, without Repeater
All other (smartphone, Tablet, PC) are working fine

with repeater:
> Booting
> Connection Failed! Rebooting...

without repeater:

> Booting
> Ready
> IP address: 192.168.2.104
> Start!

code fragment:
```
void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
//.........
}

```

igrr, 2016-05-07:
The issue was about a crash related to IGMP messages. 
Connection failure is likely something different, do you agree?
Please open another issue, and provide steps to reproduce this (i.e. how you configure your AP, your repeater, etc.)