esp8266/Arduino #1507: Exception (28) in add_hmac_digest

MarcusF2015, 2016-01-26:
I modified the HTTPS example and got an exception:

0
connecting to 149.154.167.198

Exception (28):
epc1=0x40216a22 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont 
sp: 3fff11d0 end: 3fff1520 offset: 01a0

>>>stack>>>
3fff1370:  3fff445d 00000000 3fff2aa8 402169dc  
3fff1380:  f2fedcf8 0187f7fa 8c83b36b 3fff2798  
3fff1390:  00000002 0000000f 3fff13d0 3fff445f  
3fff13a0:  3fff445d 3fff445d 3fff2b88 00001240  
3fff13b0:  000012a0 3fff56df 0000001d 00000000  
3fff13c0:  3fff2aa8 00080000 00000003 402178cd  
3fff13d0:  00020315 00000002 3fff2aa8 402182da  
3fff13e0:  00000002 00000015 00000033 00000000  
3fff13f0:  3fff2aa8 00000000 00000038 00000000  
3fff1400:  00000000 00080000 3fff2aa8 40217ed4  
3fff1410:  00000001 00000016 3fff2aa8 00000000  
3fff1420:  00000000 3fff2aa8 3fff2aa8 4021840e  
3fff1430:  00000000 3fff445d 3fff2aa8 3fff0408  
3fff1440:  3fff040c 00000000 3fff2aa8 40217f32  
3fff1450:  3fff27c0 40203436 3fff27d0 40202eea  
3fff1460:  3fff040c 3fff14b0 3fff27d0 4020319c  
3fff1470:  c6a79a95 3fff156c 3fff156c 3fff03ac  
3fff1480:  3fff03b4 3fff156c 3fff14b0 40203382  
3fff1490:  3ffe9558 c6a79a95 3fff03b4 402044d9  
3fff14a0:  3fff03b4 3fff156c 3fff03b4 40202318  
3fff14b0:  3ffe9358 00000000 00000064 0003c1dc  
3fff14c0:  00000000 3fff2798 3fff27d0 3ffe9558  
3fff14d0:  c6a79a95 00000003 3fff2b70 0000001f  
3fff14e0:  00000000 00000000 00000016 4010174d  
3fff14f0:  40203515 00000000 00000000 3fff04f8  
3fff1500:  3fffdc20 00000000 3fff04f0 4020353d  
3fff1510:  00000000 00000000 3fff0500 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

[Uploading dump.txt…]()
[Uploading HTTPS_Bot.ino.txt…]()

DedeHai, 2016-01-27:
I also very often get this exception under heavy sending load. Normal wifi client, not SSL.
Seems to be a bug in the espressif core from what I read in other forums.

devyte, 2017-10-18:
@MarcusF2015 Is this issue still valid with latest git?

MarcusF2015, 2017-10-18:
With 2.4.0-rc1 the device is running 7/24 for months without an issue. But I changed the Modul from 12E to 12F and the WiFi connection is stronger. I did not try rc2 yet.

devyte, 2017-10-18:
Closing as can't reproduce.

closed 2017-10-18 by devyte

igrr, 2017-10-20:
It can still run out of memory in axTLS, but the specific bug about
add_hmac_digest has been fixed.