esp8266/Arduino #1511: PWM output change only from 0V to 1,5V to 3,3V

qblacksheep, 2016-01-27:
HI, i want to control LEDS with my ESP. I use analogWrite function but the output voltage change only from 0 to 1,5 and 3,3. How i can change it often and more smoothly? i want to control brithness of my led strip. I connect output from ESP to gate MOSFET. But the LEDS shine decidedly to weak.(even if i have 1023 value from PWM -  3,3V at the output).

There is my code:
```javascript
 configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
 
  pinMode(13,OUTPUT);
  pinMode(12,OUTPUT);
 pinMode(14,OUTPUT);
 pinMode(A0, INPUT);    // USE_SERIAL.begin(921600);
    USE_SERIAL.begin(115200);
    delay(1000);

    //Serial.setDebugOutput(true);
    USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

      for(uint8_t t = 4; t > 0; t--) {
          USE_SERIAL.printf("[SETUP] BOOT WAIT %d...\n", t);
          USE_SERIAL.flush();
          delay(1000);
      }

    WiFiMulti.addAP("test", "prezentacja");
    delay(500);

    //WiFi.disconnect();
    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

   
    //webSocket.onEvent(webSocketEvent);

}

void loop() {
//webSocket.loop();
  String dane;
  dane = String(analogRead(A0));
 analogWrite(14, dane.toInt());
Serial.println("daneInT");
Serial.print(dane.toInt());

//webSocket.loop();
}
```

me-no-dev, 2016-01-27:
try putting delay(1000) in the loop. you call analogWrite more often than the change occurs

domonetic, 2016-01-28:
I think you should try with delay(), to see the changes of the PWM.
What is the maximum value returned from A0?

here speak of the analog values http://www.esp8266.com/viewtopic.php?f=28&t=3087#p17663

devyte, 2017-10-18:
I tested with latest git on a WittyCloud, and using analogWrite() I can vary the LEDs throughout the range.
Closing as can't reproduce.

closed 2017-10-18 by devyte