esp8266/Arduino #1513: std::unordered_map

tomkcook, 2016-01-27:
Similar to #1136, attempting to use `std::unordered_map` gives the following linker errors:

```
sketch/Test.ino.cpp.o:(.text.setup+0x8): undefined reference to `std::__detail::_Prime_rehash_policy::_M_next_bkt(unsigned int) const'
sketch/Test.ino.cpp.o: In function `setup':
/home/tkcook/Arduino/Test/Test.ino:3: undefined reference to `std::__detail::_Prime_rehash_policy::_M_next_bkt(unsigned int) const'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling.
```

this while compiling the following code:

```
#include <unordered_map>
void setup() {
  std::unordered_map<int,int> map;
}
void loop() {
}
```

I guess the relevant bit of libstdc++ is not being linked in.  Is there some (straightforward) way to get it linked in?

liquidfalcon, 2016-01-27:
I don't believe libstdc++ is linked in at all, at least in the standard setup. Add "-lstdc++" to the end of "compiler.c.elf.libs=" in platform.txt and it should link.

igrr, 2016-02-03:
@tomkcook did adding `-lstdc++` resolve this issue for you?

GalloTeo, 2016-06-06:
not for me

devyte, 2017-10-18:
I can't reproduce this with latest git, build works fine. Closing.

closed 2017-10-18 by devyte