esp8266/Arduino #1514: Board generic (platform esp8266, package esp8266) is unknown

kapilpawar, 2016-01-27:
i am using ARDUINO UNO and ESP8266 V1.

my board setting like this.

board : generic esp8266 module.
programmer : AVRISP mkII

while i upload my code then i get the error.

Board generic (platform esp8266, package esp8266) is unknown
Error compiling.

please help me 
thanks in advance

EduardUA, 2016-01-27:
I had this error when I used a version Arduino IDE 1.6.7. In version 1.6.5 all good.

Links2004, 2016-01-27:
may the same as #1377, #1387 
try to delete the 2.0.0 and rc2 folder, and then install the rc2 new.

https://github.com/esp8266/Arduino#installing-with-boards-manager

tuxedo0801, 2016-02-02:
HaD the same error. Intuitively did the same as Links2004 suggested. Works for me.

closed 2016-02-03 by igrr

SunboX, 2016-04-15:
On Mac OS X, the folder is located at
`/Users/$USER/Library/Arduino15/packages/esp8266/hardware`

pspranav7, 2016-06-12:
Check out step 5 of this [Instructable](http://www.instructables.com/id/Programming-Esp8266-12e-With-Arduino-IDE/) ...
It will solve this problem.... :)

kevin192291, 2017-07-29:
Link is Broken :( what was it?

borhanreo, 2017-10-24:
I was face same problem. I fix that problem
Goto C:\Users\"Your Username"\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266
Now you will see two version. you need to old version then restart your arduino IDE.

ksrivasu, 2017-10-26:
i am using ARDUINO UNO and ESP8266 

my board setting like this.

board : generic esp8266 module
Board esp8285 (platform esp8266, package esp8266) is unknown

Error compiling for board Generic ESP8285 Module.

castor666, 2017-11-18:
Thx! @borhanreo
greetings

palivoda, 2018-03-16:
After boards update you need to restart IDE. 
Check there is no previous version in ....\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266

SYAMBABUDARSI, 2019-02-20:
Board nodemcuv2 (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

blakeedwards823, 2019-06-17:
> #1377

A concise tutorial explaining the issue can be found here! 
https://arduino-esp8266.readthedocs.io/en/latest/faq/a04-board-generic-is-unknown.html