esp8266/Arduino #1516: wdt reset with ESP8266WiFi.h include

steenl, 2016-01-27:
Simply including ESP8266WiFi.h blinky sketch (Arduino 1.6.5 and board esp8266 v2.0.0) generates repeated wdt resets below.

This is across multiple esp-v01 modules.  AT cmds allow me to login to my AP fine, so wifi seems to work.
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```
Code that triggers wdt resets below, and any suggestions greatly appreciated. :-)
```
#include <ESP8266WiFi.h>
int pin = 2;
void setup() {
 Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println("start");
  delay(10);
  pinMode(pin, OUTPUT);
}
void loop() {
  digitalWrite(pin, HIGH);   
  Serial.println("pin high");
  delay(1000);               
  digitalWrite(pin, LOW);   
  Serial.println("pin low");
  delay(1000);               
}
```

steenl, 2016-01-29:
Something was wrong with my programming environment, perhaps an old install of 1.6.6.

closed 2016-01-29 by steenl

sonyarouje, 2016-10-28:
I am getting very similar issue. My code works fine without #include <ESP8266WiFi.h> and I can see hello world printing to my serial monitor.

I use a USB to FTDI module to upload code.

My ESP module is '[ESP8266 Serial ESP 01](http://www.ebay.in/itm/272282635593?aff_source=Sok-Goog)'

Not sure what is wrong here. I tried uploading the sketch via Arduino IDE 1.8.7 and 1.6.12. 

Any help will be really appreciated.

Below is my code.

`#include <ESP8266WiFi.h>
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("hello world");
  delay(1000);
}`

krzychb, 2016-10-31:
Hi @sonyarouje,

I suspect the `wdt reset` is cased by inadequate power supply to the ESP-01 module. 

To verify it, if you have a scope, consider checking the current spikes drawn by module with and without `ESP8266WiFi.h` included.

Regarding properly connecting ESP-01, please check section on [Generic ESP8266 modules]( https://github.com/esp8266/Arduino/blob/master/doc/boards.md#generic-esp8266-modules).

Also, for initial tests, consider using module with integrated power supply and usb to serial adapter. See section [Introduction](https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md#introduction) for examples of such modules (the rest of article is not relevant).

rashey04ec, 2019-07-19:
I have solved it !!,
Please try to comment all pin you used in programming and it's related function )like pinMode , digitalWrite etc.
 - Again save and upload it 
- check the serial monitor 
- if you not getting error that means this is because of pin declaration
-  to solve this issue : dont use #define declaration for initialize pin.
-Direct use pin in pinMode definition and read /write function
- Use GPIO no in if you are doing it in ARDUINO IDE

May be helpfull 

Thanks : 
Jaydeep Shah
E-mail : radhey04ec@gmail.com