esp8266/Arduino #1517: issue with strcpy

Duality4Y, 2016-01-27:
hi there! i am having a verry weird issue with strcpy.

here is some code:
```
// this function interprets the data.
// and executes it.
static char* str_input = NULL;
const char delim[] = " \n";
void interpretInput(String input)
{
    char *svptr = NULL;
    if(input.length())
    {
        // create a copy to be used with strtok_r. free before if: it is used before.
        if(str_input != NULL)
        {
            Serial.println("freeing str_input");
            free(str_input);
        }
        size_t str_input_len = (sizeof(char) * input.length() + 3) & (~3);
        Serial.printf("str_input_len: %d\n", str_input_len);
        str_input = (char *)malloc(str_input_len);
        Serial.println("str_input == NULL");
        if(str_input == NULL)
        {
            Serial.println("Input Copy Error.");
            return;
        }
        Serial.println("strcpy");
        memset(str_input, '\0', input.length());
        strcpy(str_input, input.c_str());
        return;
        // check for leading whitespace.
        Serial.println("isspace");
        if(isspace(str_input[0]))
        {
            Serial.println("Syntax error");
            return;
        }
        // trim trailing whitespace.
        Serial.println("strlen");
        int len = strlen(str_input);
        Serial.println("while isspace");
        while(isspace(str_input[len - 1]))
        {
            str_input[len - 1] = '\0';
            len = strlen(str_input);
            delay(0);
        }

        // strtok merges adjecent delimeters (white space in this case)
        // that is good for us it means that it doesn't matter
        // how many white space we add between arguments.
        
        // first token is the actuall command.
        Serial.printf("input_str: %s\n", str_input);
        Serial.printf("input: %s\n", input.c_str());
        Serial.printf("strlen(input_str): %d\n", strlen(str_input));
        Serial.println("command = string(strtok(str_input, delim");
        char* command = strtok_r(str_input, delim, &svptr);
        Serial.printf("command: %s\n", command);
        Serial.println("command: " + String(command) + " len: " + strlen(command));
        
        // max allowed length of argv, thus max allowed arguments.
        static int argc;
        static const uint8_t max_args = 20;
        String argv[max_args];

        // parse out all arguments for aslong as there are arguments.
        argc = 0;
        Serial.println("argument = strtok(NULL, delim");
        char* argument = strtok_r(NULL, delim, &svptr);
        Serial.printf("argument: %s\n", argument);
        Serial.println("argument != NULL");
        while(argument != NULL)
        {
            // put the arguments in order in argv
            Serial.println("argv[argc] = String(argument);");
            argv[argc] = String(argument);
            // Serial.println("Argument: " + argv[argc] + " Len: " + argv[argc].length());
            Serial.println("argument = strtok(NULL, delim);");
            argument = strtok_r(NULL, delim, &svptr);
            Serial.printf("argc++: %d\n", argc+1);
            argc++;
            // keep track of where to put argument in argv.
            // and check if we are not exeeding max arguments.
            // also is length of argv thus argc.
            Serial.println("if(argc >= max_args)");
            if(argc >= max_args)
            {
                return;
            }
        }
        // Serial.println("argc: " + String(argc));

        // check for command in commands list. and call it's handler.
        Serial.println("for command in commands");
        for(int i = 0; i < numCommands;i++)
        {
            Serial.println("command == commands[i].name");
            if(String(command) == commands[i].name)
            {
                // call that command handle
                (commands[i].handle)(argc, argv);
                // print this to show the interactive interpreter is back
                // to do our bidding!
                Serial.print(">> ");
                return;
            }
        }

        // if we reach here it's unkown input/command.
        Serial.println("unknown input: '" + input + "'");
     }
}
```

so here is the thing, 
If i put the return after the strcpy it crashes only after a while like 20 seconds later.
but! if i put the return before the strcpy it never crashes!

also when i pass this function a string that is uneven in length it crashes.
when i pass it a string that is even it doesn't !
I have tested this with a string of length 7 and string of length 4.
I have tried lots of things but can't get it to work.

maybe I am not seeing somthing.
any ideas?

here is the output:
```
loading: 
read: deadbeaf
should be: deadbeaf
valid settings!
board_name: testhost
ssid: ssid
pass: password
wifimode: no valid mode set

setting up SD.
sd begin succeeded.
setting up wifi.
connecting
.......
connected
IP: 10.42.4.36
setup ota.
setup http server.
HTTP Server started.
str_input_len: 8
str_input == NULL
strcpy

Exception (28):
epc1=0x401055b5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00002408 depc=0x00000000

ctx: sys 
sp: 3ffffd20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffec0:  40000f3d 00000023 00000000 3fffd9d0  
3ffffed0:  00000000 00000000 402399d8 4021fb9a  
3ffffee0:  00000000 3fff1a18 00000008 3fffdab0  
3ffffef0:  0000005a 3fffdad0 4021ee78 00000016  
3fffff00:  3fff1a18 06b6aec5 3ffec228 3ffec27c  
3fffff10:  00000000 3ffec25a 00000000 00000000  
3fffff20:  4021edc4 3fff1a18 3ffeef34 00000000  
3fffff30:  4021eda6 00000001 4020e207 3ffe97d4  
3fffff40:  40104ca7 00000000 3ffef230 00000001  
3fffff50:  40223530 40223529 00000000 1602db10  
3fffff60:  00000000 4022b345 3ffef208 3ffef230  
3fffff70:  1602f34c 60000600 3ffe9a00 3ffe9a00  
3fffff80:  4022b38a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffef250 3fffdad0 3fff07b0 40208edb  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

loading: 
read: deadbeaf
should be: deadbeaf
valid settings!
board_name: testhost
ssid: ssid
pass: password
wifimode: no valid mode set

setting up SD.
sd begin succeeded.
setting up wifi.
connecting
.......
connected
IP: 10.42.4.36
setup ota.
setup http server.
HTTP Server started.
```

Duality4Y, 2016-01-27:
ok here is the weird thing doesn't crash using nc when following this:
https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub

Duality4Y, 2016-01-27:
it is really around the strcpy and i don't understand why! it makes no sense!
if i bring it down to this code:
```
void interpretInput(String input)
{
    char *svptr = NULL;
    static char* str_input = NULL;
    const char delim[] = " ";
    if(input.length())
    {
        if(str_input != NULL)
        {
            free(str_input);
        }

        size_t malloc_amount = (sizeof(char) * input.length() + 3) & (~3);
        str_input = (char *)malloc(malloc_amount);
        if(str_input == NULL)
        {
            Serial.println("Input Copy Error.");
            return;
        }

        memset(str_input, '\0', input.length());
        strcpy(str_input, (char *)input.c_str());
        Serial.printf("str_input: %s\n", str_input);
        Serial.printf("strlen(str_input): %d\n", strlen(str_input));
    }
}
```

when i remove the strcpy it never crashes!,
but with strcpy it crashes.

alltheblinkythings, 2016-01-27:
strcpy will write input.length()+1 bytes as it writes a trailing NUL
character to terminate the string.  Have you tried just using
malloc(input.length()+1) ?

Duality4Y, 2016-01-27:
yes that is what i ended up doing and that worked!
stupid of me! :)

closed 2016-01-27 by Duality4Y