esp8266/Arduino #1520: ESP8266HTTPClient without timeout

PlatenySmartiks, 2016-01-27:
Why there is no timeout on handleHeaderResponse method? My code does simple POST calls and stuck sometimes waiting for a response that never comes.

Links2004, 2016-01-27:
there is, on timeout readStringUntil will return a empty string ""
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L830
the empty string will trigger here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L861

and then return:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L873
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L876
or
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L883

try to enable debug for the HTTP client:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md
and post the log here.

PlatenySmartiks, 2016-01-27:
Thanks for the answer! I'll enable debug later.
Is it possible to happen a situation in which he lost the response (internet access fails) and stays stuck like below, always connected?
_tcp->available() == 0
_tcp->connected() == 1

Links2004, 2016-01-27:
if internet access fails connected will return false.
if not there is a problem in WiFiClient or SDK.

the condition you note is may possible when the server never send anything and keeps the connection open. 
but no normal http server will do that, adding a additional time out for this case is possible.
the debug will give us more info.

closed 2016-03-09 by igrr