esp8266/Arduino #1521: SSID hide

kapilpawar, 2016-01-28:
how to hide my SSID of esp8266..

is there any AT command in esp8266....

thanks in advance ..

Links2004, 2016-01-28:
this repo is a full release for AT firmware.
with the software here you can Programm the ESP module like any other Arduino board.

for this repo its possible:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.h#L39
no idea for AT never used it.

kapilpawar, 2016-01-29:
Hi Links2004,

do i need to update the firmware of esp8266 before programing

kapilpawar, 2016-01-29:
@Links2004 

can  you please give me full steps for hiding the ssid of esp8266

Links2004, 2016-01-29:
start here:
https://github.com/esp8266/Arduino#installing-with-boards-manager

use the AP example
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino#L56

but with:
```cpp
// ssid, password, channel, hide
WiFi.softAP(ssid, password, 1, 1);
```

kapilpawar, 2016-01-29:
@Links2004 

Thanks dear

but it can not be uploaded 
it is showing some error.
the error is like ....

Sketch uses 246,926 bytes (56%) of program storage space. Maximum is 434,160 bytes.
Global variables use 38,144 bytes (46%) of dynamic memory, leaving 43,776 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed

please give me resolution...

Links2004, 2016-01-29:
i not can give you a "resolution", 
I not know how you wired the ESP with you PC or whats connected to all the GPIOs or what ESP module you use.
I only can give you hints whats going wrong.

check list:
 - select the correct board in the IDE.
 - is the wirering correct https://github.com/esp8266/Arduino/blob/master/doc/boards.md#esp-to-serial
 - is the power supply stable, 3.3V with more then 250mA (not use the usb chip as supply) https://github.com/esp8266/Arduino/blob/master/doc/boards.md#serial-adapter
 - have alle the pins the correct level at bootup (Uart mode) https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-mode

kapilpawar, 2016-02-01:
@Links2004 
Thanks

closed 2016-02-01 by kapilpawar

ariffinzulkifli, 2016-09-29:
thank you @Links2004

thebeswick, 2017-01-02:
I just had the same problem, you can set 
`WiFi.enableAP(0);`

before starting your wireless client.

majidmp97, 2018-10-21:
> WiFi.enableAP(0);

thank you very much. i need this very long time