esp8266/Arduino #1523: Exception (9) around pvPortMalloc

supersjimmie, 2016-01-28:
My ESP keeps crashing after a while, now I have cought one crash as follws:

The code part:

```
  lines = "";
  while (client.available()) {
    line = client.readStringUntil('\r');
    if ((lines.length() + line.length()) < 127) {
      lines += line;
    }
    delay(10);
    yield();
    ESP.wdtFeed();
    Serial.print(".");
  }
```

The crash:

```
Exception (9):
epc1=0x40105a11 epc2=0x00000000 epc3=0x00000000 excvaddr=0xcebfcadd depc=0x00000000

ctx: cont
sp: 3fff22e0 end: 3fff2650 offset: 01a0
```

Which, I believe, points to about this part:

```
401059d8 <pvPortMalloc>:
401059d8:	e0c112        	addi	a1, a1, -32
401059db:	61e9      	s32i.n	a14, a1, 24
401059dd:	2139      	s32i.n	a3, a1, 8
401059df:	3109      	s32i.n	a0, a1, 12
401059e1:	41c9      	s32i.n	a12, a1, 16
401059e3:	51d9      	s32i.n	a13, a1, 20
401059e5:	02cd      	mov.n	a12, a2
401059e7:	04dd      	mov.n	a13, a4
401059e9:	ee1c01        	l32r	a0, 4010125c <__muldi3+0x70>
401059ec:	0000c0        	callx0	a0
401059ef:	ffb6e1        	l32r	a14, 401058c8 <ets_timer_arm_new+0xf4>
401059f2:	7e08      	l32i.n	a0, a14, 28
401059f4:	40cc      	bnez.n	a0, 401059fc <pvPortMalloc+0x24>
401059f6:	fff701        	l32r	a0, 401059d4 <xPortWantedSizeAlign+0x24>
401059f9:	0000c0        	callx0	a0
401059fc:	1cac      	beqz.n	a12, 40105a21 <pvPortMalloc+0x49>
401059fe:	0c2d      	mov.n	a2, a12
40105a00:	fffac5        	call0	401059b0 <xPortWantedSizeAlign>
40105a03:	02cd      	mov.n	a12, a2
40105a05:	829c      	beqz.n	a2, 40105a21 <pvPortMalloc+0x49>
40105a07:	8e28      	l32i.n	a2, a14, 32
40105a09:	9e08      	l32i.n	a0, a14, 36
40105a0b:	12bc27        	bgeu	a12, a2, 40105a21 <pvPortMalloc+0x49>
40105a0e:	ffef71        	l32r	a7, 401059cc <xPortWantedSizeAlign+0x1c>
40105a11:	1038      	l32i.n	a3, a0, 4
40105a13:	0048      	l32i.n	a4, a0, 0
40105a15:	2cb3c7        	bgeu	a3, a12, 40105a45 <pvPortMalloc+0x6d>
40105a18:	94ac      	beqz.n	a4, 40105a45 <pvPortMalloc+0x6d>
40105a1a:	007d      	mov.n	a7, a0
40105a1c:	040d      	mov.n	a0, a4
40105a1e:	fffbc6        	j	40105a11 <pvPortMalloc+0x39>
40105a21:	0d0c      	movi.n	a13, 0
40105a23:	ee3601        	l32r	a0, 401012fc <pp_soft_wdt_feed_local+0x94>
40105a26:	0000c0        	callx0	a0
40105a29:	61e8      	l32i.n	a14, a1, 24
40105a2b:	9dcc      	bnez.n	a13, 40105a38 <pvPortMalloc+0x60>
40105a2d:	0c3d      	mov.n	a3, a12
40105a2f:	ffe821        	l32r	a2, 401059d0 <xPortWantedSizeAlign+0x20>
40105a32:	ee8101        	l32r	a0, 40101438 <ppTxqUpdateBitmap+0x28>
40105a35:	0000c0        	callx0	a0
```

How could this be solved? (I do not really understand how to track this further)

supersjimmie, 2016-01-30:
I suspect it is this line:
```
lines += line;
```
If that is the problem, is that memory fragmentation, and how can I prevent that?
(line.concat() does the same?)

Links2004, 2016-01-30:
you can try to allocate the needed size before using 
the String class has a function called ```reserve```

the 
```
    yield();
    ESP.wdtFeed();
```
are not needed the delay will use yield and the SDK behind will then feed the SW wdt.

supersjimmie, 2016-01-30:
Yes I have seen reserve(), but I also noticed that in the concat function the reserve is also called. I believe that is the reserve that causes the fragmentation (because it keeps adding extra pieces).
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.cpp#L266-L277
Even if I do a reserve in advance, the concat will still do it again with small parts, causing fragmentation?

igrr, 2016-01-30:
Reserve check if (buffer && capacity >= size), so if you reserve enough in
advance, it won't cause more allocations.

supersjimmie, 2016-01-30:
Ah I missed that, thanks. I will do a reserve in advance then!
But am I right that the exception is caused by fragmentation?

supersjimmie, 2016-02-01:
Did some more testing and I'm quiet convinced that multiple times += or .concat() is the cause.
Solved it by rewriting the code so this is no longer needed.

supersjimmie, 2016-02-04:
Reopening this...  It seems not to be concatenating.
I still get the Exception (9) at this point.
```
Exception (9):
epc1=0x40105b9a epc2=0x00000000 epc3=0x00000000 excvaddr=0x3af0e1bf depc=0x00000000
ctx: cont
```
Global:
```
String line;
```
In setup():
```
  line.reserve(512);
```
(large beause I am using it at several points but it never gets filled so much)

In a function:
```
    line = client.readStringUntil('\r');
```
The code is in here:
```
40105b44 <pvPortRealloc>:
40105b44:	e0c112        	addi	a1, a1, -32
40105b47:	31d9      	s32i.n	a13, a1, 12
40105b49:	1109      	s32i.n	a0, a1, 4
40105b4b:	0139      	s32i.n	a3, a1, 0
40105b4d:	51f9      	s32i.n	a15, a1, 20
40105b4f:	21c9      	s32i.n	a12, a1, 8
40105b51:	6149      	s32i.n	a4, a1, 24
40105b53:	41e9      	s32i.n	a14, a1, 16
40105b55:	054d      	mov.n	a4, a5
40105b57:	05ed      	mov.n	a14, a5
40105b59:	02cd      	mov.n	a12, a2
40105b5b:	61f8      	l32i.n	a15, a1, 24
40105b5d:	032d      	mov.n	a2, a3
40105b5f:	0f3d      	mov.n	a3, a15
40105b61:	ffe745        	call0	401059d8 <pvPortMalloc>
40105b64:	02dd      	mov.n	a13, a2
40105b66:	329c      	beqz.n	a2, 40105b7d <pvPortRealloc+0x39>
40105b68:	1c9c      	beqz.n	a12, 40105b7d <pvPortRealloc+0x39>
40105b6a:	0c3d      	mov.n	a3, a12
40105b6c:	0148      	l32i.n	a4, a1, 0
40105b6e:	f59d01        	l32r	a0, 401031e4 <wDevDisableRx+0x40>
40105b71:	0000c0        	callx0	a0
40105b74:	0e4d      	mov.n	a4, a14
40105b76:	0f3d      	mov.n	a3, a15
40105b78:	0c2d      	mov.n	a2, a12
40105b7a:	fff245        	call0	40105aa0 <vPortFree>
40105b7d:	1108      	l32i.n	a0, a1, 4
40105b7f:	21c8      	l32i.n	a12, a1, 8
40105b81:	41e8      	l32i.n	a14, a1, 16
40105b83:	0d2d      	mov.n	a2, a13
40105b85:	51f8      	l32i.n	a15, a1, 20
40105b87:	31d8      	l32i.n	a13, a1, 12
40105b89:	20c112        	addi	a1, a1, 32
40105b8c:	f00d      	ret.n
40105b8e:	610000        	excw
40105b91:	8f          	.byte 0x8f
40105b92:	ff          	.byte 0xff
40105b93:	0658      	l32i.n	a5, a6, 0
40105b95:	06b527        	bgeu	a5, a2, 40105b9f <pvPortRealloc+0x5b>
40105b98:	056d      	mov.n	a6, a5
40105b9a:	0558      	l32i.n	a5, a5, 0
40105b9c:	f83527        	bltu	a5, a2, 40105b98 <pvPortRealloc+0x54>
40105b9f:	1678      	l32i.n	a7, a6, 4
40105ba1:	376a      	add.n	a3, a7, a6
40105ba3:	079237        	bne	a2, a3, 40105bae <pvPortRealloc+0x6a>
40105ba6:	1248      	l32i.n	a4, a2, 4
40105ba8:	062d      	mov.n	a2, a6
40105baa:	447a      	add.n	a4, a4, a7
40105bac:	1649      	s32i.n	a4, a6, 4
40105bae:	1278      	l32i.n	a7, a2, 4
40105bb0:	872a      	add.n	a8, a7, a2
40105bb2:	0f9587        	bne	a5, a8, 40105bc5 <pvPortRealloc+0x81>
40105bb5:	ff4481        	l32r	a8, 401058c8 <ets_timer_arm_new+0xf4>
40105bb8:	7888      	l32i.n	a8, a8, 28
40105bba:	101587        	beq	a5, a8, 40105bce <pvPortRealloc+0x8a>
40105bbd:	1598      	l32i.n	a9, a5, 4
40105bbf:	0558      	l32i.n	a5, a5, 0
40105bc1:	997a      	add.n	a9, a9, a7
40105bc3:	1299      	s32i.n	a9, a2, 4
40105bc5:	0259      	s32i.n	a5, a2, 0
40105bc7:	011267        	beq	a2, a6, 40105bcc <pvPortRealloc+0x88>
40105bca:	0629      	s32i.n	a2, a6, 0
40105bcc:	f00d      	ret.n
40105bce:	085d      	mov.n	a5, a8
40105bd0:	fffc46        	j	40105bc5 <pvPortRealloc+0x81>
40105bd3:	ffff00        	excw
40105bd6:	fffd      	excw
```
Specifically this line then:
```
40105b9a:	0558      	l32i.n	a5, a5, 0
```

igrr, 2016-02-04:
Which version of the core are you using? In case that's 2.1.0-rc2, could you please try git version? The issue is that heap is getting corrupted, and one particular reason for this has been fixed recently.

supersjimmie, 2016-02-04:
I am using GIT from 13 januari 2016.
As far as I could see, nothing specific changed between my last download and now?

supersjimmie, 2016-02-04:
ah my bad...
now I see RC2 is from 24th so my git is older.
Will update!

supersjimmie, 2016-02-04:
I did a complete GIT download again.
So far no new Exception (9), but instead I've had an Exception (0) while it was just printing a short line to Serial with ```Serial.println()```, and an Exception (29) while doing a web request with the WiFiClient. (something with ppCheckTxIdle).

supersjimmie, 2016-02-09:
Problem did not occur anymore, seems to be solved with latest git.

closed 2016-02-09 by supersjimmie