esp8266/Arduino #1524: Make SD Class be able to coexist with SPIFFS

me-no-dev, 2016-01-28:
we should rework this to use the same API (will look into it)
but this is for now
https://github.com/me-no-dev/Arduino/commit/72c2c3f4e2a0c4de29ce5f946417aed6130748d5

tzapu, 2016-03-14:
hi, if this is to be done at some point, maybe it s worth adding long file names as well and this suggestion here
https://github.com/esp8266/Arduino/issues/641
SdFat lib with LFN support https://github.com/tannewt/SdFat

igrr, 2016-06-02:
Fixed in 43fb139

closed 2016-06-02 by igrr

DedeHai, 2016-10-15:
Do you have an example on how to use the two file systems side by side? Something like open, access and close a file on SPIFFs then on SD card.

Bushii7, 2017-09-16:
full code please :/

duckylock, 2019-02-21:
This not really works for me. The namespace error disappeared but the read / write experience seams to not working for me. In your code you dont use any include SD.h is it a mistake or...? And if it missed where is it suppose to go? Before? After? Thanks

d-a-v, 2019-02-21:
@duckylock you may try https://github.com/esp8266/Arduino/pull/5525 (soon to be merged)