esp8266/Arduino #1525: what is the excption?

EagleSmith, 2016-01-29:
dhcp server start:(ip:192.168.4.1,mask:255.
Exception (0):
epc1=0x402103f4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc10 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdb0:  40223e00 3fff6f50 00000010 60000600  
3ffffdc0:  00000001 4021f774 3fffc250 4000050c  
3ffffdd0:  400043d5 00000030 00000016 ffffffff  
3ffffde0:  400044ab 3fffc718 3ffffed0 08000000  
3ffffdf0:  60000200 08000000 00000003 00000000  
3ffffe00:  0000ffff 00000001 04000002 003fd000  
3ffffe10:  3fff7188 000003fd 3fff2564 00000030  
3ffffe20:  40101709 00000008 00000008 00000020  
3ffffe30:  c1948db3 394c5e70 7f2060f2 c6ba0c87  
3ffffe40:  3fff7058 00000001 40238d41 3fff6ff0  
3ffffe50:  3fff6f50 00000010 60000600 00000020  
3ffffe60:  402301a8 3fff7098 3fff7014 40238c77  
3ffffe70:  4022fb6c 40230ebe 3fff1a5b 3fff6f00  
3ffffe80:  3ffffec8 00000010 40231061 3fff0f90  
3ffffe90:  3fff6848 3ffed0c0 60000600 3fff6ae0  
3ffffea0:  3fff0f90 3fff0f90 3fff6848 3fff6d40  
3ffffeb0:  3fff28e8 40101233 d634fe1a fffeffff  
3ffffec0:  00000001 00000000 4022d5d6 3fff6848  
3ffffed0:  00000002 4000410f 3fff2394 3fff6848  
3ffffee0:  3fffc718 40004a3c 000003fd 3fff7188  
3ffffef0:  3fffc718 40101510 00000378 3fff1a5b  
3fffff00:  000003fd 4021d2e7 00000378 000003ff  
3fffff10:  00001000 4021d37d 3fff2564 000003ff  
3fffff20:  000003fd 60000600 003fd000 3fff2564  
3fffff30:  ffffff00 55aa55aa 00000312 0000001c  
3fffff40:  0000001c 0000008a 0000006d 000003ff  
3fffff50:  4021d224 3ffecf90 00000000 3ffed0c0  
3fffff60:  00000001 4021c2e9 00000003 3fff1238  
3fffff70:  4021c071 3ffecf84 3ffecf30 0026a2b0  
3fffff80:  4021c0b6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffecf40 3fffdab0 00000000 3fffdcc0  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

EagleSmith, 2016-01-29:
this is the code?

WiFi.disconnect();
    apnanme = loadConfig("AP_NAME");
    appassword = loadConfig("AP_PASSWORD");
    WiFi.softAP(apnanme.c_str(), appassword.c_str());
    delay(1000);

    //pinMode(SCK, OUTPUT);
    //pinMode(DOUT, INPUT);
    //digitalWrite(SCK, LOW);

    //host web server
    handleWebServer();
    configMode = true;
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);

CombiesGit, 2016-01-29:
> Exception (0):
> epc1=0x402103f4
An illegal instruction als address 0x402103f4

you can use "objdump -d -S   prog.elf" to see, where it points to.

EagleShi, 2016-01-30:
40210358 <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE>:
40210358:	c0c112        	addi	a1, a1, -64
4021035b:	b1f9      	s32i.n	a15, a1, 44
4021035d:	03fd      	mov.n	a15, a3
4021035f:	4c0f62        	l8ui	a6, a15, 76
40210362:	d1d9      	s32i.n	a13, a1, 52
40210364:	f109      	s32i.n	a0, a1, 60
40210366:	e1c9      	s32i.n	a12, a1, 56
40210368:	c1e9      	s32i.n	a14, a1, 48
4021036a:	02dd      	mov.n	a13, a2
4021036c:	053d      	mov.n	a3, a5
4021036e:	042d      	mov.n	a2, a4
40210370:	56cc      	bnez.n	a6, 40210379 <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0x21>
40210372:	0d69      	s32i.n	a6, a13, 0
40210374:	1d69      	s32i.n	a6, a13, 4
40210376:	002406        	j	4021040a <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0xb2>
40210379:	00c705        	call0	40210fec <_Z13getSpiffsModeN2fs8OpenModeENS_10AccessModeE>
4021037c:	042f32        	l32i	a3, a15, 16
4021037f:	f44020        	extui	a4, a2, 0, 16
40210382:	00a052        	movi	a5, 0
40210385:	234b      	addi.n	a2, a3, 4
40210387:	20cf32        	addi	a3, a15, 32
4021038a:	f21445        	call0	402024d0 <SPIFFS_open_by_dirent>
4021038d:	0e0c      	movi.n	a14, 0
4021038f:	023d      	mov.n	a3, a2
40210391:	07a2e7        	bge	a2, a14, 4021039c <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0x44>
40210394:	0de9      	s32i.n	a14, a13, 0
40210396:	1de9      	s32i.n	a14, a13, 4
40210398:	001b86        	j	4021040a <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0xb2>
4021039b:	c23c00        	excw
4021039e:	4139      	s32i.n	a3, a1, 16
402103a0:	01e9      	s32i.n	a14, a1, 0
402103a2:	11e9      	s32i.n	a14, a1, 4
402103a4:	fed085        	call0	4020f0b0 <_Znwj>
402103a7:	02cd      	mov.n	a12, a2
402103a9:	0129      	s32i.n	a2, a1, 0
402103ab:	4138      	l32i.n	a3, a1, 16
402103ad:	4f28      	l32i.n	a2, a15, 16
402103af:	ec9c      	beqz.n	a12, 402103d1 <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0x79>
402103b1:	ff3441        	l32r	a4, 40210084 <_ZN14SPIFFSFileImpl5closeEv+0x20>
402103b4:	1c29      	s32i.n	a2, a12, 4
402103b6:	0c49      	s32i.n	a4, a12, 0
402103b8:	045c32        	s16i	a3, a12, 8
402103bb:	2ccb      	addi.n	a2, a12, 12
402103bd:	203ee0        	or	a3, a14, a14
402103c0:	2ca042        	movi	a4, 44
402103c3:	c33d01        	l32r	a0, 402010b8 <uart_write_char_d+0x98>
402103c6:	0000c0        	callx0	a0
402103c9:	0c2d      	mov.n	a2, a12
402103cb:	384ce2        	s8i	a14, a12, 56
402103ce:	ffe0c5        	call0	402101dc <_ZNK14SPIFFSFileImpl8_getStatEv>
402103d1:	020c      	movi.n	a2, 0
402103d3:	2129      	s32i.n	a2, a1, 8
402103d5:	421c      	movi.n	a2, 20
402103d7:	01c8      	l32i.n	a12, a1, 0
402103d9:	fecd45        	call0	4020f0b0 <_Znwj>
402103dc:	c28c      	beqz.n	a2, 402103ec <_ZN13SPIFFSDirImpl8openFileEN2fs8OpenModeENS0_10AccessModeE+0x94>
402103de:	130c      	movi.n	a3, 1
402103e0:	1239      	s32i.n	a3, a2, 4
402103e2:	2239      	s32i.n	a3, a2, 8
402103e4:	ffa731        	l32r	a3, 40210280 <_ZN14SPIFFSFileImplC1EP10SPIFFSImpls+0x34>
402103e7:	32c9      	s32i.n	a12, a2, 12
402103e9:	006232        	s32i	a3, a2, 0
402103ec:	1138      	l32i.n	a3, a1, 4
402103ee:	1129      	s32i.n	a2, a1, 4
402103f0:	218b      	addi.n	a2, a1, 8
402103f2:	2139      	s32i.n	a3, a1, 8
402103f4:	008085        	call0	40210c00 <_ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE0EED1Ev>
402103f7:	0128      	l32i.n	a2, a1, 0
402103f9:	1138      	l32i.n	a3, a1, 4
402103fb:	0d29      	s32i.n	a2, a13, 0
402103fd:	020c      	movi.n	a2, 0
402103ff:	1129      	s32i.n	a2, a1, 4
40210401:	0129      	s32i.n	a2, a1, 0
40210403:	1d39      	s32i.n	a3, a13, 4
40210405:	214b      	addi.n	a2, a1, 4
40210407:	007f85        	call0	40210c00 <_ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE0EED1Ev>
4021040a:	f108      	l32i.n	a0, a1, 60
4021040c:	0d2d      	mov.n	a2, a13
4021040e:	e1c8      	l32i.n	a12, a1, 56
40210410:	d1d8      	l32i.n	a13, a1, 52
40210412:	c1e8      	l32i.n	a14, a1, 48
40210414:	b1f8      	l32i.n	a15, a1, 44
40210416:	40c112        	addi	a1, a1, 64
40210419:	f00d      	ret.n
4021041b:	d07800        	subx2	a7, a8, a0
4021041e:	fe          	.byte 0xfe
4021041f:	3f          	.byte 0x3f

i can't find the program ,please help me to see,thank you !

Links2004, 2016-01-30:
the error happens by the call of SPIFFS Dir openFile
check you usage of SPIFFS

EagleShi, 2016-01-31:
thank you very much.

devyte, 2017-10-18:
Closing due to age and SPIFFS working correctly in latest git.

closed 2017-10-18 by devyte