esp8266/Arduino #1526: EEPROM crashing the sketch

shayo, 2016-01-29:
Hi,

I have an annoying problem where EEPROM writes to memory cause the 8266 to crash.
I'm running with Huzzah (Esp12) from AdaFruit, and using the latest(?) code 2.0.0.
See code and output below.

What am I missing?

Code:
------------------------------

#include <EEPROM.h>
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  EEPROM.begin(512);

  for (int k=0;k<512;k++)
  {
    byte b = k / 10;

    Serial.print("now writing in address ");
    Serial.print(k);
    Serial.print(" the value ");
    Serial.println(b);
    EEPROM.write(k, b);
    EEPROM.commit();
  }
  Serial.println("Done");
}

void loop() {
  // put your main code here, to run repeatedly:

}
----------------------------------------------------------

Output:

now writing in address 0 the value 0
now writing in address 1 the value 0
now writing in address 2 the value 0
now writing in address 3 the value 0
now writing in address 4 the value 0
now writing in address 5 the value 0
now writing in address 6 the value 0
now writing in address 7 the value 0
now writing in address 8 the value 0
now writing in address 9 the value 0
now writing in address 10 the value 1
now writing in address 11 the value 1
now writing in address 12 the value 1
now writing in address 13 the value 1
now writing in address 14 the value 1
now writing in address 15 the value 1
now writing in address 16 the value 1
now writing in address 17 the value 1
now writing in address 18 the value 1
now writing in address 19 the value 1
now writing in address 20 the value 2
now writing in address 21 the value 2
now writing in address 22 the value 2
now writing in address 23 the value 2
now writing in address 24 the value 2
now writing in address 25 the value 2
now writing in address 26 the value 2
now writing in address 27 the value 2
now writing in address 28 the value 2
now writing in address 29 the value 2
now writing in address 30 the value 3
now writing in address 31 the value 3
now writing in address 32 the value 3
now writing in address 33 the value 3
now writing in address 34 the value 3
now writing in address 35 the value 3
now writing in address 36 the value 3
now writing in address 37 the value 3
now writing in address 38 the value 3
now writing in address 39 the value 3
now writing in address 40 the value 4
now writing in address 41 the value 4
now writing in address 42 the value 4
now writing in address 43 the value 4
now writing in address 44 the value 4
now writing in address 45 the value 4
now writing in address 46 the value 4
now writing in address 47 the value 4
now writing in address 48 the value 4
now writing in address 49 the value 4
now writing in address 50 the value 5
now writing in address 51 the value 5
now writing in address 52 the value 5
now writing in address 53 the value 5
now writing in address 54 the value 5
now writing in address 55 the value 5
now writing in address 56 the value 5
now writing in address 57 the value 5
now writing in address 58 the value 5
now writing in address 59 the value 5
now writing in address 60 the value 6
now writing in address 61 the value 6
now writing in address 62 the value 6
now writing in address 63 the value 6
now writing in address 64 the value 6
now writing in address 65 the value 6
now writing in address 66 the value 6
now writing in address 67 the value 6

Soft WDT reset

ctx: cont 
sp: 3ffef850 end: 3ffefa50 offset: 01b0

>>>stack>>>
3ffefa00:  003fb000 40201bb5 3ffee8ec 3ffee8ec  
3ffefa10:  3fffdc20 3ffee8ec 3ffee9e8 40201880  
3ffefa20:  00000000 00000000 00000000 3ffeea1c  
3ffefa30:  3fffdc20 00000000 3ffeea14 4020241e  
3ffefa40:  00000000 00000000 3ffeea30 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

WereCatf, 2016-01-29:
Your for - loop is probably taking too long and triggering watchdog. Add a yield() or delay() to your for-loop to allow for background-tasks to do their thing. Also, why do you commit EEPROM after every single write? It'd be a lot more efficient to just commit all the 512 bytes at once by moving commit() outside the for-loop.

shayo, 2016-01-29:
OK. Got it. I added a delay(1) and things are now working. Thank you!

closed 2016-01-29 by shayo