esp8266/Arduino #1528: ESP8266-12E and Ethernet Shield

folny, 2016-01-29:
Hi Friends

Someone has tried to use ESP and Ethernet Shield (W5100) together ? If so what you should do to make it work and what should be the involvement of pins thanks.

pieman64, 2016-01-29:
@folny can't you tie the ESP to the Arduino without the Ethernet Shield?

folny, 2016-01-29:
I would like to use ESP and connecting via ethernet cable would that be ?. Have you tried it ?.

asetyde, 2016-01-29:
It’s possible but i’ve no information about this

folny, 2016-01-29:
He writes about it in this post but I'm not sure what you should do to make it work.

https://github.com/esp8266/Arduino/issues/962

asetyde, 2016-01-29:
Simple , connect all necessary pin , load ethernet library and use esp as cpu only ,but it's good use nodemcu with more pin .

Links2004, 2016-01-29:
yes its possible:
https://github.com/esp8266/Arduino/tree/master/libraries/Ethernet
connect the W5100 to the SPI pins of the ESP.
if you have the uno compatible board of the W5100 use the SPI pins of the ISP header to connect.
the normal one are not routed.

folny, 2016-01-29:
Yes it works now the problem was the involvement of pins thanks for the advice.

closed 2016-01-31 by folny

burner-, 2016-02-01:
I currently try draw nodemcu - arduino uno adapter. I mean adapter where is slot for nodemcu2 and it provide arduino uno layout for ethernet shield. But I dont know what are right pins? There is only info that use SPI pins but at nodemcu there is several SPI pins so what are right ones?

dvjcodec, 2016-02-16:
Maybe will help
http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/esp12.png

mkeyno, 2016-05-15:
hi @folny , do you have any success to bin ESP and Ethernet?

burner-, 2016-05-15:
If someone manage to use that code, can he or she please post what pin out he use. I have hit head to wall with this but I must say that given links does not help. They give hint how to you may connect your esp but really there is still multiple ways and none of them seems to work. Also do I need pull up or down resistors?

tusharvb19, 2017-04-20:
I got it working but the example code web-server does not getting the static ip rather it takes 192.168.0.240
Dont know why!
I have used the normal SPI pins on NodeMCU which are:

#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)

Arslanqayyum, 2018-03-06:
Hi Friends

Any success in connecting ESP8266 with Ethernet ?

burner-, 2018-03-06:
I did long while a go BUT it was quite unstable after all because shared WDT interrupt. I suggest that use esp32 because it have proper WDT.