esp8266/Arduino #1531: New library

KaloNK, 2016-01-29:
Hello,
 I have made two new libraries, but not sure what is the policy for adding them here (i.e. create a pull request) or creating a separate repository to host them?

1. SNTPtime - https://github.com/KaloNK/Arduino/tree/SNTPtime
 The potential problem here is that i borrowed some code from http://www.arduino.cc/playground/Code/Time and https://github.com/JChristensen/Timezone and modified it so the library can work without installing those libs or use them if available. Both are noted in SNTPtime.h with their initial sources, but not sure if it's OK

2. JSonWebConfig - https://github.com/KaloNK/Arduino/tree/JSonWebConfig
 It uses the SNTPtime library and ArduinoJson library https://github.com/bblanchon/ArduinoJson

igrr, 2016-01-29:
The most flexible solution is to add these libraries to Library Manager, this way you will be able to make changes and update libraries as often as you like.
Also do add these libraries to the list of supported libs in our docs.

devyte, 2017-10-18:
Closing per #3655 .

closed 2017-10-18 by devyte