esp8266/Arduino #1533: HttpClient POST?

shayo, 2016-01-29:
I'm trying to use HTTPClient to request a web page with POST, but this doesn't work.
I'm not sure if the problem is how I do it, or if there is a bug in the library.
Trying to send the variables with GET via the URL works fine.
Any ideas?

In 8266:
`http.begin("www.mywebpage.com", 80, "/php_page.php");
int httpCode = http.POST("VAR=123");
`
On the web:
`
php_page.php:
<? php
  $v = $_POST['VAR'] ;
   echo "<html><body><p> $v </p></body></html>";
?>
`

Links2004, 2016-01-29:
for url encode post add the header for url encode.
example:
```cpp
HTTPClient http;
http.begin("http://www.mywebpage.com/php_page.php");
http.addHeader("Content-Type", "application/x-www-form-urlencoded");
http.POST("title=foo&body=bar&userId=1");
http.writeToStream(&Serial);
http.end();
```
Note:  the example has no error handling, use the same style like in the GET example.

shayo, 2016-01-29:
Gotcha. I was missing the addHeader. I assumed it added this automatically when using the POST method. It might be useful to add this example to the release.
Thank you!

closed 2016-01-29 by shayo