esp8266/Arduino #1537: How to save variable into flash.

qblacksheep, 2016-01-30:
Hi, 
i want to save variable ( in my case it will be String device_id) into flash memory.
I tried to use arduino library progmem but it doesnt working.
Have u and idea how to do it?
i want to save it and when ESP8 reset i want load my variable from flash

WereCatf, 2016-01-30:
Use EEPROM, see the examples provided with the library.

qblacksheep, 2016-01-30:
When i am trying to read value from EEPROM its gives me and exception and program crash.
[image]

I want to read value from EEPROM when i Connected to WebSockets, it is my device_id
```javascript
 int address = 0;
         const char* id = (char*)EEPROM.read(address);
         address = address + 1;
          if (address == 512)
            address = 0;

        Serial.println("id z flash:");
        Serial.println(id);
        updateDeviceID(id);
```

and there is function when i want to save id into flash memory.
``` javascript
String newIDs = (char*)newID;
 Serial.println("do eeprom:");
 Serial.println(newIDs);
 int addr =0;
  EEPROM.write(addr, newIDs.toInt());
  addr = addr + 1;
  if (addr == 512)
  {
    addr = 0;
    EEPROM.commit();
  }
```

Links2004, 2016-01-30:
do you call ```EEPROM.begin(<needed size here>);```?

qblacksheep, 2016-01-30:
yes, in the Setup

qblacksheep, 2016-01-30:
How can i write String with using this library? I want to read value from EEPROM after connected to WebSockets and when its = 255( this is when i have nothing in EPPROM) i want to send undefined device_id but when something is in the EEPROM i want to read it and send it:
``` javascript
  int address = 0;
        byte id;
         id = EEPROM.read(address);
         address = address + 1;
          if (address == 512)
            address = 0;
        
        Serial.println("id z flash:");
        Serial.println(id, DEC);
        if(id = 255)
        {
        webSocket.sendTXT(registration);
        }
        else
        {
           const char* upid = (char*)id;
           updateDeviceID(upid);
           webSocket.sendTXT(registration);
        }
```
Here i save it:

``` javascript
void saveDeviceID(const char* id)
{
 
 Serial.println("do eeprom:");
 Serial.println(id);
   String newIDs = String(id);
  int addr =0;
  EEPROM.write(addr, newIDs.toInt());
  addr = addr + 1;
  if (addr == 512)
  {
    addr = 0;
    EEPROM.commit();
  }
  
}
``` 
its doesnt working.

WereCatf, 2016-01-30:
I don't even understand what you're trying to do. Every time you call saveDeviceID(), for example, addr = 0, so what's the point of comparing it to 512 since it'll never be that?

qblacksheep, 2016-01-30:
When i am connectiong to WebSockets, i am sending a Json with have device_ide:undefinded, server returns me device_id, exp. dev_0 and i am updating this id in my json.
 
I want:
1) save this id in EEPROM 
2) every time when i connect to WS, check is in EEPROM saved any id, if yes i want to use it in my json and send to server, if no i will get in from server and want to save in EEPROM.

my id is for example: dev_1

Duality4Y, 2016-01-30:
I think werecatf's point is that you never reach EEPROM.commit() and thus the settings are never storen in flash.

qblacksheep, 2016-01-30:
Do you know how to write String in the flash?

WereCatf, 2016-01-30:
Yes.
`size_t i;
char yourString[] = "Some text here.";
for(i=0; i<strlen(yourString); i++)  EEPROM.write(i, (uint8_t) yourString[i]);
EEPROM.commit();`

qblacksheep, 2016-01-30:
Is it the corret way to read it?
``` javascript
  char buf[5];
  for(i=0; i<=5; i++)
  {
  buf[i] = (char)EEPROM.read(i);
  }
```

Should i write it to adress 0? At adress 0x0000 i have boot.bin. At adress 0xfe000 i have init data defualt. Maybe i should write it on 0xfe000? but how to do it?

WereCatf, 2016-01-30:
EEPROM is an area located after your sketch and SPIFFS, it's not the same area as where the code resides, and when you give an address of 0 to the EEPROM-functions you're not giving an address on the physical flash layout, you're giving an index to where the EEPROM resides -- if the EEPROM was at address 0xffaa00, for example, and you gave an address of 5 to EEPROM.write() the function would actually write to 0xffaa00 + 5, not to 0x05. That is to say, yes, you should use value 0 and upwards from there to whatever you supplied to EEPROM.begin().

See https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md and https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#eeprom

Also, yes, reading it like that should work.

qblacksheep, 2016-01-30:
Thanks a lot! In example its work fine. But when i want use it in my program i still have problem, i wrote function wchich will save device id and console retrun me this:variable-sized object 'idtab' may not be initialized

```javascript
void saveDeviceID(const char* id)
{
  int siz =  strlen(id);
   size_t i; 
  char idtab[siz] = id;
  for(i=0; i<=strlen(idtab); i++) 
  {
  EEPROM.write(i, (uint8_t) idtab[i]); 
  }
  EEPROM.commit();
}
```

WereCatf, 2016-01-30:
Try this, instead.
```c++
void saveDeviceID(const char* id)
{
  size_t i; 
  for(i=0; i<=strlen(id); i++) 
  {
    EEPROM.write(i, (uint8_t) id[i]); 
  }
  EEPROM.commit();
}
```

Also, you're trying to define a fixed buffer, but then you're telling the compiler to use a variable to define its size -- it doesn't work like that. If you wanted a variable buffer-size you'd have to use malloc() to allocate the memory for it at runtime and then free() to release the allocated memory before leaving the function.

qblacksheep, 2016-01-31:
Ok, its working, but when i am reading form EEPROM i get: my id and .? after  recevied data. 
I wrote dev_5 to memory and i read dev_5.?

qblacksheep, 2016-01-31:
OKay, i fix it. Thanks for help mate!

closed 2016-02-03 by igrr