esp8266/Arduino #1538: Timer1 causing resets

bjpirt, 2016-01-30:
I'm trying to use timer1 to accurately drive some stepper motors using interrupts but I've run into a problem where when I use it, the Arduino will reset every 10 or so seconds. I've boiled down the code to this:

```C
void timer_callback(void){
  digitalWrite(13, !digitalRead(13));
}

void setup() {
  pinMode(13, OUTPUT);
  
  timer1_disable();
  timer1_attachInterrupt(timer_callback);
  timer1_enable(TIM_DIV1, TIM_EDGE, TIM_LOOP);
  timer1_write(1);
}

void loop() {
  delay(0);
}
```

Any ideas what I'm doing wrong? Also, I'd love to know what the number passed into timer1_write represents - it seems to be about 2 and a bit microseconds.

igrr, 2016-01-30:
You need to mark your timer callback with ICACHE_RAM_ATTR.

bjpirt, 2016-01-30:
That doesn't seem to have any effect unfortunately - still resets exactly the same :-(

bjpirt, 2016-01-31:
Figured it out - it needed `timer1_isr_init();` in there before enabling it.

closed 2016-01-31 by bjpirt

ThoughtFisher, 2017-06-13:
This is the answer to your problem bro:

void timer_callback(void){
  digitalWrite(2, !digitalRead(13));
  Serial.println("1");
  wdt_reset();
}
void setup() {
pinMode(2, OUTPUT);
 Serial.begin(115200);
  timer1_disable();
  timer1_attachInterrupt(timer_callback);
  timer1_isr_init();
  timer1_enable(TIM_DIV1, TIM_EDGE, TIM_LOOP);
  timer1_write(1000);

  //Set up ESP watchdog
  ESP.wdtDisable();
  ESP.wdtEnable(WDTO_8S);
}
void loop() {
  
}

you had problems with the watchdog because you dont have lines, then  watchdog detect inactivity and reset the esp8266 supposing an error.

kamalesh07, 2017-07-20:
i want to use timer as well as i need to receive data from mobile through wifi how it is possible