esp8266/Arduino #1539: Store any struct in flash.

Duality4Y, 2016-01-30:
Hi there I have been writing some code to store any struct that you can create.

so here is the functional code.
```cpp
void storeSettings(void *data_source, size_t size)
{
  EEPROM.begin(size * 2);
  for(size_t i = 0; i < size; i++)
  {
    char data = ((char *)data_source)[i];
    EEPROM.write(i, data);
  }
  EEPROM.commit();
}

void loadSettings(void *data_dest, size_t size)
{
    EEPROM.begin(size * 2);
    for(size_t i = 0; i < size; i++)
    {
        char data = EEPROM.read(i);
        ((char *)data_dest)[i] = data;
    }
}
```

it is verry simple every struct is passed by reference,
say  you have this code:
```cpp
typedef struct {
  char board_name[64];
  char ssid[64];
  char pass[64];
} settings_t __attribute__ ((packed));

settings_t settings
{
    "test_host",
    "ssid",
    "pass"
};
```

and you want to store that struct in eeprom/flash! you'd do:
```cpp
storeSettings(&settings, sizeof(settings));
```

and to load it:
```cpp
loadSettings(&settings, sizeof(settings));
```

here is my test code and that passes:
```cpp
void printCmpSettings(settings_t s1, settings_t s2)
{
    Serial.printf("board_name: %s | %s \n"
                  "ssid: %s | %s \n"
                  "pass: %s | %s \n"
                  "pincode: %d | %d \n",
                  s1.board_name, s2.board_name,
                  s1.ssid, s2.ssid,
                  s1.pass, s2.pass,
                  s1.currentmode, s2.currentmode
                  );
}

memcpy(&settings_original, &settings, sizeof(settings));
storeSettings(&settings, sizeof(settings));
loadSettings(&settings, sizeof(settings));
if(!memcmp(&settings_original, &settings, sizeof(settings)))
{
    Serial.println("settings are the same!");
    printCmpSettings(settings_original, settings);
}
else
{
    Serial.println("settings are not the same!");
    printCmpSettings(settings_original, settings);
}

memcpy(&settings_original, &settings, sizeof(settings));
strcpy(settings.board_name, "test_host");
storeSettings(&settings, sizeof(settings));
loadSettings(&settings, sizeof(settings));
if(!memcmp(&settings_original, &settings, sizeof(settings)))
{
    Serial.println("settings are the same!");
    printCmpSettings(settings_original, settings);
}
else
{
    Serial.println("settings are not the same!");
    printCmpSettings(settings_original, settings);
}
```

it is really nice because you can make those structs as complicated or simple as you want.
a few notes. your struct needs to be packed!
and it needs to be alligned with flash!
also if you want to put structs in a struct you want to store do it like this:
```cpp
struct Employee
{
   char ename[20];
   int ssn;
   float salary;
   struct date
       {
       int date;
       int month;
       int year; 
       }doj;
}emp1;
```

I Hope it helps someone out this code :)

marvinroger, 2016-01-30:
That's nice, thanks for sharing. But, there are already `EEPROM.get` and `EEPROM.put` that do exactly the same thing.

Duality4Y, 2016-01-30:
There is no example of this for the esp8266, so I decided to write it myself :)

marvinroger, 2016-02-05:
This is a great way to learn, good idea. The fact is the EEPROM API is the same as the Arduino one, except for `EEPROM.begin()` and `EEPROM.commit()`, so `put` and `get` are implemented just like on AVR. Given there are 300+ issues, I guess we can close this one, if you don't mind. ;)

closed 2016-02-05 by Duality4Y

shinnlu, 2017-05-03:
found a problem when writing a String variable in struct, esp8266 crash with Exception(3).

Makuna, 2017-05-03:
@shinnlu You can't store "complex" objects this way.  Any class that allocates memory to store anything will require special work to serialize and de-serialize that member otherwise you are just storing the pointer to the memory.  String is such a class.  

A common way to store variable length strings is to convert them to a BSTR (basic string) format, which starts with a 8 bit or 16 bit value that is the length of the string, followed by a buffer that is that length long containing the string.

shinnlu, 2017-05-03:
@Makuna i will change to char array or save with Json object, thanks.

YasirShahahzad, 2018-07-04:
Great Help, some people are saying its alternatives. But some times we need this type of code, thanks once again

pbravoz2, 2018-12-31:
Hi Robert,
                    this is a very usefull sketch for use efficiently and easy the EEPROM in NODEMCU. It solves a lot of problems for me. Congratulations!!. Is excellent!!
With best regards,
Pedro (from Chile)

lasofre, 2019-01-18:
Hi i haves this error 
warning:'packed' attribute ignored [-Wattributes]}settings_t __attribute__ ((packed));  
what can a i do ?

Duality4Y, 2019-01-18:
I might have put the packed attribute in the wrong place :)

doditsuprianto, 2020-08-13:
in August 2020, your code very useful. this solves my problem.
thanks, Duality4Y

Shumatic, 2023-06-20:
> ```c++
> EEPROM.begin(size * 2);
> ```

I knw it is anold post... but
Why do you multiply size with 2?