esp8266/Arduino #1540: unwanted reset after calling an URL

p1337, 2016-01-30:
Hello ESP-Ardu community ! 
I´m working with the ESP 12-e for a while - after connecting the accesspoint as client the ESP resets:

```
void trigger(){                  // TRIGGER START
  
Serial.println("Trigger set");
nrloops = 0;
Serial.println("Reset nrloops");
 
   WiFiClient client;
      const int httpPort = 80;
  Serial.println(host);

      if (!client.connect(host, httpPort)) {
        delay(100);
        Serial.println("connection failed - Status");
        return;
      }

String url = "/gp/gpControl/status";
```
And the error: 
```
Exception (0):
epc1=0x40201730 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff01f0 end: 3fff0650 offset: 01a0

>>>stack>>>
3fff0390:  4000050c 00000000 00000000 00000000  
3fff03a0:  400043e1 00000030 00000019 ffffffff  
3fff03b0:  400044ab 3fffc718 3fff04a0 08000000  
3fff03c0:  60000200 08000000 08000000 00000000  
3fff03d0:  0000ffff 00000001 04000002 0007e000  
3fff03e0:  3fff1848 0000007e 3ffee934 00000030  
3fff03f0:  00000000 00000000 00000000 00000000  
3fff0400:  00000000 00000000 00000000 00000000  
3fff0410:  00000000 00000000 00000000 00000000  
3fff0420:  00000000 00000000 00000000 00000000  
3fff0430:  00000000 00000000 00000000 00000000  
3fff0440:  00000000 00000000 00000000 00000000  
3fff0450:  00000000 00000000 00000000 00000000  
3fff0460:  00000000 00000000 00000000 00000000  
3fff0470:  00000000 00000000 00000000 00000000  
3fff0480:  00000000 00000000 00000000 00000000  
3fff0490:  00000000 00000000 00000000 00000000  
3fff04a0:  00000002 4000410f 00000000 00000000  
3fff04b0:  3fffc718 40004a3c 0000007e 3fff1848  
3fff04c0:  3fffc718 401014d0 00000378 00000000  
3fff04d0:  0000007e 402103c3 00000378 0000007f  
3fff04e0:  00001000 40210459 3ffee934 0000007f  
3fff04f0:  0000007d 00000000 0007e000 3ffee934  
3fff0500:  ffffff01 55aa55aa 00000007 0000001c  
3fff0510:  0000001c 0000009e 000000d9 0000007f  
3fff0520:  4021086d 3ffe94a0 00000001 3fff0590  
3fff0530:  00000001 3fff05b0 40210973 3ffe94a0  
3fff0540:  00000001 3fff0590 00000000 3ffee824  
3fff0550:  3ffee9f0 3fff05f1 00000001 402109aa  
3fff0560:  3ffef4f8 00000000 3ffef408 3ffef624  
3fff0570:  3fff05b0 00000007 3ffe84ab 3ffef544  
3fff0580:  40202eec 00000008 3fff06a0 402050f5  
3fff0590:  3ffe8e00 00000200 3fff06a0 402050f5  
3fff05a0:  3fff06a0 00000001 3ffe8e95 40205498  
3fff05b0:  3fff0600 00000000 3ffe84f4 3ffef624  
3fff05c0:  3ffef408 00000001 3fff06a0 402050f5  
3fff05d0:  3ffe8e94 3ffe833c 3fff06a0 402050f5  
3fff05e0:  3ffe84f4 3ffef624 3fff06a0 4020556c  
3fff05f0:  402012e6 00000064 3fff06a0 402055d8  
3fff0600:  3fffdc20 3ffef4f8 3fff06a0 4020248e  
3fff0610:  402012e6 00000064 3fff06a0 402055d8  
3fff0620:  3fffdc20 3ffe833c 3fff06a0 402025b6  
3fff0630:  00000000 00000000 3ffef61c 4020488e  
3fff0640:  00000000 00000000 3ffef630 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
`

what does this mode mean?

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)`

greetins from south germany.

Robert

Links2004, 2016-02-02:
with out a objdump or at least a call stack analyses its hart to say where the error is coming from.
 - install the ESP Exception Decoder tool https://github.com/me-no-dev/EspExceptionDecoder
 - flash the ESP new from the IDE.
 - wait for stack dump
 - c&p the error to analyses 
 - post the result here

[image]

p1337, 2016-02-02:
Thanks for your help! This is a nice tool: 

this is the errorcode:
```
Decoding 26 results
0x402050a9: IPAddress at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/IPAddress.cpp:87
 (inlined by) __static_initialization_and_destruction_0 at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/IPAddress.h:90
 (inlined by) _GLOBAL__sub_I__ZN9IPAddressC2Ev at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/IPAddress.cpp:113
0x40105427: ets_timer_arm_new at ??:?
0x402034f1: WiFiClient::connect(char const*, unsigned short) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp:362
0x402012d0: delay at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring.c:70
0x40205588: Print::println(String const&) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp:135
0x402026e8: trigger() at C:\Program Files (x86)\Arduino/functions.ino:137
0x40205458: Print::write(unsigned char const*, unsigned int) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp:39
0x40106ef2: interrupt_handler at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring_digital.c:156
0x40106e6c: interrupt_handler at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring_digital.c:156
0x4000050c: ?? ??:0
0x40000f68: ?? ??:0
0x40000f58: ?? ??:0
0x4021be55: tcp_input at ??:?
0x40105427: ets_timer_arm_new at ??:?
0x40233370: inet_chksum_pbuf at ??:?
0x4021d986: sys_timeouts_init at ??:?
0x40207aa3: pp_enable_noise_timer at ??:?
0x40207b0c: pp_enable_noise_timer at ??:?
0x401052ce: ets_timer_setfn at ??:?
0x40224c01: ets_timer_handler_isr at ??:?
0x40224c20: ets_timer_handler_isr at ??:?
0x40224c01: ets_timer_handler_isr at ??:?
0x40224c46: ets_timer_handler_isr at ??:?
0x402047f3: loop_task at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp:43
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0

```

Links2004, 2016-02-03:
you nee to add the
```epc1=0x40201730 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000```
line too since 0x40201730 is the address of the last executed command.

p1337, 2016-02-03:
This is the actual result:

`
Exception (9):
epc1=0x401052a6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys 
sp: 3ffffc10 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdb0:  0905050a 00000008 3fff048c 402050d9  
3ffffdc0:  c0033035 40105427 3ffffe00 402034f1  
3ffffdd0:  00000000 00000001 402012d0 00000064  
3ffffde0:  3ffef36c feefef00 3fff048c 402055b4  
3ffffdf0:  feefeffe 3ffe8370 3fff048c 402026e8  
3ffffe00:  3ffe8ac8 00000000 000003e8 feefeffe  
3ffffe10:  00000000 00000000 feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe 04080000 001c110c 01010000  
3ffffe50:  3fff048c 00000003 3fff0400 00000001  
3ffffe60:  3ffe84e5 40205458 3fff0400 00000001  
3ffffe70:  c0033035 0000000e 00000000 40106ef2  
3ffffe80:  ffffffff 00000020 00000000 4000050c  
3ffffe90:  00000000 00000000 0000001f 00000022  
3ffffea0:  3fffc200 40106e6c 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 0000001b ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  0021deda 00000001 00003a98 00000000  
3ffffee0:  ffffffff 3fffc6fc 00000001 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffeea68 00000030  
3fffff00:  fdb2d531 d7a81518 d98a861f 21a05724  
3fffff10:  a271a285 ffffffff ffffffff 4021be7d  
3fffff20:  3ffe8fe4 40105427 402333a0 4021d9ae  
3fffff30:  00000000 00000001 40207acb 3ffe8fe4  
3fffff40:  3ffeea40 3ffe8fe0 3ffeea40 00000001  
3fffff50:  40207b34 3ffe8fe4 3ffeea40 006c2091  
3fffff60:  401052ce 40224c29 3ffeea18 3ffeea40  
3fffff70:  40224c48 60000600 ffffffff 40224c29  
3fffff80:  40224c6e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeea68 3fffdad0 3ffef410 402047f3  
3fffffa0:  40000f49 48374cf5 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
--------------------------------------------------------------------------------------------------
Decoding 29 results
0x401052a6: ets_timer_setfn at ??:?
0x402050d9: Print::write(char const*) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.h:60
0x40105427: ets_timer_arm_new at ??:?
0x402034f1: WiFiClient::connect(char const*, unsigned short) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp:362
0x402012d0: delay at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring.c:70
0x402055b4: Print::println(char const*) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp:143
0x402026e8: trigger() at C:\Program Files (x86)\Arduino/functions.ino:137
0x40205458: Print::write(unsigned char const*, unsigned int) at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp:39
0x40106ef2: interrupt_handler at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring_digital.c:156
0x4000050c: ?? ??:0
0x40106e6c: interrupt_handler at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_wiring_digital.c:156
0x4000050c: ?? ??:0
0x40000f68: ?? ??:0
0x40000f58: ?? ??:0
0x4021be7d: tcp_tmr at ??:?
0x40105427: ets_timer_arm_new at ??:?
0x402333a0: inet_chksum_pbuf at ??:?
0x4021d9ae: sys_timeout at ??:?
0x40207acb: pp_enable_noise_timer at ??:?
0x40207b34: pp_noise_test at ??:?
0x401052ce: ets_timer_setfn at ??:?
0x40224c29: ets_timer_handler_isr at ??:?
0x40224c48: ets_timer_handler_isr at ??:?
0x40224c29: ets_timer_handler_isr at ??:?
0x40224c6e: ets_timer_handler_isr at ??:?
0x402047f3: loop_task at C:\Users\p1337\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp:43
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0
0x4010f000: ?? ??:0
`

is this the problem, what does
```

Exception (9):
epc1=0x401052a6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

```

`0x401052a6: ets_timer_setfn at ??:?`

say ?

Links2004, 2016-02-03:
what core version do you use?
at the first look it looks like a NULL ptr usage or flash access problem during interrupt.
do you use any interrupt?

p1337, 2016-02-03:
Hi, thanks for the fast reply - 
I´m using V2.1.0-rc2 
with arduino IDE 1.6.5

i am using two interrupts ... the interrupt for the trigger() funktion is called before reset.

` pinMode(BTN_2, INPUT_PULLUP);
  attachInterrupt( BTN_2, trigger, FALLING ); //TRIGGERBUTTON`

Links2004, 2016-02-03:
try to move your ```trigger``` function to ram.

```cpp
void ICACHE_RAM_ATTR trigger() {

}
```

igrr, 2016-02-03:
Wait, you're using WiFiClient from an interrupt handler? This won't work,
you need to set a flag in the ISR and check the flag in your loop function.

p1337, 2016-02-03:
oh okay thx :) could you help me out how to do this, with the ISR ?

Links2004, 2016-02-03:
```cpp
bool request = false;
void trigger() {
    request = true;
}

void loop() {
   if(request) {
   WiFiClient client;
      const int httpPort = 80;
  Serial.println(host);

      if (!client.connect(host, httpPort)) {
        delay(100);
        Serial.println("connection failed - Status");
        return;
      }

......
        request  = false;
    }
}
```

p1337, 2016-02-03:
okay, i checkt this out

p1337, 2016-02-03:
NO CRASHED ANY MORE!!!!  Thank you ----- i disabled only the interrupt .... asking in the loop for a button state .

`void loopstodeepsleep(int loops){
  Serial.print ("LOOPING-MAIN-LOOP ");
Serial.print (nrloops);
Serial.print (" of ");
Serial.println (loops);

  nrloops++;
   if (digitalRead(BTN_1) == HIGH ){
    trigger();
    }
  delay(500);
  if (nrloops >= loops) {       
    deepsleep();
    }
  }`

closed 2016-02-03 by igrr

dthompso99, 2018-06-26:
Life Saver!  Attempting to use wifi on an esp was causing a similar crash, when used inside a timer.

jasonmanuel13, 2018-07-03:
im also have same problem 

`
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <MFRC522.h>
#include <EEPROM.h>

#define SS_PIN 16
#define RST_PIN 10

MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class

MFRC522::MIFARE_Key key; 

// Init array that will store new NUID 
byte nuidPICC[3];
byte A[4];
byte value;
int address = 999;
int minaddr=1000;
int maxaddr=2000;
int maxcard=1;
int ncard;
boolean match=false;
boolean q=false;

#define relay 5

void bacaRFID();
void regisRFID();
void clearEEP(int minADDR,int maxADDR,int Value);

char x;
int i=0;
char inSerial[10];

void setup() 
{
  //============================================
  //inisialisasi
  Serial.begin(115200);
  SPI.begin(); // Init SPI bus
  rfid.PCD_Init(); // Init MFRC522 
  EEPROM.begin(maxaddr);
  clearEEP(998,maxaddr,0);
  pinMode(relay,OUTPUT);
  //============================================
  //cek EEPROM
  Serial.println(); 
     for ( int j = 0; j < 1+(maxcard*4); j++ )
     {
      value = EEPROM.read(address+j);
      Serial.print(address+j);
      Serial.print("\t");
      Serial.print(value, DEC);
      Serial.println();
      delay(10);
     }
  //============================================
      
  for (byte i = 0; i < 4; i++) 
  {
    //key.keyByte[i] = 0xFF;
    nuidPICC[i]= 0xFF;
  }
//  rfid.PCD_SetAntennaGain(rfid.RxGain_max);
  //===========================================
  //menunggu komunikasi serial tersedia
  while (!Serial){;}
[... 188 lines cut ...]

`Exception (9):
epc1=0x40203e4b epc2=0x00000000 epc3=0x00000000 excvaddr=0x4010ffff depc=0x00000000

ctx: cont 
sp: 3ffef8d0 end: 3ffefac0 offset: 01a0

>>>stack>>>
3ffefa70:  00020004 3ffee84c 3ffeea68 402024d8
3ffefa80:  3ffe84dc 3ffee808 3ffee80c 4020216b
3ffefa90:  3fffdad0 00000000 3ffeea8c 402021f0
3ffefaa0:  3fffdad0 00000000 3ffeea8c 40203c30
3ffefab0:  feefeffe feefeffe 3ffeeaa0 40100710
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld`

can you guys help me ?