esp8266/Arduino #1542: ESp8266 Panic

efuturetoday, 2016-01-31:
Hi, i got some troubles finding my bug in my code.
I have a TLS Server with a 2048bit Key and the ESP8266 connecting to it.
It works and the server send every second a heartbeat to the ESP and the ESP simply send it back to the server. All is running fine i thought until i noticed that the connection is lost after some random time:
The ESP got a panic error:
```
Panic .pioenvs\esp12e\FrameworkArduino\core_esp8266_main.cpp:114 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  4010395e 00000002 00000000 3fff1020  
3fffff90:  00000000 3fffdc20 3fff1018 40201e59  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
```

i compiled the programm with the latest platformIO release.
```
Platform espressif
--------
Updating toolchain-xtensa package:
Versions: Current=2, Latest=2    [Up-to-date]
Updating tool-esptool package:
Versions: Current=4, Latest=4    [Up-to-date]
Updating tool-mkspiffs package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=7, Latest=7    [Up-to-date]
Updating ldscripts package:
Versions: Current=18, Latest=18          [Up-to-date]

Updating [ 64 ] Json library:
Versions: Current=0cfc25d751, Latest=0cfc25d751          [Up-to-date]
```

and finally here is my code on the ESP:
```
#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <WiFiClientSecure.h>
#include <Ticker.h>
#include <ArduinoJson.h>

extern "C" {
    #include "user_interface.h"
}

#define MAJOR_VERSION 0
#define MINOR_VERSION 1
#define BUFF_SIZE 512

const char* ssid = "xxxxxxxxxx";
const char* password = "xxxxxxxx";

const char* host = "192.168.1.156";
const int port = 4444;

// SHA1 fingerprint of the certificate
const char* fingerprint = "XX XX XX XX XX XX XX XX XX XX 79 XX XX XX XX XX XX XX XX XX";

// Use WiFiClientSecure class to create TLS connection
WiFiClientSecure client;
Ticker tickerOSWatch;

void connectBackend() {
  if (client.connected()) {
    return;
  }
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }

  Serial.println("connection success");

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
    client.stop();
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Start");
  });
  
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  
  ArduinoOTA.begin();
  
  Serial.println("OTA Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  connectBackend();
  tickerOSWatch.attach_ms(5000, connectBackend);
}

void safeSend(char *payoad) {
    if (client.connected()) {
       //client.flush(); // i dont care about other messages - bec i can only process one at once
       client.print(payoad);
       yield();
    }
}

void turnOff() {
    digitalWrite(13, LOW);
}

String IP2Str(IPAddress address) {
 return String(address[0]) + "." + 
        String(address[1]) + "." + 
        String(address[2]) + "." + 
        String(address[3]);
}

/*
    JSONBUFFER SIZE ON ESP8266
    JsonVariant	8 <-- Reference either of jsonArray or jsonObject: https://github.com/bblanchon/ArduinoJson/issues/195
    JsonArray of N element	8 + 12 * N
    JsonObject of N element	8 + 16 * N
*/
void processCommand(char* msgBuffer) {
    StaticJsonBuffer<BUFF_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(msgBuffer);
        
    if (!root.success()) {
        Serial.println("parseObject() failed");
        return;
    }
    
    yield();
    const char* cmd = root["cmd"];
    char msg[1024];
    if (strcmp(cmd, "heartBeat")  == 0) {
        root.printTo(msg, sizeof(msg));
        safeSend(msg);
    } else if (strcmp(cmd, "identify")  == 0) {
        StaticJsonBuffer<1024> jsonBuffer;
        JsonObject& json = jsonBuffer.createObject();
        
        //SIZE FOR BASE INFO WITHOUT SERVICES MAY BE ~512 so there 512 jsonBuffer for services array: https://github.com/bblanchon/ArduinoJson/wiki/Memory-model
        json["cmd"] = "identify";
        json["chipId"] = ESP.getChipId();
        json["freeSketchSpace"] = ESP.getFreeSketchSpace();
        
        JsonObject& version = json.createNestedObject("version");
        version["sdk"] = system_get_sdk_version();
        version["major"] = MAJOR_VERSION;
        version["minor"] = MINOR_VERSION;
        
        /*
        enum rst_reason {
            REANSON_DEFAULT_RST = 0, // normal startup by power on
            REANSON_WDT_RST = 1, // hardware watch dog reset
            REANSON_EXCEPTION_RST = 2, // exception reset, GPIO status won’t change
            REANSON_SOFT_WDT_RST = 3, // software watch dog reset, GPIO status won’t change
            REANSON_SOFT_RESTART = 4, // software restart ,system_restart , GPIO status won’t change
            REANSON_DEEP_SLEEP_AWAKE = 5, // wake up from deep-sleep
        };
        */

        rst_info* resetInfo = system_get_rst_info();
        JsonObject& reset = json.createNestedObject("reset");
        reset["reason"] = resetInfo->reason;
        reset["exccause"] = resetInfo->exccause;
        reset["epc1"] = resetInfo->epc1;
        reset["epc2"] = resetInfo->epc2;
        reset["epc3"] = resetInfo->epc3;
        reset["excvaddr"] = resetInfo->excvaddr;
        reset["depc"] = resetInfo->depc;
        
        JsonObject& flash = json.createNestedObject("flash");
        flash["id"] = ESP.getFlashChipId();
        flash["size"] = ESP.getFlashChipSize();
        flash["realSize"] = ESP.getFlashChipRealSize();
        flash["speed"] = ESP.getFlashChipSpeed();
        FlashMode_t ideMode = ESP.getFlashChipMode();
        flash["mode"] = (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN");
        
        const char* phyModes[] = { "", "B", "G", "N" };
        JsonObject& wifi = json.createNestedObject("wifi");
        wifi["channel"] = wifi_get_channel();
        wifi["reconnect"] = wifi_station_get_auto_connect();
        wifi["phyMode"] = phyModes[wifi_get_phy_mode()];
        wifi["ssid"] = WiFi.SSID();
        wifi["bssid"] = WiFi.BSSIDstr();
        wifi["mac"] = WiFi.macAddress();
        wifi["ip"] = IP2Str(WiFi.localIP());
        wifi["mask"] = IP2Str(WiFi.subnetMask());
        wifi["gateway"] = IP2Str(WiFi.gatewayIP());
        wifi["dns"] = IP2Str(WiFi.dnsIP());
        wifi["hostname"] = WiFi.hostname();
        
        JsonArray& services = json.createNestedArray("services");
        for (unsigned int i = 0; i < 10; i++) {
            JsonObject& service = services.createNestedObject();
            service["keyA"] = "AAA";
            service["keyB"] = "BBB";
            service["keyC"] = 111;
            yield();
        }
        
        json.printTo(msg, sizeof(msg));
        safeSend(msg);
    } else if(strcmp(cmd, "restart")  == 0) {
        ESP.restart();
    } else if(strcmp(cmd, "perform")  == 0) {
        
    }
}

//String ICACHE_FLASH_ATTR Stream::readStringUntil(char terminator) { //blockiert max 1 sekunde
char msgBuffer[BUFF_SIZE];
unsigned int pos = 0;
void loop() {
  ArduinoOTA.handle();
  
  //process incoming data
  while (client.available()) {
    char c = client.read();
    if (pos >= BUFF_SIZE) {
        pos = 0;
    } else if (c == '\n') {
        msgBuffer[pos] = '\0';
        client.flush();
        processCommand(msgBuffer);
        pos = 0;
    } else {
        msgBuffer[pos++] = c;
    }
    yield();
  }
}
```

igrr, 2016-01-31:
Do not make network calls from Ticker callback. Instead, set a flag and
check for it in your loop function.

efuturetoday, 2016-01-31:
@igrr thank you for your quick response - i just removed the whole ticker includes and the ticker itself and will test without the reconnect ticker. I will report soon - my test is currently running.

efuturetoday, 2016-01-31:
okay - the same issue after about 785sec the ESP reboots with this message:
```
Panic .pioenvs\esp12e\FrameworkArduino\core_esp8266_main.cpp:114 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  4010395e 00000002 00000000 3fff1020  
3fffff90:  00000000 3fffdc20 3fff1014 40201e59  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
```

if you want i can create a repo for this issue tracking?
i have also a elf-dump if it helps.

igrr, 2016-02-03:
The `panic` line comes from the [stack overflow check](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_main.cpp#L117).

Usual suspects are large objects/arrays allocated on the stack, like `StaticJsonBuffer<BUFF_SIZE> jsonBuffer;`, `char msg[1024];`, `StaticJsonBuffer<1024> jsonBuffer;` (did you really need two `jsonBuffer` variables?)

Please check if you can make these objects global or allocated on the heap.

devyte, 2017-10-18:
This looks like user error, closing.

closed 2017-10-18 by devyte