esp8266/Arduino #1545: Nodemcu Devkit v0.9 not recognized by computer, unable to upload new code.

sokeeffe10, 2016-01-31:
Hi there , I was given the nodemcu devkit v0.9 with code on it already uploaded using Arduino IDE, I have been trying to upload to this again but the Board would appear in device manager but then I would try to upload using arduino IDE and receive , com error. 

I have tried different usb cables, different ports , and now there is simply a blue light on steady when the board is plugged into usb. The user button no longer turns on the small blue light , and my comp does not recognize the board being plugged in our out at all. 

Please help as this has eben very frustrating.

tuxedo0801, 2016-02-02:
Sounds like module is broken?

sokeeffe10, 2016-02-02:
Yeah it was,  I bricked it stupidly,  thanks anyway

tuxedo0801, 2016-02-02:
Issue can be closed?

sokeeffe10, 2016-02-02:
Yes thank you

tuxedo0801, 2016-02-02:
Then close it?

closed 2016-02-02 by Links2004

rolf5649, 2026-02-23:
> Jo, byl, hloupě jsem to zablokoval, každopádně díky.