esp8266/Arduino #1546: How to configure DHCP on ESP8266 AP using Hardware Serial

rwkiii, 2016-02-01:
I have an ESP8266 AP setup on a Mega 2560 that is working. I also have an ESP8266 Client setup that is able to connect to the AP and get a DHCP issued IP address - but the client is unable to access anything except for by IP address. There is no DNS configured.

I wasn't able to use any of the WiFi* libraries, so all of my commands on the AP (using hardware serial)  are made using AT commands. I don't see any way of setting the gateway and DNS for DHCP leases using AT commands. I see there are commands to set these using the WiFi libraries, but how do I configure DHCP using AT commands?

Don't think it matters, but my ESP8266 is connected on the Mega using TX1 and RX1.

AT+GMR
AT version:0.40.0.0(Aug  8 2015 14:45:58)
SDK version:1.3.0
Ai-Thinker Technology Co.,Ltd.
Build:1.3.0.2 Sep 11 2015 11:48:04
OK

Can anyone confirm that I am forced to use AT commands in my sketch since I am using hardware serial? None of the ESP8266-specific libraries will work, of course, and I haven't had any luck using WiFi.h either. I think I've tried all libraries out there! :-S

Thanks.

Links2004, 2016-02-01:
this repo is a full replace for AT firmware.
with the software here you can Programm the ESP module like any other Arduino board.
we not deal with any AT command.

may this docs helps you to understand it:
https://github.com/esp8266/Arduino/blob/master/README.md#arduino-core-for-esp8266-wifi-chip
http://esp8266.github.io/Arduino/versions/2.1.0-rc2/doc/boards.html

closed 2016-02-01 by Links2004

rwkiii, 2016-02-01:
Thank you for the links. 

I didn't want to use AT commands, but all of the Arduino libraries seem to require the presence of a WiFi shield. I'm not sure how to satisfy this requirement using only an ESP8266-01 connected to TX1 and RX1. So, I tried AT commands and at least I now have a functional AP... don't like the AT command approach though... I feel stuck with it.

I will review your links and try to understand better.

Links2004, 2016-02-01:
the ESP8266 itself is a micro controller with integrated WiFi.
you can program it like the ATmega329p or a Mega2560.
the ESP-01 is limited to less pins may take a look to a ESP-12 or nodeMCU (ready to go ESP-12).