esp8266/Arduino #1548: Possibility to debug Heap fragmentation/corruption?

nouser2013, 2016-02-01:
Greetings, is there any way to debug the heap management system? I see random freezes on the IP subsystem and sometimes even reboots when heavy class creation/destruction is done. This is mainly the cause with the regular webserver and @me-no-dev s Async Webserver, and the symptom is that the ESP continues to run with the main loop and all tickers, but I cannot ping it anymore and it will not accept any more IP packets (UDP/TCP tried).

So I was wondering if there is a way to "see" / debug the fragmentation, since getFreeHeap() always returns a sufficient amount (~30k).

Thanks.

Duality4Y, 2016-02-01:
Oh yea that would be nice if such a thing existed :)

me-no-dev, 2016-02-01:
I am running umm_malloc instead of the default heap management. Is there anything I can try to test this theory here?
Also would like to point you to this tool (in case you use the Arduino IDE) which can help tracing the causes of the failures: [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

nouser2013, 2016-02-01:
Unfortuantely, my IP system freezes will not output any error on serial. And the ESP kinda sorta still runs, I can press buttons, the LED keeps blinking and my serial interactive shell works as expected. Other than that, it does not send nor respond to queries made via WiFi/IP.

I also don't have an idea where to begin with debugging...

closed 2016-03-11 by igrr