esp8266/Arduino #1549: Arduino 1.6.8 for Raspberry Pi error

AgreeDK, 2016-02-01:
I know that 1.6.8 is not supported and that I should use 1.6.5 , but I have not been able to get 1.6.5 installed on my Raspberry Pi - and now we can download 1.6.8 rc1 direct to RPi and it works fine with other boards. I have installed the ESP boards, selected Generic ESP, Im able to open the com port and send AT commands to the board, but when I try to compile I get this error:

"fork/exec /home/pi/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++: exec format error
Error compiling."

Hope you can assist me in getting a working version installed on my Raspberry Pi ?

Thx in advance for your support

me-no-dev, 2016-02-01:
it detects your PI as Linux and the wrong toolchain probably.
You can try and compile the toolchain yourself (it's pretty straight forward) hen replace the one in the packet with the one you compiled and should be good to go

AgreeDK, 2016-02-01:
I did not compile 1.6.8 it is downloaded as binary,  so I can not just replace source  code.

me-no-dev, 2016-02-01:
it has nothing to do with Arduino itself.
For you to be able to compile on any OS for ESP8266 you need a toolchain. That is usually automatically downloaded by the package manager. BUT Raspberry Pi is not one of the supported platform and since it's Linux, the package manager downloaded the toolchain for linux.
Search on the web for toolchain for ESP8266 and RasberryPi and replace the one in /home/pi/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc with the one you find (or compile yourself)

devyte, 2017-10-18:
We're now on 1.8.x, closing.

closed 2017-10-18 by devyte