esp8266/Arduino #1550: Same value pointers issue in Arduino

sajjad400w, 2016-02-01:
hi friends.
please excuse me for my bad english.

I think it is a Bug in arduino.
I was involved in almost 12 hors.

`#include <ESP8266WiFi.h>`

`char    *in_password   = "400400400";`
`char    *out_password   = "400400400";`

`void setup() {`
      `uint8_t i;`
      `Serial.begin(115200);`
      
      `Serial.println(in_password);            //  Result:  400400400`
      `Serial.println(out_password );         //  Result:  400400400`
      
     `*out_password = "500500500";`
     
      `Serial.println(out_password );         //  Result:  500500500`

      `// Problem:`
      `Serial.println(in_password);            //  Result:  500500500 !!!!!!!!!!!!`
`}`
       
  
its mean both of pointers are the same value even thouth they are different name.

supersjimmie, 2016-02-01:
AFAIK, you should never modify a char *.
(it's not an array that you can modify, but it's a pointer to somewhere you don't control)

d-anders, 2016-02-02:
You probably mean    out_password="500500500";

igrr, 2016-02-03:
Modifying `const char*` is an undefined behaviour. If you enable compiler warnings (Arduino Preferences > Warnings > "More") you will see compiler complaining about that line.

closed 2016-02-03 by igrr