esp8266/Arduino #1554: webserver stops responding during sent

ba-us, 2016-02-02:
I have a strange issue here that I have trouble searching for information on the net.
I found some similar issues but not exactly what I got here...

I am using Arduino 1.6.5 and the latest Arduino-master downloaded today.
The sketch is getting data over UART and outputing a html-table with this data.
As the table is quite big, it is streamed out in bits of about 2k.

During the command "pclient->print(sTable);" the server stops sending data and the table is not updated in the browser.
A new request from the browser works fine. (With an older version of Ardiuno-master the device did reset with wdt_reset.)

debug output on a failing transmission is:

> new client
> GET / HTTP/1.1
> :wr
> :sent 86
> :ww
> :wr
> :sent 576
> :ww
> :wr
> :sent 1460
> :sent 673
> :ww
> :wr
> :sent 1460
> :sent 667
> :ww
> :wr
> :er -8 2108 1
> :ww
> :wr !_pcb
> :wr !_pcb
> :wr !_pcb
> :ur 1
> WS:dis
> :del
> Client disconnected
> ip:192.168.0.100,mask:255.255.255.0,gw:192.168.0.254
>                                                     wifi evt: 3

It happens in different positions of the transmission and after different time/number of request from the browser.
Transmitting more data seems to make the error appear sooner, but I cannot confirm that for sure.

What's more, I get this debug output

> ip:192.168.0.100,mask:255.255.255.0,gw:192.168.0.254
>                                                     wifi evt: 3

every few minutes, even with the ESP with no activity UART or WiFi. It's basicallly idling.

I hope that I described this behaviour in an intelligible manner...

best regards,
Bob

igrr, 2016-02-03:
Hi, could you please share the minimal example which leads to this issue?

ba-us, 2016-02-04:
Of course.

I threw out the whole UART thing and stripped it down to just creating one html table that auto-loads every two seconds.

Took a few minutes but now it hangs. Requesting the page again, and all is fine (until it hangs for the next time...)

Last debug output:
> WS:ac
> :rn 318
> :ref 1
> WS:av
> :ref 2
> :ur 2
> :wr
> :sent 85
> :ww
> :wr
> :er -8 1349 1
> :ww
> :wr !_pcb
> :ur 1
> WS:dis
> :del

Here's the code, I hope it's ok to post directly...

```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

// WiFi connection
char Receive_ssid[33] = "TP-LINK_791EC4";
char Receive_password[65] = "C2791EC4";

unsigned long ulReqcount;       // how often has a valid page been requested
unsigned long ulReconncount;    // how often did we connect to WiFi

int WiFiStatus = WL_IDLE_STATUS;

// storage for Measurements; keep some mem free; allocate remainder
#define KEEP_MEM_FREE 10240

// Create an instance of the server on Port 80
WiFiServer server(80);

// needed to avoid link error on ram check
extern "C"
{  
  #include "user_interface.h"
}

/////////////////////
// the setup routine
/////////////////////
void setup()
{
  // setup globals
  ulReqcount = 0;
  ulReconncount = 0;

  // start serial
  Serial.begin(115200);
  Serial.println("  ");
  Serial.println("Webserver-Test");
  
  WiFi.mode(WIFI_STA);
  WiFiStart(Receive_ssid, Receive_password);
  
  Serial1.begin(115200);
  Serial.setDebugOutput(true);
[... 263 lines cut ...]
    sResponse2 = "";
  }

  else
    ////////////////////////////
    // 404 for non-matching path
    ////////////////////////////
  {
    sResponse = "<html><head><title>404 Not Found</title></head><body><h1>Not Found</h1><p>The requested URL was not found on this server.</p></body></html>";

    sHeader  = F("HTTP/1.1 404 Not found\r\nContent-Length: ");
    sHeader += sResponse.length();
    sHeader += F("\r\nContent-Type: text/html\r\nConnection: close\r\n\r\n");

    // Send the response to the client
    client.print(sHeader);
    client.print(sResponse);
  }

  // and stop the client
  client.stop();
  Serial.println("Client disconnected");
  delay(50);
}
```

ba-us, 2016-02-08:
could anyone recreate this behaviour or could it be my hardware setup?

ba-us, 2016-02-17:
Got the same issue when using  ESP8266WebServer.h
and the corresponding WebServer.send() command

Requests from the browser are processed correctly, but after a while sending just stops.

debug output:
> URL is: /
> HTTP Method on request was: GET
> Number of query properties: 0
> :wr
> :sent 117
> :ww
> :wr
> :sent 1460
> :ww
> :wr
> :sent 1460
> :ww
> :wr
> :er -8 1460 1
> :ww
> :ur 1
> WS:dis
> :del

ba-us, 2016-02-17:
I would even be happy with a workaround...

ba-us, 2016-02-17:
I found that it has to do with tcp_output( _pcb ); in ClientContext.h

But I don't know how I could debug deeper...
I also tried to enable LWIP debugging but didn't succeed.
Any ideas?
I think I lack some basic knowledge about Arduino and how libraries are included...

mcr-ksh, 2017-05-16:
+1. got a !_pcb during send in a TCP connection.

devyte, 2017-10-18:
@d-a-v is this worth testing with lwip2?

d-a-v, 2017-10-18:
I think this was solved by PR #2874 (issue #2871) and also described in #2925.
The truth is I can't reproduce the issue with lwip1.4 (on master) with the sketch above, much bigger data, 300 iterations and still running.

It's nice to have sketches to try :)

devyte, 2017-10-18:
I figured, that's partly why I'm throwing them your way :)
Closing as can't reproduce.

closed 2017-10-18 by devyte