esp8266/Arduino #1557: Minor warning under eclipse IDE (esp8266 core version 2.0)

far5893, 2016-02-02:
Description	Resource	Path	Location	Type
Class 'WiFiUDP' has virtual method 'stop' but non-virtual destructor	WiFiUdp.h	/esp8266_rgbwebstrip/Libraries/ESP8266WiFi/src	line 40	Code Analysis Problem

Description	Resource	Path	Location	Type
No return, in function returning non-void	libc_replacements.c	/esp8266_rgbwebstrip/arduino/core	line 141	Code Analysis Problem

devyte, 2017-10-18:
This is still valid in latest git. If there are virtual methods, then the destructor should also be virtual. If the class shouldn't allow inheritance, then the destructor should be protected.
A quick search in the repo shows that currently no class inherits from WiFiUdp.
@igrr I'm inclined towards removing all virtual decorators. What do you think?

devyte, 2017-10-18:
A quick repo search shows other classes with the same issue. I'd say we need to revisit all classes with virtual methods.

igrr, 2017-10-22:
Many methods of WiFiUDP are virtual, because it inherits from UDP -> Stream -> Print.
To make things more obvious I would change declarations from

```
virtual void foo();
```
to
```
void foo() override;
```
in WiFiUDP. And also make the destructor virtual, of course.

earlephilhower, 2019-01-16:
Anyone want to volunteer to do a pass through the code for this?  It's a good note.

liebman, 2019-01-17:
If there is no rush I can take a look.

d-a-v, 2019-01-19:
Fixed by #5637, thanks!

closed 2019-01-19 by d-a-v