esp8266/Arduino #1559: share same flash string with printf, printf_P and other _P calls

hallard, 2016-02-03:
Sorry guys, I've got a problem with Flash string syntax, it won't stay in my head and give me some headaches each time, so once forever, I'm trying to optimize Flash to avoid string duplication. 
So what I'm trying to do is to declare Flash String "globally" like that but I want to use it with _P function as classic such as printf.

This sketch seems to work, but is it the best way? I'm not sure with 4 bytes flash alignment and flash storage. Any confirmation would be helpful!

```arduino
// Declare this string once to be used multiple time with one storage in flash
const char flash_str[] PROGMEM = "test %d";

void setup() {
	Serial.begin(115200);
}

void loop() {
	static uint16_t cpt = 0;
	char buff[32];

  // This one works, use FPSTR for 4 bytes alignment
  // see https://github.com/esp8266/Arduino/issues/1143
  Serial.println( FPSTR(flash_str) );

  // Use stored Flash STR, what about byte aligment ?
  sprintf_P( buff, flash_str, cpt++ );
  Serial.println(buff);

  // Use hard coding Flash STR, what about byte alignment?
  // FPSTR() nor F() does compile, only PSTR
  sprintf_P( buff, PSTR("test %d"), cpt++ );
  Serial.println(buff);

  delay(1000);
}
```

Makuna, 2016-03-09:
asking this on the gitter channel might have been a better approach than an issue, as you haven't listed a bug or a problem.

If you are following the Arduino guidelines for PROGMEM and F(), then you shouldn't have to worry about alignment, all the fun macros/functions hide the details of it from you.

In your example:  
where is 'str' defined?  
you will have two duplicate strings in flash, from these two lines...
```
const char flash_str[] PROGMEM = "test %d";
```
and
```
sprintf_P( buff, PSTR("test %d"), cpt++ );
```
but I suspect you knew this as you coded the line to use the original in the first place.

devyte, 2017-10-18:
Non-issue, closing.

closed 2017-10-18 by devyte