esp8266/Arduino #1562: PWM doesn't working

qblacksheep, 2016-02-03:
Hi,

Links2004, 2016-02-03:
the WebSockets example for RGB LED dimming, is working for me,
but i not use a Mosfet its a single RGB LED.
what version of the core do you use?
there where some bug in the PWM handling, its fixed in 2.1.0-rc2

qblacksheep, 2016-02-03:
Thanks, in this version PWM its working. But my leds still shine decidedly to weak in the full cycle (1023). Maybe ESP output gives me too low current? I have ~42mA in the full cycle, from arduino i have ~60mA

Links2004, 2016-02-03:
the Arduino runs with 5V the ESP with 3.3V.
so the lower current looks in the expected range of change to me.
how you wiring with the mos looks like?

qblacksheep, 2016-02-03:
But i measure the current from arduino when i had ~3.3V on the output.
[image]

i am using P16NF0b

Links2004, 2016-02-03:
your R9 is on the wrong place.
try:
[image]

qblacksheep, 2016-02-03:
Yes, these schematic was wrong, i uploaded a new. Sorry

Links2004, 2016-02-03:
no problem,

have take a look to the datasheet of the MOS.
the Threshold Voltage (Vgs) is not optimal for the ESP.
the MOS needs up to 4V to work properly. 

[image]

qblacksheep, 2016-02-03:
Ah, so MOS will be full opened when Vgs > 4V? You suggest to find new mosfet 
which will be full open at Vgs> 1-2,5V or try to add amplifiler at the output and get about ~5v from 3,3v

Links2004, 2016-02-03:
AO3400 is a good chose for the ESP, can handle 4.5A and up to 30V and its small.
yes the MOS open fully in between from 2V to 4V depending on manufacturing process.

qblacksheep, 2016-02-03:
Thanks! I will buy it tomorrow and test. If it works, ill write  there. Thanks again:)

qblacksheep, 2016-02-10:
Yea, its working. Thanks for help. Ill close it.

closed 2016-02-10 by qblacksheep

vargham, 2016-02-27:
> yes the MOS open fully in between from 2V to 4V depending on manufacturing process.
No, the opposite is true. The MOSFET will be fully closed somewhere between these values, depending on manufacturing process.
You have to check other part of the datasheet: RDS(ON) It is usually a sub-table, with specific Vgs values, resistance and maximum Id (Drain current). You will see, lower gate voltage means lower maximum drain current.
Logic level MOSFETs usually will be fully close between 2 and 4 V and will be fully opened at 4.5 or 5 V, so the full opening will be somewhere near 4.5 - 5 V, not between 2 and 4. You heave to be sure, so use voltage values given in RDS(ON).

Source:
[How to Read a Power MOSFET Datasheet](http://www.embeddedrelated.com/showarticle/809.php)

vargham, 2016-02-27:
> AO3400 is a good chose for the ESP

Yes, this is a good choice.
It handles 5.8A at Vgs = 10 V, and 4A at Vgs = 2.5 V

[AO3400 datasheet](http://www.aosmd.com/pdfs/datasheet/ao3400.pdf)