esp8266/Arduino #1565: Wiring Connections from teensy 3.2 to Esp8266 not working or something with script wrong?

Tejkaran, 2016-02-03:
Hi All

First of all, I apologise if this is not meant to be here. I was not sure where else to go, so if this is in the wrong place, could you redirect me please. Thank you.

I am not a an expert with circuit boards etc but I feel comfortable doing them. However I am having trouble connecting, or at least getting some sort of response from the ESP8266.
My wiring goes as such:
VCC & CH_PD are connected to 3.3V (driven by a 3.3V 1.5A regulator supplied by 5V 60A power supply)
GND, GP015 & GP100 all to the ground (on the power supply)
TXD goes to Pin 0 on the Teensy 3.2 (RX1)
RXD goes to Pin 1 on the Teensy 3.2 (TXD)

I run the Arduino IDE with this script on this page -> half way down (https://kaspars.net/blog/electronics/esp8266-teensy-arduino) and nothing comes up on my serial monitor. I have it set at 115,200 baud and have messed around with different settings too. In addition I have both NL and CR as setting too. I am using Port COM4 (that is the only option). When I type in AT in the serial monitor nothing happens at all.

When I use this code (https://forum.pjrc.com/threads/27850-A-Guide-To-Using-ESP8266-With-TEENSY-3), the first part of it. It works as suggested with the LEDs turning on/off very quickly if I type anything into the Serial Monitor. However if I try to type any code such as AT, AT + RST etc. into it to get something proper then nothing happens,
My question is, what do I need to do now to get the ESp8266 up and working? I think I am nearly there, maybe just a novice error somewhere?

Any help would be appreciated

Thanks

Tej

drmpf, 2016-02-04:
You could have a look at the bottom half of this page
http://www.forward.com.au/pfod/ESP8266/index.html
But as it says there using the AT command set is not recommended.

Suggest you just configure the ESP8266 as a wifi shield, as described here
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html

Tejkaran, 2016-02-04:
Thanks for the help drmpf
I have the ESP8266-12 module. i have done as is said on the second link (mostly). so for example i have not used an Uno as I have a Teensy. however to the best of my knowledge the rest is the same, mainly the wiring is the same. I do not have any ground wires running directly from the Teensy to the ESP8266, does this matter? If I should where should I link them?
That said. when I use the Arduino IDE it comes up with the error message:
error: Failed to open COM5
error: espcomm_open failed

I do not understand why this is happening. Also worth mentioning, in the code (http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/ESP8266_WifiShield.ino.txt) I changed the password as suggested with the java tool. No problem.  However I am not sure I have the correct pin number as I am using a ESP-12. I looked around on the internet and it seems that pin 4 is the correct number, though I do not say this with a certianty, and I therefore used 4(after trying 2 with this error message, also tried 11,12&15). Do you know any better? 
Im really struggling here :(

drmpf, 2016-02-05:
error: Failed to open COM5
error: espcomm_open failed

For programming the Teensy?
try unplugging the USB and pluging it in again to prod the computer to find the port.

here are my notes from a while back on programming Teensy
```
Run serial_install.exe to add inf for serial drivers

Run teensydunio.exe
Version 1.23 supports 1.0.6 (do not use 1.0.6), 1.6.1, 1.6.3, 1.6.4 (do not use 1.6.2 buggy)
to add support to arduino
this checks in serial driver (above) installed

after install to Arduino 1.6.3 or 1.6.4
choose board Teensy LC
and USB type Serial

To use first time (per board??)
just load a program e.g. blink and then compile
as per HowToUseImage
then after compile
a popup shows
FirstTimeImage

then click press the reset button on the board and see hardware found and installed and code uploaded and run

Serial USB uses Serial
for hardware serial use Serial1, 2 or 3
```
You definitely need a GND connection between the two boards.  This the return for the TX and RX lines. Just connect the two -ve supply lines together.

Nothing special about Pin 2 you can use any free GPIO input to detect if you should go into config mode.  Check out this issue https://github.com/esp8266/Arduino/issues/437
OR just hard code the ssid, password and IP:port and skip the config altogether to get up and running.

Tejkaran, 2016-02-05:
Hey drmpf
I am convinced that my wiring is fine now. I don't think anything is wrong there. In addition, the blue light is lit up constantly on the Esp8266 when i supply power, which it has not done so before.
I am less convinced with the code, as in I am doing something wrong.
It still says failed to open COM5. I did a quick test of the blink app on the teensy and that worked fine, and then i started doing some more slightly more difficult stuff but it still worked fine. For the code you provided I am uploading to the ESP right, and not to the teensy. Because in the IDE i select the ESP8266 generic one (even though i have a 12)
To hard code, how would I do that? A further question what is the purpose/difference between the method in the link you provided and hard coding it?
Thanks a tonne :)

drmpf, 2016-02-05:
OK, from what I understand your Teensy is working. 
It is the ESP that is giving you problems?  
Yes my code is for the ESP 

Try programming a simple blink example on the ESP and get that working first, then try programming my code. 

I have an ESP-01 and an OMILEX board and the code works on both so I don't think there will be a problem with ESP-12,  If you have an ESP-12 select that in the IDE.

Tejkaran, 2016-02-06:
Yeah it is the ESP that is giving me the problems.
The progress that I have made is that the ESP has the blue light on at all times. Suggests to me that it is ready to be used.
In the IDE I have tried to do blink but failed, and it is down to the same issue of:
error: Failed to open COM5 (I did try taking the cable out and putting it back in too)
error: espcomm_open failed

I tried it with the setting of Board:
-Generic ESP8266 module
-Node MCU 0.9 (ESP 12-12 Module) 
but neither did anything
I also changed my USB cable connecting to the Teensy thinking that possibly something could be up with that, but that was hopeful thinking more than anything else.
I used a new ESP and a new Teensy - i have also ordered a ESP -1 so that it may be more simple?

Regarding firmware/software downloads is there something I should be doing? Because at the moment the only software I am using is the Arduino IDE. I havent downloaded anything specific for the ESP (other than the ESP board for the IDE)
The Teensy is working fine, i am sure of that. But when I do the most simple of sketches on the ESP it isnt working because of the Port issue (which does not occur with thwe Teensy) Also, just to get my head around this, for the ESP to work, it is feeding information to the ESP via the Teensy module, right?

any suggestions on what I should try or do next?

thanks

I hope this all makes sense :/

drmpf, 2016-02-06:
OK lets assume you have your ESP and Teensy wired up as shown here

[image]

Note the protection resistors, R1 and R2,  They prevent the Teensy / ESP shorting each other out by mistakes in programming.

Then you need to program the ESP.  First you need to program the Teensy with an empty sketch so the Teensy is not driving D0/D1 otherwise it will upset the ESP programming.

Then to program the ESP you need a USB to 3v3 serial cable,  I used an OMILEX  USB serial cable F  [https://www.olimex.com/Products/Components/Cables/USB-Serial-Cable/USB-Serial-Cable-F/](url)
but you can use any USB to 3V3 serial cable.  Just be sure the TX/RX connections are 3V3.
other cables you can use are [https://www.sparkfun.com/products/12977](url) and [http://www.adafruit.com/products/954](url)
**Note: those two cables have USB power leads as well but those leads are +5V and should NOT be connected to the ESP VCC which only accepts 3V3**

Then to program, turn the power off to the board, short GPIO0 to gnd using the link, apply power and program the ESP via the IDE.  Then remove the GPIO0 link.

See if you can get the ESP blink sketch running this way to start with.

Tejkaran, 2016-02-09:
Quick question, what is/are JP1 and JP3?
I purchased the OMILEX USB serial cable F so I am using that now.

drmpf, 2016-02-09:
In my circuit above JP1 is the link you short, before powering up, to get the ESP into Programming mode
JP3 is to put the ESP into WiFi AP config mode on start up.

_Actually that circuit is incorrect JP3 should connect **GPIO2 to GPIO0  NOT GPIO2 to GND**_

Then in the startup make GPIO2 output low and check the input on GPIO0. If it is low then start AP config.    see http://www.forward.com.au/pfod/ESP8266/GPIOpins/ESP8266_01_pin_magic.html
for more details

drmpf, 2016-02-09:
revised previous post.

Tejkaran, 2016-02-12:
hey drmpf

i have had significant prgoress (I feel), and this is how it stands.
Teensy is doing nothing, and has no program on it.
The ESP is wired up as you wanted however, for some reason it was not working. the reason for this was that it kept on saying' could not COM(X)'. Installed software for the OMILEX USB serial cable F and that was sorted.
Now it says:
warning: espcomm_sync failed
error: espcomm_open failed

To solve this, from the other threads (https://github.com/esp8266/Arduino/issues/466) on this forum it says to connect GPI00 to ground, which I am doing and it should be fine. IE 
Reset (which I do by removing power) and ensure that GPI00 is connected to ground, then reattach power. And it should be good to upload, however this is not happeneing. 
There is mention of GPI02, t attach that to 3V3, I do that and it makes no difference.

Am I doing the above correctly?
I know I am sooo close now! driving me nuts, lol..
What do you think I need to do to overcome the sync/open fails?

FYi I found this really useful (https://www.youtube.com/watch?v=8J7zflVO8K0). very dumbed down for simpletons like me :) plus the guy was using the module 12 version which was nice to compare with.

Many, many thanks

Tej

drmpf, 2016-02-12:
Hi, I think you should take a step back and remove the teensy altogether and just get the programming of the ESP running reliably.

Do a clean install the Arduino V1.6.5 IDE, and then the ESP addon  
(see https://github.com/esp8266/Arduino/)

Checkout the wiring shown on this page
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#adafruit-huzzah-esp8266-esp-12
and wire up you esp12 board

You can use your 3V3 supply board to power the ESP.
Then try and get a simple blink example loaded and running in the ESP

Tejkaran, 2016-02-16:
Hey drmpf

So i followed those steps, did a clean install of both and there has been a marginal improvement, though I am not sure if this is because i re-wired everything (this is what it looks like now(https://imageshack.com/i/pnEcH81lj), or because of the reinstall. I now have some gibberish coming up on the IDE screen (https://imageshack.com/i/pnNRgfs8p) when i type anything, which has not happened before. I have read around and I assume this is to do with a lack of current? i am currently using a 1.5A regulator so this sounds surprising to me, because to my knowledge the ESP only needs 300mA. In the meanwhile I have ordered a 3A 3V3 regulator to see if this is really the problem. Do you think power is the problem? 
*On the image, the VCC, CH_PD and RESET are on the left and connectedf to the positive rail.
GND/GPI015/GPI00 are connected to the ground rail
GPI02 is connected to nothing, as it stands, but will be for the LED.
From the USB cable, which is the one you suggested (3V3) the ground to groun, TX to RX and RX to TX.

When i (finally) get to the point of uploading blink to the ESP, I assume it would be a case of attaching the positive leg of the led to the GPI02 pin and the negative to the ground rail. Also , I am aware I have not used the resistors just yet. I will use them once I get things going.

drmpf, 2016-02-16:
Actually I normally pullup GPIO2 with a resistor also.
When using the OLIMEX USB cable you need to connect GND to GND, TX to RX and RX to TX and add a 3V3 supply   Add a 10uF 16V or higher cap and a 0.1 ceramic cap  between the 3V3 ESP pin and ground pin as close to the esp as possible to reduce the possibiliity of supply problems.

To program with the OLIMEX cable, power down and then connect GPIO0 to ground via a link (but leave the pullup resistor in place) then power up, then program.  Do it again if it does not work the first time.
If it still does not work try unplugging the USB cable from the PC and plug it in again.

You will always get some output after programming due to ESP debug
See this post http://www.esp8266.com/viewtopic.php?f=33&t=2308

You don't need to add a led.  The ESP-12 has a built in green led that will flash when you program the ESP2866 blink example that is added to the IDE when you install ESP support.

Tejkaran, 2016-02-16:
Yuo said:
_When using the OLIMEX USB cable you need to connect GND to GND, TX to RX and RX to TX _
I am fine with this bit

_and add a 3V3 supply _
where do i add this 3V3 power too? Do you mean add it to the USB somehow? I am confused here?

_Add a 10uF 16V or higher cap and a 0.1 ceramic cap between the 3V3 ESP pin and ground pin as close to the esp as possible to reduce the possibiliity of supply problems._
done this, thanks

Other than that one exception regarding power, i have done all the above and the same error message keeps coming up:
warning: espcomm_sync failed
error: espcomm_open failed
....

drmpf, 2016-02-17:
The OLIMEX USB cable does not supply power so you need to power your ESP from something else.
You cannot use a USB supply directly since it is 5v.  I understood you had a 3V3 supply board available.

When I get that message I try
i) power cycle the ESP again with GPIO0 grounded  (I try this a few times, allowing time for the power capacitors to discharge completely)
ii) unplug and plug the USB cable into the PC usb socket (so the computer detects it again), check the COM port has been found is ticked
iii) close the IDE, unplug the USB cable, power down the ESP and then start the IDE, plug in the USB, power up the ESP and check the COM port has been found is ticked.

AND/OR  try swapping the Tx, Rx cables over incase you got them wrong.

drmpf, 2016-02-17:
Also check out this link
https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/discuss

Tejkaran, 2016-02-18:
Hey drmpf, I got it working on blink. Yay! Haha, this took so long
Form my understanding, once again it is hard to pinpoint what was wrong, but I kept on getting th sync error message, So I purchased a new ESP and rewired in exactly the same way, it now works.
Initially i did get this message where there were loads of '..................' and then it just said fail.
So i did a power cycle, re connected USB and open/close IDE as you said, as well as the reset and it worked.

Next step WiFi shield...lets hope this is easier!

Tejkaran, 2016-02-18:
And thank you ever so much for your help!
One question I do have, the wiring coming out from the USB seems to be really fragile. Any suggestions on how to 'strengthen' them?

Tejkaran, 2016-02-18:
I am probably a little bit too happy but I have managed to do the hello world as a web server. Really happy to finally be making some progress..
Any suggestions on what I should do next?

drmpf, 2016-02-18:
The world is you oyster! 
 Data logging, monitoring, control, you name it. 
 www.pfod.com.au has a number of diverse projects,  www.instructables.com has a lot.

Tejkaran, 2016-02-18:
I will definitely look into the pfod app....just out of interest is that something you have made?
i am going to bed now (its past 0100 in England!)

drmpf, 2016-02-18:
Yes I have been developing pfodApp for some years now

Tejkaran, 2016-03-04:
Hey drmpf,

I have been having some fun with esp recently, however I would like to now take things a bit further, but obviously I have hit a stumbling block.

I have an LED strip connected to my teensy3.2 with a sequence that I would like it to follow.
I would then like it so that I can turn the LED strip on or off for a set period of time from the internet, via the ESP. e.g.I can turn it on for5mins, then turn if off, then turn it on for 1hour etc.

I notice with the pfod app you can do similar stuff, but how would I do this?

thanks :)

Tej

drmpf, 2016-03-04:
Hi Tej,
Try using the (free) pfodDesigner to create a menu item that has a 
pulsed output connected to the pin controlling the power to you Led strip.
The pfodDesigner will let you set any pulse width from 1/10 sec to 49days
If you want two different on pulse times then creating two menu items 
seems easiest
pfodDesigner will then generate the code for the ESP8266 so you can 
drive it from your Android mobile via pfodApp
regards
matthew

Tejkaran, 2016-03-04:
ah right, ok, thanks. and then do i need to use the pFod app to use it?

drmpf, 2016-03-05:
yes

Tejkaran, 2016-03-05:
Ok. I'll get onto it.
Thanks for the help. You (guys) are awesome

Tejkaran, 2016-03-08:
hey drmpf,
could you email me at: ninthtej@gmail.com
just feel i am going wildly off topic here, so would be good to converse elsewhere.
thanks
Tej

closed 2016-06-02 by igrr