esp8266/Arduino #1566: Get web client's IP address and MAC address

EnochHwang, 2016-02-04:
My ESP8266 is setup in AP mode as a web server. When a web client browses to the ESP8266, is there a function that returns the client's IP address and the client's MAC address?

nouser2013, 2016-02-05:
Well, I'm using @me-no-dev 's excellent [AsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP) and his [AsyncWebserver](https://github.com/me-no-dev/ESPAsyncWebServer). In the AsyncClient header file, you'll find the members `IPAddress remoteIP();` and `uint16_t remotePort();`.

So, with this lib you'll get at least the IP address. Not sure about the MAC though. It may not even be exposed through the ESP framework.

igrr, 2016-02-05:
remoteIP and remotePort methods are also available in WiFiClient class. MAC address isn't exposed though.

EnochHwang, 2016-02-05:
Thanks for the info. Currently the remoteIP and remotePort methods are only
listed as available for the UDP class. Should update the doc of their
availability in the WiFiClient class.

closed 2016-03-09 by igrr

Crypter, 2018-01-19:
I'm happy to report that now you can get the MAC address off a device. Take a look here:
https://github.com/Crypter/MAChaser