esp8266/Arduino #1568: Non-persistent behavior with WebSocketServer_LEDcontrol

maccody, 2016-02-04:
I've been getting up to speed with a [SeeedStudio ESP-12E module](http://www.seeedstudio.com/depot/ESP8266-based-WiFi-module-SPI-supported-p-2486.html) and programming it using Arduino when I encountered an odd problem using the [WebSocketServer_LEDcontrol sketch](https://github.com/Links2004/arduinoWebSockets/blob/master/examples/WebSocketServer_LEDcontrol/WebSocketServer_LEDcontrol.ino).  I was able to get the sketch working successfully.  Oddly, though, the strings I used in the line 'WiFiMulti.addAP("SSID", "passpasspass");' did not appear to be used.  Instead the default SSID used by the ESP-12E module was used and no password was needed to attach to the AP.

The greater oddity is that cycling the power on the ESP-12E caused the sketch to hang at the while loop with the conditional 'WiFiMulti.run() != WL_CONNECTED'.  Apparently the conditional is never met.

In order to get the sketch working again, I retraced my previous programming experiments.  I discovered that if I first compiled, uploaded, and ran the [Simple HTTP get webclient test](https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide#connecting-via-wifi) on the Adafruit webpage [Using Arduino IDE](https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide), followed by the WebSocketServer_LEDcontrol sketch, it would then work properly.  Cycling the power again would cause the WebSocketServer_LEDcontrol sketch to fail at the same point.  It appears that there is some non-persistent behavior in the sketch or the libraries used.  Anyone have any ideas what the problem may be?

I'm using Arduino 1.6.7 with ESP8266 support installed directly from Github, rather than through the JSON install process.

Links2004, 2016-02-04:
the WiFiMulti may dont find your WiFi.

i think its linked to this issues: #1355
@igrr fixed it today:
https://github.com/esp8266/Arduino/commit/1d149560beba3b851dce7df364d8576ebaa42a0c

how new is your git version of the core?

maccody, 2016-02-04:
The WebSocketServer_LEDcontrol sketch sets up the ESP-12E as an access point, so it is not looking for a WiFi to connect to.

As a sanity check, I reinstalled Arduino 1.6.7 and the ESP Arduino support via Github between 3:00 and 4:00 UTC on February 4, 2016 (local time between 9:00 PM and 10:00 PM CST February 3, 2016).  Your fixes may have already been put in place by that time.  The issues still persist.

Links2004, 2016-02-04:
the WebSocketServer_LEDcontrol example is designed for connecting to a WiFi, no code for AP mode in there.
do you run the code 1:1 like its on the git?

please enable debug for wifi + core, and post the result here.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

maccody, 2016-02-05:
Below are the results of my tests.  In the first test, I was not selecting one of the listed APs. I thought that the function WiFiMulti.addAP() was creating an AP, not selecting an AP.  In the second test, I set up the SSID and passphrase for my WiFi (2WIRE665).  For some reason, it is not connecting.  The third test has two parts.  The first part uses the Adafruit sketch mentioned in my original post.  This worked properly.  In the second part of the test, the WebSocketServer_LEDcontrol sketch is reloaded.  As can be seen, now it works.  The interesting thing (shown towards the end of the test) is that I was also able to connect through an apparently open Access Point on the ESP8266.  This is part of what caused my confusion about the sketch being an AP as well as a server.  That is a hole that needs to be plugged!

***** START FIRST TEST ******
[SETUP] BOOT WAIT 4...
[SETUP] BOOT WAIT 3...
[SETUP] BOOT WAIT 2...
[SETUP] BOOT WAIT 1...
[WIFI][APlistAdd] add SSID: LightTest
[WIFI] delete old wifi config...
[WIFI] start scan
[WIFI] scan done
[WIFI] 12 networks found
       0: [1][94:10:3E:98:BC:17] sparmage (-87) *
       1: [11][00:1E:C7:B4:57:C1] 2WIRE665 (-64) *
       2: [3][00:1B:5B:30:DD:C9] 2WIRE750 (-87) *
       3: [11][94:C1:50:3C:C2:F1] 2WIRE160 (-93) *
       4: [5][B0:C2:87:C0:7C:17] MUSTANG2 (-90) *
       5: [6][00:21:7C:2C:2C:41] 2WIRE767 (-83) *
       6: [7][00:22:A4:E3:A3:31] 2WIRE751 (-86) *
       7: [7][34:EF:44:E2:DD:B1] 2WIRE961 (-86) *
       8: [7][AC:5D:10:07:83:D9] Pickle (-90) *
       9: [11][90:1A:CA:01:E9:20] Bridges Wifi (-91) *
       10: [9][00:22:A4:80:0B:39] 2WIRE889 (-83) *
       11: [9][90:1A:CA:ED:AE:B0] CatLover (-91) *

[WIFI] no matching wifi found!
[WIFI] delete old wifi config...
[WIFI] start scan
[WIFI] scan done
[WIFI] 14 networks found
       0: [1][94:10:3E:98:BC:17] sparmage (-91) *
       1: [1][00:23:51:DB:8E:71] Thom718 (-92) *
       2: [11][00:1E:C7:B4:57:C1] 2WIRE665 (-64) *
       3: [11][90:1A:CA:01:E9:20] Bridges Wifi (-92) *
       4: [3][00:1B:5B:30:DD:C9] 2WIRE750 (-88) *
       5: [5][B0:C2:87:C0:7C:17] MUSTANG2 (-91) *
       6: [6][00:21:7C:2C:2C:41] 2WIRE767 (-82) *
       7: [7][34:EF:44:E2:DD:B1] 2WIRE961 (-87) *
       8: [7][00:22:A4:E3:A3:31] 2WIRE751 (-86) *
       9: [7][AC:5D:10:07:83:D9] Pickle (-91) *
       10: [9][00:22:A4:80:0B:39] 2WIRE889 (-84) *
       11: [9][90:1A:CA:ED:AE:B0] CatLover (-90) *
       12: [11][94:C1:50:3C:C2:F1] 2WIRE160 (-90) *
       13: [11][64:51:06:72:C7:9D] HP-Print-9D-Officejet 4630 (-91) *

[WIFI] no matching wifi found!
[WIFI] delete old wifi config...
[WIFI] start scan
[WIFI] scan done
[WIFI] 11 networks found
       0: [1][94:10:3E:98:BC:17] sparmage (-88) *
       1: [11][00:1E:C7:B4:57:C1] 2WIRE665 (-64) *
       2: [3][00:1B:5B:30:DD:C9] 2WIRE750 (-86) *
       3: [5][B0:C2:87:C0:7C:17] MUSTANG2 (-91) *
       4: [6][00:21:7C:2C:2C:41] 2WIRE767 (-85) *
       5: [6][00:1D:5A:83:CC:F1] 2WIRE950 (-91) *
       6: [7][00:22:A4:E3:A3:31] 2WIRE751 (-85) *
       7: [7][34:EF:44:E2:DD:B1] 2WIRE961 (-87) *
       8: [9][00:22:A4:80:0B:39] 2WIRE889 (-82) *
       9: [9][90:1A:CA:ED:AE:B0] CatLover (-90) *
       10: [11][90:1A:CA:01:E9:20] Bridges Wifi (-92) *

[WIFI] no matching wifi found!
[WIFI] delete old wifi config...
[WIFI] start scan
[WIFI] scan done
[WIFI] 11 networks found
       0: [1][94:10:3E:98:BC:17] sparmage (-88) *
       1: [3][00:1B:5B:30:DD:C9] 2WIRE750 (-87) *
       2: [5][B0:C2:87:C0:7C:17] MUSTANG2 (-89) *
[... 544 lines cut ...]
:rdi 7, 7
:c0 7, 13
[0] get Text: #a86080
:rn 13
:rd 2, 13, 0
:rdi 13, 2
:rd 4, 13, 2
:rdi 11, 4
:rd 7, 13, 6
:rdi 7, 7
:c0 7, 13
[0] get Text: #a86097
station: 24:0a:64:b9:4b:69 leave, AID = 1
rm 1
wifi evt: 6
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: 24:0a:64:b9:4b:69 join, AID = 1
wifi evt: 5
****** END THIRD TEST - PART 2 **********

maccody, 2016-02-05:
I neglected to mention that there is a minor difference in my code.  The #define for the red LED is 14 rather than 15.  I don't believe that should be a significant difference.