esp8266/Arduino #1569: suddenly my codes are not compiling

shirish47, 2016-02-04:
I have been using this programs from last 2-3 months it used to compile fine till tomorrow...
I have not updated anything in the arduino IDE or libraries.
:( does arduino update it self??

In file included from c:\users\asus-pc\appdata\roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:79:0,
                 from C:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/WiFiClient.h:28,
                 from C:\Users\ASUS-PC\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:32,
                 from ESP_V2.ino:3:
c:\users\asus-pc\appdata\roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional: In static member function 'static void std::_Function_handler<void(_ArgTypes ...), _Functor>::_M_invoke(const std::_Any_data&, _ArgTypes ...) [with _Functor = void (*)(); _ArgTypes = {}]':
c:\users\asus-pc\appdata\roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:2073:7: internal compiler error: Segmentation fault
       }
       ^
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.

shirish47, 2016-02-04:
humm I don't know what happened here ..but restart of my computer solved problem. :(

igrr, 2016-02-04:
You can close the issue if it requires no further attention.

closed 2016-02-04 by shirish47