esp8266/Arduino #1572: Exception (29) at ppCheckTxIdle, what is it?

supersjimmie, 2016-02-04:
I get this exception

```
Exception (29):
epc1=0x4020cbe5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000592 depc=0x00000000
ctx: sys
```

```
4020cbe5:	113426        	beqi	a4, 3, 4020cbfa <ppCheckTxIdle+0x6a>
```
But I have no idea where to start searching why/what?
It occured after client.connect(host, port), during the (first part of) communication.
I sent the request with client.print, or at least tried to (cannot see if the request really got out), but never got past client.available.

Links2004, 2016-02-04:
the ppCheckTxIdle function its from the SDK.
what do you use as power source, may its instable?

supersjimmie, 2016-02-05:
I think my powersource is good enough (using 1A source, several extra capacitors, etc), I am able to run much code with the "2.0.0 Stable" without ptoblems for days long.

devyte, 2017-10-18:
Old issue over a year old. No basic info, no sketch. Closing.

closed 2017-10-18 by devyte