esp8266/Arduino #1575: Sketch upload works fine, SPIFFS Upload fails

salildel, 2016-02-04:
IDE - 1.6.7
ESP8266 core - 2.1.0-rc2
Upload tool - ESP8266FS-0.2.0 (https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md)
IDE boards setting - Generic ESP8266 module
Board - ESP8266 ESP-01

I tried different upload speeds, but 115200 is the only speed where it would at least start the upload (dots indicating progress) but then stops with error.
Does it matter what Flash size I select? I am only trying to upload a file of size 57 bytes. Anyway, I tried all available combinations (with 1 MB and 512 Kb flash). Sketch uploads work fine with 512K(64k SPIFFS) and 1M(512k SPIFFS). ESP.getFlashChipRealSize indicates 1 M flash. How do I know what the real SPIFF size is?

Please help...

[SPIFFS] data   : C:\Users\supagupthan\Documents\Arduino_Code\ESP8266\AP_WebServ\AP_Webserv\data
[SPIFFS] size   : 64
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/tst.html

[SPIFFS] upload : C:\Users\SUPAGU~1\AppData\Local\Temp\buildd892b1a9281902446aab159486bf14a5.spiffs/AP_Webserv.spiffs.bin
[SPIFFS] address: 0xEB000
[SPIFFS] reset  : ck
[SPIFFS] port   : COM18
[SPIFFS] speed  : 115200

warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x08

Uploading 65536 bytes from C:\Users\SUPAGU~1\AppData\Local\Temp\buildd892b1a9281902446aab159486bf14a5.spiffs/AP_Webserv.spiffs.bin to flash at 0x000EB000

........warning: espcomm_send_command: didn't receive command response

warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed

warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04

SPIFFS Upload failed!

nouser2013, 2016-02-05:
Had similar issues, see [here](https://github.com/esp8266/arduino-esp8266fs-plugin/issues/7). Perhaps try the new version of the _plugin_, not just the IDE...

salildel, 2016-02-05:
Thanks for the reply, nouser. I am using 0.2.0. Is there a newer one available?
In your case you were getting comm errors:
```
warning: espcomm_sync failed
error: espcomm_open failed
```

In my case, comm starts, though after showing this warning:

`warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x08`

and then:

`warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed`

Still stuck at this :(

Any help will be appreciated.

nouser2013, 2016-02-05:
> How do I know what the real SPIFF size is?

Well, look on the label of the flash IC, it's the one with 8 legs. May be a bit difficult to read, but type the description in Google, and it will give you a link to the datasheet or even already the full specs on a web page. Look for the size figure (likely given in mega _bit_, not byte) and devide by 8. Voila.

> Does it matter what Flash size I select?

Definitely. the plugin compiles a complete FFS image and uploads the full 512k, even if you just put a single file with 100 bytes in the data folder.

> The dots above indicates transfer has started, and the number of dots vary with each attempt, before it fails with the warning...

Sorry to hear, it may be hardware related, e.g. if the power supply is not stable and the SPI flash does not get enough current to erase a page. Or even a faulty SPI flash module. If you can, try a different module and hook it up to a stabilized PSU with 1 A output minimum. I've seen short spikes of up to 500mA when programming the chip.

salildel, 2016-02-06:
Thanks again Nouser, appreciate your help.
The marking on the 8 legged chip said 25Q80ASSIG - It appears to be a 1M chip. Though, I still can't figure out how to know the SPIFFS size.

So I wanted to eliminate the power supply issues - I got hold of a desktop PC SMPS, and used the 3.3 V out to power the ESP - same error. So it's not likely the power supply. Though now I am loving this new supply - the label on the SMPS box says 3.3v @ 28A! Guess I'll keep using this till some one complains about the broken home desktop :D.

I had a brand new 12-E module lying around. This one turned out to be a bit difficult to wire up - it now looks like a spider with colorful legs on the breadboard. Anyway had the same problems. Sketch upload worked fine, though it started with this warning:
`warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x08`

When I tried sketch data upload, got this error:

`[SPIFFS] data   : C:\Users\....\data
[SPIFFS] size   : 1004
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/tst.html

[SPIFFS] upload : C:\Users\....\AppData\Local\Temp\buildd892b1a9281902446aab159486bf14a5.spiffs/AP_Webserv.spiffs.bin
[SPIFFS] address: 0x300000
[SPIFFS] reset  : ck
[SPIFFS] port   : COM18
[SPIFFS] speed  : 115200

Uploading 1028096 bytes from C:\Users\SUPAGU~1\AppData\Local\Temp\buildd892b1a9281902446aab159486bf14a5.spiffs/AP_Webserv.spiffs.bin to flash at 0x00300000

......warning: espcomm_send_command: didn't receive command response

warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed

warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04

SPIFFS Upload failed!
`

Should I try this on my desktop or another PC?

nouser2013, 2016-02-07:
Alright, this sounds like a bug in the uploader plugin. There's not much we can do I'm afraid.

salildel, 2016-02-07:
Thanks again nouser :).

Just to make sure it was not my laptop or the particular Arduino installation acting up, I booted up my desktop after a long time (yes I had to install back the SMPS I had borrowed from it). It was never used for anything Arduino - so was a kinda clean slate for this testing. I Installed the 1.6.7 Arduino IDE, latest ESP8266 core, and SPIFFS uploader plugin. Also made some changes to the ESP rig up to make sure I am using all the right pull up/pull down resistors for the ESP-12E. Guess what - exact same error. So I guess this is indeed a bug. Will have to wait for the final release of this plugin I guess.

Any one else experiencing this issue? Am I the only one? Is there any other tool (standalone/plugin) I can use to upload files when using the Arduino firmware?

Thanks!

me-no-dev, 2016-02-14:
are you able to upload a sketch using the same settings? The plugin calls the same esptool command that the IDE is calling to upload the sketch. If you have your reset method wrong for instance, the reset to bootloader will not succeed and the result above will show. Try having GPIO tied to GND and try the upload. If success then you had your reset method wrong.

salildel, 2016-02-15:
me-no-dev, Thanks for responding.
Yes, Sketch upload works fine - no problems there. I had tied GPIO (I am guessing you meant GPIO0) tied to ground all the while. I am not quite sure what you meant by the reset method. I had the Reset pin tied to VCC with a resistor.

me-no-dev, 2016-02-15:
there are usually two reset methods "ck" and "nodemcu" depending on the module/board you have.
If sketch uploads fine though, that means that the SPIFFS should also upload fine. It is calling the exact same command that is called to upload the sketch from IDE with the difference that it is going to a different location on the flash. What OS IDE and Core versions are you on? Also how did you install the core? package manager?
I'm asking to try and replicate your environment and see if I can get that error

salildel, 2016-02-15:
Thanks again,
I am using Windows 8, 64 bit. (also tried the same on Windows 7 64bit with the same results).
Arduino IDE - 1.6.7
ESP8266 core - 2.1.0 rc2
Core was installed through Boards manager (http://arduino.esp8266.com/staging/package_esp8266com_index.json)

salildel, 2016-02-22:
me-nodev, were you able to replicate the issue?

me-no-dev, 2016-02-22:
have been preoccupied by family and work matters but have seen some more chatter here: https://github.com/esp8266/arduino-esp8266fs-plugin/issues/12
it will be really interesting to find out what change in what config is causing that. It certainly is not the IDE plugin because it just calls the regular tools based on the settings selected by the user and provided in platform.txt and hardware.txt.
Have you by the way tried to upload the SPIFFS using OTA?

ibaranov-cp, 2016-04-16:
SPIFFS via OTA works fine by the way.
To try: load Simple example, then open and compile FSBrowser, and upload.
Takes about 20 seconds.

brynnbp, 2016-08-05:
For posterity, someone in the link me-no-dev posted above said it worked for them by changing 3M SPIFSS to 1M SPIFSS. This also worked for me. Using OTA wasn't an option because I was having unrelated issues with getting my board to show up under ports.

devyte, 2017-10-18:
I can't reproduce this, the SPIFFS upload works well for 1M or 3M on a variety of ESP12-based devices for me.
Closing as can't reproduce.

closed 2017-10-18 by devyte

JinZhenlin, 2017-10-26:
I still can't get this to work! Why?

Pablo2048, 2017-10-26:
Hard to guess - You didn't write any info about Your configuration/setup/sketch...

eketjall, 2017-11-03:
I also get the same error. Sketch upload works fine but spiffs data upload fails. 
ESP12 module with 4MB Flash
Arduino IDE 1.8.4
ESP8266 core - 2.3.0

Tried diffrent Flash speed, flash mode, Spiffs size, reset method, upload speed but this is the furthest a get:

[SPIFFS] data   : D:\arduino-1.8.4\portable\sketchbook\libraries\ESP8266Audio-master\examples\PlayMP3FromSPIFFS\data
[SPIFFS] size   : 1004
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/pno-cs.mp3

[SPIFFS] upload : C:\Users\eketjall\AppData\Local\Temp\arduino_build_926134/PlayMP3FromSPIFFS.spiffs.bin
[SPIFFS] address: 0x300000
[SPIFFS] reset  : ck
[SPIFFS] port   : COM4
[SPIFFS] speed  : 256000

Uploading 1028096 bytes from C:\Users\eketjall\AppData\Local\Temp\arduino_build_926134/PlayMP3FromSPIFFS.spiffs.bin to flash at 0x00300000

................................................................................ [  7% ]

................................................................................ [ 15% ]

................................................................................ [ 23% ]

................................................................................ [ 31% ]

................................................................................ [ 39% ]

................................................................................ [ 47% ]

................................................................................ [ 55% ]

................................................................................ [ 63% ]

.....................................warning: espcomm_send_command: didn't receive command response

warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed

warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04

error: espcomm_upload_mem failed

SPIFFS Upload failed!

eketjall, 2017-11-03:
After reading https://github.com/esp8266/arduino-esp8266fs-plugin/issues/12#issuecomment-274476072
I tried to use a different USB-TTL converter.

Now it works!!

Both converters are FTDI FT232RL based. (probably clones)

TobiasJiupato, 2017-12-19:
Sketch upload works fine, but SPIFFS Upload fails
The problem occours with FTDI FT232RL, now I'm using a PL2303HX model and works fine!
Another tip is use "OTA Upload"

TobiasJiupato, 2017-12-19:
I had a problem with FTDI FT232RL too, now I'm using a USB-TTL PL2303HX model and works fine!
Another tip is to use "OTA Upload"

Skytrobb, 2018-01-13:
When I have my serial monitor open in arduino it won't let me upload SPIFFS. I closed it and it worked fine. I'm pretty sure the nodeMCU can only handle 1 serial connection at a time so maybe the SPIFFS tries to open up another serial? Just my hypothesis, let me know if this helps.

ondabeach, 2018-01-13:
@Skytrobb that is normal. You must close the monitor to upload SPIFFS.

Skytrobb, 2018-01-13:
@ondabeach and here I was thinking I was clever......

techiear, 2018-02-10:
my nodemcu lolin(esp8266 12 E) was working fine for blynk sketch but when i reprogrammed​it for eeprom write operation it shows the same error (error: didn't receive response ... expected 0x1 and 0x4) and now when i am trying​to upload the blynk sketch again it shows me the same error it is not uploading the normal blynk sketch after i tried to upload the eeprom write function on it. seems like my module is malfunctioning now.

rskandhan, 2018-05-05:
i am using nodemcu and for me the issue got solved after changing flash size to 4M(1M) from 4M(3M)... its available in tools>flash size

mbenitez01, 2018-08-06:
LOL!!
I had used this tool in the past very sucessfully...

Now I'm having the same errors as all listed above..

FORGOT I HAD TO CLOSE SERIAL MONITOR!!!

SOLVED THE PROBLEM...

thEpisode, 2018-11-17:
Jajaja same problem, I left the serial monitor open jaja Thank you guys!