esp8266/Arduino #1577: Tracking down that darn "delayed ACK" (200ms) bug...

nouser2013, 2016-02-05:
Greetings advanced dev's, since I'm working on a Windows machine, I'm really mad about the ESP not sending more than one packet before waiting for an ACK. Here's what I observe with different clients on Windows:

- Small data (200 bytes), webbrowser: if CONTENT_LENGTH matches, connection is closed by instantly browser. Overall TX time: 10-15ms.
- Large data (~8000 bytes), web browser: each completely filled packet is ack'd after 200ms. Overall TX time: ~1s (sic!)
- if it's my own client (python etc.) I can set the timeout accordingly and teardown the connection earlier for one packet, but with more packets same as webbrowser. I don't have any influence.

Then, I tried playing with the options in [`lwipopts.h`](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/include/lwipopts.h). In theory, if the `TCP_SND_BUF` buffer is at least twice the size of `TCP_MSS`, the stack would send exactly this amount of data (because it has it in memory for retransmission) until RX window at the receiver (initially set to ~8k by Windows) is full. So, `1460 * 4 = 5840` (4 being set in the opts file) which fits perfectly into Windows' initial RX window. Long story short: as configured it _should_ work.

Still, the ESP would send only one packet before waiting for an ACK. Then, I decreased `TCP_MSS` in the `lwipopts.h` to `128`, just to see if I can see this in Wireshark. Oh wonder, I cannot:
[image]
Same thing with `TCP_CALCULATE_EFF_SEND_MSS` set to `0` by the way.

I tried to force the values by `#define`ing those directly at the top of my sketch before any includes. Same result.

So, could it be that those limits are hardcoded in the SDK `liblwip.a` library file? Can one of the dev's include lwIP's source in the github release in order to have a) the latest version and be) have it recompiled with new options?

igrr, 2016-02-05:
These defines need to be set when building LWIP, of course. Note that for
SDK 1.5.1 we don't have LwIP source available, as Espressif has released it
only for 1.5.

Links2004, 2016-02-05:
which side has the delay?
windows has a 200ms delay for acks.
see:
https://support.microsoft.com/en-us/kb/823764 (Method 3)
https://support.microsoft.com/en-us/kb/328890

nouser2013, 2016-02-05:
Exactly, I can only circumgo this by sending at least 2 packets. And if TCP_SND_BUF is at least 2*TCP_MSS, the stack has enough RAM to store both full sized packets for retransmission should the ACK from the remote not arrive. And in the supplied `lwipopts.h` the `TCP_SND_BUF` is set to 2*TCP_MSS.

So, in theory, there should be no problem with windows, since lwip is configured to send TCP_SND_BUF data completely (if it fits into Receiver=Windows RX buffer) before waiting for ACK. And Windows, receiving those two (or even more) data packets will send the ACK right away after the second packet.

What I don't know of course is which setting is in the supplied liblwip.a; even if they claim in the source to have TCP_SND_BUF to 2*TCP_MSS, we still cannot be sure. The *.a may be from a different build, may not have been updated or similar.

Does anybodoy have a Windows machine an can confirm the behaviour (packet ping-pong with 200ms delay)? It may be my setup after all.

Links2004, 2016-02-05:
can confirm the windows delay of 200ms.
have changed the registry long time ago.
but the sync TCP that the webserver uses will sure run in this "problem".

nouser2013, 2016-02-07:
Alright, I've tested extensively over the last 2 days.

1. the `liblwip.a` file definitely has a TCP_SND_BUF size of `2 * TCP_MSS = 2 * 1460 =` **`2920`**. I found out by simply outputting the `size_t room` in `clientcontext::write()`.
1. The current write-mechanism definitely does not make use of this buffer size. Instead, the `WiFiClient::write()` only ever writes up to 1460 bytes to `ClientContext::write()`, which, in turn, waits for ACKs and interrupts the `delay(5000)` inside its `write()` from within the `err_t _sent()` callback. So, half of the buffer always remains empty.
1. It doesn't make a difference whether the `tcp_output()` is called. I'm guessing that lwip uses Nagle as well to determine when to send a packet. In theory, if you disable NAGLE, every tcp_write() should result in a TCP packet sent immediately.
1. I experimented with the PSH flag, but it does not seem to influence Windows' delayed ACK behaviour.
1. At least with all webbrowsers on Windows I tested, if they received the expected CONTENT_LENGTH bytes, they'll close the connection, thus forcing windows to send the ACK/FIN packet right away.
1. I still have trouble with small responses: Header goes in one ~250 byte packet, delayed ACK, JSON body goes into 10 byte packet, ACK/FIN. Overall response time: 250ms. If I manually put everything into one single packet (by removing web server and just sending on a simple WiFiServer/WiFiClient implementation), response time drops to **10ms** for the same request!

So, in order to improve performance, I did the following to `ClientContext.h`:
```c
size_t write(const char* data, size_t size) {
	if(!_pcb) {
		DEBUGV(":wr !_pcb\r\n");
		return 0;
	}

	if(size == 0) return 0;

	size_t room;
	// Wait for room in send buffer. Before this makes it to
	// master, definitely add a wait to determin from "outside"
	// how much data can be written!
	while (true) {
		room = tcp_sndbuf(_pcb);
		if (room<size) { delay(1); } else { break; }
	}
	size_t will_send=size;  // = (room < size) ? room : size;
	err_t err = tcp_write(_pcb, data, will_send, TCP_WRITE_FLAG_MORE); // No Copy, set PSH-Flag
	if(err != ERR_OK) {
		DEBUGV(":wr !ERR_OK\r\n");
		return 0;
	}

	_size_sent = will_send;
	DEBUGV(":wr\r\n");
	tcp_output( _pcb );
	// Let LWIP handle the sending completely!
	//_send_waiting = true;
	//delay(5000); // max send timeout
	//_send_waiting = false;
	DEBUGV(":ww\r\n");
	return will_send - _size_sent;
}
```

```c
err_t _sent(tcp_pcb* pcb, uint16_t len) {
  DEBUGV(":sent %d\r\n", len);
  //_size_sent -= len;
  //if(_size_sent == 0 && _send_waiting) esp_schedule();
  return ERR_OK;
}
```

With this, I get the following nice Wireshark for 5k `WiFiClient::write_P()`...
[image]

...and 17 bytes JSON `WiFiClient::write()`:
[image]
Unfortunately, forcing the output with `tcp_output()` superseeds Nagles algorithm, Browser receives http header only and waits for data, windows ACKs after 200ms and Webserver then sends the body. ~230ms response time for 17 bytes.

If you comment out the `tcp_output()`, the 17 bytes are fine (Nagle kicks in):
[image]
Now: ~50ms response time for 17 bytes. As I said, with a custom C string functions based parser I can get this down to 10ms.

But for some reason, the `write_P()` behaves much worse:
[image]

We may think about write paradigm. I say if you put your data to LWIP stack and LWIP accepts the data via `tcp_write()` without returning an error,  its not the application's problem anymore.

The dirty hack would be to still use tcp_output but force the ESP Webserver to break with convention and send header and body in one go. This would seem to do the trick...

igrr, 2016-02-07:
>    2. The current write-mechanism definitely does not make use of this
>    3. It doesn't make a difference whether the tcp_output() is called.
>    4. I experimented with the PSH flag, but it does not seem to influence
>    5. At least with all webbrowsers on Windows I tested, if they received
>    6. I still have trouble with small responses: Header goes in one ~250

Did you check the pull request related to WiFiClient refactoring?
WiFiClient will now sends up to TCP_SND_BUF.

nouser2013, 2016-02-07:
I just did, switched to the branch and pulled "as-is". Better, but not optimal.

5k transcript:
[image]

That the http response header is in one single packet I can understand, but why does Nagle not kick in? Also, look at the last two packets, 1514 and 404 should have been sent in one rush? Do I need to change something in the application source code?

17bytes transscript:
[image]
Again, shouldn't Nagle kick in here?

igrr, 2016-02-07:
Sorry, what do you mean by "Nagle kicks in"? We call tcp_output after each
write, which forces any pending data to be sent, so why should Nagle work
here?

nouser2013, 2016-02-07:
well, kinda my point. Why is the `setNoDelay()` function exposed, then? It's rendered effectively useless with the forced sending, is it not? Shouldn't we let lwip handle sending completely?

igrr, 2016-02-07:
OK, that's the question of semantics. WiFiClient::write should return the
number of bytes delivered, so there is no way to aggregate data from
multiple WiFiClient::write calls into a single packet.
I understand that this creates performance issues. I'll take a look if we
can relax this requirement without breaking too much code. For instance I
know that people have WiFiClient::write calls at the end of functions where
WiFiClient is a local variable. Currently WiFiClient destructor may call
tcp_abort, so the data from the final write may not be sent yet if we don't
wait for it to be ACKed. Changing WiFiClient::write semantics can
potentially break a lot of things, and I'm not sure that increased
performance is worth that. But I will take a look, thanks for pointing this
out.

drmpf, 2016-02-08:
The problem of breaking that type of code could be avoided by providing a buffered wrapper for WiFiClient that the user can use if they want.

nouser2013, 2016-02-19:
A tiny update here: I am using latest GIT code and still it's not behaving as expected. @igrr , would it be a compromise to intoduce a `WiFiClient::confirmReception()` method or, semantically more correct `WiFiClient::waitForResult()` and otherwise remove all the strategy code again (doesn't seem to work anyway)?

Then I could enable Nagle, fill the buffer with a lot of tiny writes, LWIP sends large packets and I either get blocked by the `WiFiClient::write()` call for as long as there is no room in the buffer, or I get blocked by the new function as long as there is no ACK from the other side. `WiFiClient::confirmReception()` could at the same time return an (LWIP-) error code.

Regarding the destructor issue: You could make that configurable, meaning option 1="abort on destroy", 2="block until result (ACK/NACK)", where of course one could no react to anymore, but hence the new function to cope with that.

I haven't had coffee yet so this may be complete misthinking :)

igrr, 2016-02-19:
I haven't yet merged the "strategies" branch (and will not merge it in the current state anyway), so I don't think there is anything that needs to be "removed" from the latest git code:)
I'm thinking about the way to introduce a "switch" between blocking and non-blocking mode. Having a getter and setter is a poor option, but I have an idea how to make this distinction type-safe.

All these changes are currently on hold though. I'm working on an automated test system which is needed to ensure these "optimizations" do not introduce regressions in existing code.

cnlohr, 2016-10-04:
Did anyone ever find a solution to this with the standard SDK?  I can't seem to send a null-push using their stack, it's like they specifically detect it and are like "nope, you're not allowed to fix that".  TCP null-pushs are the best solution I've found to get around this for situations where you need an ack immediately from Windows, and they work quite well!

https://github.com/cnlohr/avrcraft/blob/master/ipcore/tcp.c line 451 - basically just creates an empty TCP psh packet and Windows says "aaha, I'll IMMEDIATELY respond!"  When I try anything with the ESP Open SDK, it refuses to send anything at all.

devyte, 2017-10-18:
@igrr what is the status of this? It seems to me that current performance is far better than what is explained here, so maybe this no longer applies?

d-a-v, 2018-03-04:
In current git master, nagle (default) and ```client.setNoDelay(true)``` are both working. It is quite hard to detect because the nagle's not-sending-condition is rather complex. I could see nagle in action with this sketch:
```
#include <ESP8266WiFi.h>

const char* ssid = "open";
const char* psk = "";

WiFiServer server(51);
WiFiServer server2(52);
WiFiServer server3(53);
char* data;
const int datasize = 10000;

void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println(datasize);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, psk);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print('.');
    delay(500);
  }
  Serial.println(WiFi.localIP());
  server.begin();
  server2.begin();
  server3.begin();

  data = (char*)malloc(datasize);
  memset(data, 'a', datasize);
  data[datasize - 2] = 'z';
  data[datasize - 1] = 0;
}

struct tcp_pcb;
extern struct tcp_pcb* tcp_tw_pcbs;
extern "C" void tcp_abort (struct tcp_pcb* pcb);
void tcpCleanup (void) {
  while (tcp_tw_pcbs)
    tcp_abort(tcp_tw_pcbs);
}

void loop()
{
  tcpCleanup();

  {
    WiFiClient client = server.available();
    if (client)
    {
      Serial.print(client.write(data, datasize));
      Serial.println(-(int)ESP.getFreeHeap());
    }
  }

  {
    WiFiClient client2 = server2.available();
    if (client2)
    {
      client2.setNoDelay(false);
      Serial.println("nagle");
      client2.write("nagle1\n",8);
      client2.write("nagle2\n",8);
      client2.write("nagle3\n",8);
      client2.write("nagle4\n",8);
      delay(1000); // prevent flush
    }
  }

  {
    WiFiClient client3 = server3.available();
    if (client3)
    {
      client3.setNoDelay(true);
      Serial.println("smast");
      client3.write("smast1\n",8);
      client3.write("smast2\n",8);
      client3.write("smast3\n",8);
      client3.write("smast4\n",8);
    }
  }
}
```

On the PC, I run a ```tcpdump```, and on another console I run ```nc```, both below.

On port 52, nagle is enabled:
```
$ nc 192.168.1.239 52
nagle1
nagle2
nagle3
nagle4
04:22:39.656191 IP 192.168.1.118.37488 > 192.168.1.239.52: Flags [S], seq 2005212935, win 29200, options [mss 1460,sackOK,TS val 2024108028 ecr 0,nop,wscale 7], length 0
04:22:39.658353 IP 192.168.1.239.52 > 192.168.1.118.37488: Flags [S.], seq 12171, ack 2005212936, win 5840, options [mss 1460], length 0
04:22:39.658505 IP 192.168.1.118.37488 > 192.168.1.239.52: Flags [.], ack 1, win 29200, length 0
04:22:39.661706 IP 192.168.1.239.52 > 192.168.1.118.37488: Flags [.], seq 1:9, ack 1, win 5840, length 8
04:22:39.661772 IP 192.168.1.118.37488 > 192.168.1.239.52: Flags [.], ack 9, win 29200, length 0
04:22:39.665078 IP 192.168.1.239.52 > 192.168.1.118.37488: Flags [.], seq 9:33, ack 1, win 5840, length 24
04:22:39.665180 IP 192.168.1.118.37488 > 192.168.1.239.52: Flags [.], ack 33, win 29200, length 0
04:22:39.668435 IP 192.168.1.239.52 > 192.168.1.118.37488: Flags [F.], seq 33, ack 1, win 5840, length 0
04:22:39.668656 IP 192.168.1.118.37488 > 192.168.1.239.52: Flags [F.], seq 1, ack 34, win 29200, length 0
04:22:39.671705 IP 192.168.1.239.52 > 192.168.1.118.37488: Flags [.], ack 2, win 5839, length 0

```
We can see 2 packets of respective lengths 8 and 24(=3*8). It is expected that we don't have 4 8-bytes packets.
The first one is not nagled for an unknown reason to me (you are free to [enlighten me](https://github.com/esp8266/Arduino/blob/a2e7c7d0166c0ded47509fda1ec6542beb1e67ce/tools/sdk/lwip/include/lwip/tcp_impl.h#L84))

On port 53, nagle is disabled:
```
$ nc 192.168.1.239 53
smast1
smast2
smast3
smast4
04:29:33.814526 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [S], seq 969597367, win 29200, options [mss 1460,sackOK,TS val 2024522173 ecr 0,nop,wscale 7], length 0
04:29:33.817638 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [S.], seq 12221, ack 969597368, win 5840, options [mss 1460], length 0
04:29:33.817758 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [.], ack 1, win 29200, length 0
04:29:33.821423 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [.], seq 1:9, ack 1, win 5840, length 8[|domain]
04:29:33.821491 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [.], ack 9, win 29200, length 0
04:29:33.824718 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [.], seq 9:17, ack 1, win 5840, length 8[|domain]
04:29:33.824784 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [.], ack 17, win 29200, length 0
04:29:33.828075 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [.], seq 17:25, ack 1, win 5840, length 8[|domain]
04:29:33.828142 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [.], ack 25, win 29200, length 0
04:29:33.828243 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [.], seq 25:33, ack 1, win 5840, length 8[|domain]
04:29:33.828261 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [.], ack 33, win 29200, length 0
04:29:33.831584 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [F.], seq 33, ack 1, win 5840, length 0
04:29:33.831844 IP 192.168.1.118.41252 > 192.168.1.239.53: Flags [F.], seq 1, ack 34, win 29200, length 0
04:29:33.834962 IP 192.168.1.239.53 > 192.168.1.118.41252: Flags [.], ack 2, win 5839, length 0
```
As expected, we have four packets of size 8. I can't explain tcpdump's ```[|domain]``` yet.

This is with lwIP-v1.4. lwIP-v2 behaves the same.

To be noted: results are not the same with ```client.println("hello");``` because ```println``` writes two times (one for the text, another one for ```\n```, and it may be calling ```::flush``` or doing similar).

I can confirm ```tcp_output()``` **does not** superseeds nagle.

You can see that there is no PUSH flags in my logs. This is because I am experimenting things in ClientContext, and I am too lazy to re-copy-paste real logs that are the same with the P flag.

lwIP and ClientContext seems OK here, and if the issue persists, try with current git master. Then we would have to watch into libraries.

liebman, 2018-03-04:
> I can't explain tcpdump's [|domain] yet.

Port 53 is DNS, its trying to decode the DNS request....

d-a-v, 2018-03-04:
of course :) thanks
I will choose my ports number more carefully in my tests

hossamfadeel, 2019-04-26:
I know it is too late to respond, however, maybe someone looking for solution of this issue and come through this. 
So, what I have done and it solved my issue was if the buffer out of memory the wait for the acknowledge to come (By receiving the Ack it means there will be a free space in your buffer for sending further data)

Hope it helps.

Regards,

devyte, 2019-04-26:
@hossamfadeel can you please provide more details?

hossamfadeel, 2019-04-27:
> @hossamfadeel can you please provide more details?

OK.
Assume you have code that you're using for sending a large amount of data once. If you just rely on sending only then you buffer will stuck and you will get out of memory (tcp_out error message -1). 
I solved this by doing the following pseudo code.

while(your condition of sending is true)
{
1- data = whatever data you want to send (divide them to be less or equal to the maximum buffer size or you can use multiples of the MSS (Maximum Segment Size) - I used 2*MSS - but I think it can be more ---I didn't try to increase as it suffice my requirements. )
while(tcp_send_buff > length of your data)
{
2- use tcp_write (by determining the data you want to send and the size you are considering)
3- use tcp_out to output the data immediately -- this doesn't guarantee your buffer will be free so you will need to go to step 4
= if tcp_send_buff buffer size is less than data size then the loop will break and goes to outer loop
}

4- enable receiving ack from the receiver side --- you can use callback function to notify you that ACK has been received --- once ACK is received, it means there is some free space in the sending buffer so you can send other data 

}

Hope I answered you.

Regards,

devyte, 2019-04-27:
Why do you manually fragment to MSS? Fragmentation should be automatic, done within lwip.

@d-a-v assuming this makes sense to you, is there any part that you think has merit?

d-a-v, 2019-04-27:
This pseudo-code is targetting lwIP API. This is common sense about how TCP works.
WiFiClient (probably AsyncLibs too) is implemented that way and its use is [documented](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/client-examples.html?highlight=availableforwrite#general-client-loop). Additions or comments are welcome.

As said by @hossamfadeel, fragmentation is necessary. With MSS, it is fair but using the hacking way.
This is generalized with WiFiClient::availableForWrite(). TCP will always fragment automatically to chunks of MSS.

devyte, 2019-10-17:
Ok I don't see anything further to do here, so I'm closing. If anything is pending, let's handle it in a new issue.

closed 2019-10-17 by devyte