esp8266/Arduino #1578: AP + STA mode need to enable

Shaheers, 2016-02-05:
hi, its found that as in AT mode we can enable AP + client mode and also Lua have this capability to oprate in that mode, if we implement that in it, we can the enable the mash network buy using a simple protocol or either use MQTT or such

marvinroger, 2016-02-05:
Actually, AP + client mode is [already implemented](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp#L43), just like [a mesh library](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFiMesh/examples/HelloMesh/HelloMesh.ino).

pgScorpio, 2016-06-15:
But all "standard" services like mdns, webserver and ftp only bind to the AP interface in this mode and can't be accessed via STA interface !!!!

devyte, 2017-10-03:
@Shaheers wifi mode with AP+STA is already implemented.
Closing.

closed 2017-10-03 by devyte