esp8266/Arduino #1579: Suggestion for annotated Figure "ESP-to-Serial"

pcvm, 2016-02-05:
I have an annotated circuit diagram that might help newcomers to the ESP8266.

It isn't a good idea to have a 3.3V supply on an esp8266 directly connected to what is typically a limited output 3.3V supply of the USB adapter. I've broken that direct link in the circuit and added a note to say that the power needs of the esp8266 justify a separate supply.

Also, target systems can have programs downloaded and launched successfully but doing something as simple as opening the serial comms window (or quitting from the Arduino IDE) can cause a reset of the esp8266. Suggesting a dual pole switch on the RTS and DTR lines might might reduce confusion.

Hope this is of use.

[image]

h4rm0n1c, 2016-02-06:
It's appreciated, but probably unnecessary:
1. The Power supply issue is already documented. (https://github.com/esp8266/Arduino/blob/master/doc/boards.md#generic-esp8266-modules)
2. RTS/DTR is a software issue with the serial terminal in the Arduino IDE, other terminal software generally doesn't do this, you'd be better off with Termite anyway. (http://www.compuphase.com/software_termite.htm)

pcvm, 2016-02-06:
OK although the error of connecting two power supplies in parallel will remain. Thanks.

closed 2016-06-02 by igrr