esp8266/Arduino #1580: WifiClient print in loops - bad performance

moelski, 2016-02-05:
Hi !

I have an Array which is stored in the RAM (24*12 float values). When I try to send the data with a WifiClient in this way: 
```
    for ( byte i = 0; i < 24; i++ ) {
      for ( byte j = 0; j < 12; j++ ) {
         Client.print(day_temp[i*12 + j]);
         Client.print(" ");
      }
      Client.print("\r\n");
    }
```
it takes really long to finish sending the data. It´s about half a minute!

If I send it in one block all is ok and fast (data is a String): 
```
    for ( byte i = 0; i < 24; i++ ) {
      for ( byte j = 0; j < 12; j++ ) {
         data.concat(day_temp[i*12 + j]);
         data.concat(" ");
      }
      data.concat("\r\n");
    }
    client.println(data);
```

So why is it so damn slow to send in a Loop? 

The same Problem exists in ESP8266WebServer class. But I think it is the same Problem ... 

Any ideas? 

Dominik

igrr, 2016-02-05:
Each time you call print, WiFiClient will send data, wait for an ACK to
arrive, and return the number of bytes delivered to the client. So each
write incurs a penalty of approximately twice the network latency. This is
done so that write method is reliable, i.e. it returns exactly the number
of bytes delivered.

moelski, 2016-02-05:
Hi igrr, 

thx for that Information. 
But what is the best way to send something like an Array? 
Since my Array isn´t really large it works now. But if you Need to send larger data structures this could be a Problem. 

Dominik

drmpf, 2016-02-05:
My version of the ESP8266WiFi library, pfodESP2866WiFi,
on http://www.forward.com.au/pfod/pfodParserLibraries/index.html
 includes a buffered client which buffers up to a full packet before sending 
OR just sends what it has after 10mS of no writes.

You could extract that buffer code to use in your application at a higher level.

igrr, 2016-02-05:
I have a pull request up for review which introduces different strategies
for WiFiClient, and it already improves performance for large payloads. I
imagine adding buffering strategy for small payloads will be pretty
trivial. I'll update the pull request.

closed 2016-06-02 by igrr

flodmotorgrodan, 2017-11-14:
I noticed long delays in "SEND OK" from the device even though data has already reached the destination server been validated and sent back.

Waiting for someone to explain at the Espressif forum:
[http://bbs.espressif.com/viewtopic.php?f=16&t=8348&p=18141#p18141](http://bbs.espressif.com/viewtopic.php?f=16&t=8348&p=18141#p18141 )

devyte, 2017-11-14:
Try latest git with lwip2.

flodmotorgrodan, 2017-11-28:
Strange. No response to the "SEND OK" delay [problem reported](http://bbs.espressif.com/viewtopic.php?f=16&t=8348&p=18141#p18141). To me it looks like a major bug in the ESP2866 firmware. Has this issue been addressed in some ESP2866 firmware update?

devyte, 2017-11-28:
@flodmotorgrodan first, this repo has nothing to do with the AT command firmware. 
Then, all comm layers up to link-layer are handled by the Espressif proprietary SDK code, to which we don't have access. Comm layers above link-layer, including TCP/UDP, are handled by lwip, to which we have access, but don't own. The code in this repo is built on top of the Espressif SDK and lwip, and user apps that use this code are built on top of that. What I mean by all this is: "major bug in the ESP8266 firmware" is very vague.
And finally, if you can reproduce the issue with the firmware in this repo, then I already suggested trying a build with lwip2, which solves a whole lot of issues present in lwip1.4 (distributed with the Espressif SDK). I think that includes some observed delays in sending replies. You'll need latest git for that.

devyte, 2017-11-28:
Oh, and Espressif is not very good at replying about the SDK lately, they are lately mostly focused on IDF for the ESP32.